ControlSystemStudio/cs-studio #2378: How to interact (via script?) with Data Browser widget

claudio-rosati, 2018-01-22:
Hello all,

a colleague would like to have the channels displayed in a Data Browser widget being made visible/hidden through check-boxes somewhere else in the OPI itself. The idea is similar to what done in the `2_5_2_XY_Graph.opi` example.

Is that possible to achieve? How (via scripts, for example)?

Thank you in advance for your help.

kasemir, 2018-01-22:
I'm reluctant to hand out script examples.
If people need to ask how to implement a script, maybe they shouldn't use them, because they'll just have trouble maintaining them down the road.
But here we go...

Attach for example a checkbox widget to 'loc://show(1)'.
Then attach a script to the data browser widget, triggered by 'loc://show(1)':

````
# widget: Data Browser widget
# pvs[0]: PV used to show or hide the first trace
#
# Ideally, could be expanded to allow any number of input PVs,
# and then loop over the pvs[i] array to update trace[i].setVisible(..)

from org.csstudio.opibuilder.scriptUtil import PVUtil
show = PVUtil.getLong(pvs[0]) > 0

# To implement this example, or to later update it from
# BOY to another version of the display tool:
#
# 1) Check what the widget class is -> DataBrowserWidgedEditPart
#     print widget.__class__.__name__
# 2)  Use the force, read the source, and figure out how to manipulate the widget class.
#      The DataBrowserWidgedEditPart and Data browser model API are
#      pretty straight forward except that the model items are only accessible by name,
#      or via an iterator. To get the "first" trace thus looks like this:
trace = widget.getDataBrowserModel().getItems().iterator().next()

# Now change the trace
trace.setVisible(show)
````

kasemir, 2018-01-22:
````
# widget: Data Browser widget
# pvs[0]: PV used to show or hide the first trace
# pvs[1]: PV used to show or hide the second trace
# pvs[..]: As many as you want
from org.csstudio.opibuilder.scriptUtil import PVUtil
items = widget.getDataBrowserModel().getItems()
for i in range(len(pvs)):
    show = PVUtil.getLong(pvs[i]) > 0
    trace = items.iterator().next()
    trace.setVisible(show)
````

claudio-rosati, 2018-01-22:
Thank you Kay,

I’ll forward to the requester.

The real problem here is not the script by itself, but the fact that the Display Builder XY Graph differs a lot from the BOY one, and the working example from BOY uses the XY Graph.

Can be worth adding to the Data Browser widget a property showing/hiding a list of check boxes allowing to show/hide the channels?

Best regards,
Claudio

_____________________________

Claudio Rosati
Software Engineer
Software Group, ICS Division

European Spallation Source ERIC
P.O Box 176, SE-221 00 Lund, Sweden
Visiting address: Tunavägen 24, Lund

Phone: +46 46 888 34 24
Mobile: +46 721 79 24 24
E-mail: claudio.rosati@esss.se<mailto:petra.ribb@esss.se>

www.esss.se<http://www.esss.se/>

[cid:F13AFA79-E9F7-44E8-95C8-95C364416F06]

_____________________________

EMAIL CONFIDENTIALITY NOTICE

This message is private and confidential.
If you have received this message in error,
please notify me and remove it from your system.