ControlSystemStudio/cs-studio #2383: EDM to OPI conversion of group content results in negative coordinates

kasemir, 2018-01-30:
When converting groups in *.edl files, the converter adjusts the coordinates of the group content in a way that they can end up with negative coordinates.
That results in content which is then clipped by the outline of the group.

demo1.edl contains a group with x=47 and width=16.
Inside that group are two arcs.
One with x=47 and width=8, another with x=55 (=47+8) and width=8.
Overall, the two arcs thus cover a width of 16, matching the width of the group.

When converted to *.opi, the coordinates of the content are adjusted because they need to be relative to the group, not the display.
The first arc ends with x=-1, the second with x=7, which seems to be off by one pixel.

When this group is further embedded into another group as in demo2.edl, the first arc ends up with x=-3, and it's visibly clipped.

[image]

Because of the internal differences between group handling in EDM and BOY it's understandable that the coordinates might need adjustment, but that shouldn't clip the content.
If the content needs to move by 1 pixel to the left to remain in the same place, maybe that means the group needs to move 1 pixel left, instead of placing the content at x=-1?

[edl.zip](https://github.com/ControlSystemStudio/cs-studio/files/1678491/edl.zip)

mfurseman, 2018-02-13:
At Diamond we have a set of post processing Python scripts that we run after conversion. As the Java converter creates one to one conversions of widgets these Python scripts are used to tidy the conversions by looking at the relationship between widgets. One of these [scripts](https://github.com/dls-controls/css-converter-python/blob/0dbfb42761fdcdd1b988cb32eefc5c57e3e00224/convert/groups.py) resizes the grouping boxes to fit their contents while keeping the same absoloute child coordinates.

kasemir, 2018-02-13:
Excellent, you confirm exactly what I was seeing:
"in EDM widgets which have dimensions exceeding the grouping container are drawn, in CSS these widgets are clipped to the grouping container area. <..need to modify groups> so that they match the contents."

Would it make sense to put that logic into the converter, i.e. into the Java code? At first glance, the `Opi_activeGroupClass`, https://github.com/ControlSystemStudio/cs-studio/blob/master/applications/opibuilder/opibuilder-plugins/org.csstudio.opibuilder.converter/src/main/org/csstudio/opibuilder/converter/writer/Opi_activeGroupClass.java, is what's writing the `groupingContainer`. It has all the child widget information, so it could patch the sizes and positions.

Do you remember if there was a reason to instead handle this with a python script after the conversion? Something that makes it impractical to be handled within the converter?

mfurseman, 2018-02-14:
We were already using Python scripts to fix other issues, looking at the Java converter there doesn't seem to be a good reason not to have done the conversion there.

kasemir, 2018-02-14:
OK, I'll try to add the correction to the Java code.
Will your python script be OK as long as the child widget coordinates are all located inside the group? I.e. if there's nothing to correct, it'll leave it untouched?

mfurseman, 2018-02-14:
I don't believe it will cause us any issues. If the group box is larger than it's content it will still be cropped to fit by the Python script.