ControlSystemStudio/cs-studio #2387: MEDM converter tweaks

kasemir, 2018-01-31:
From Mark Rivers:

..conversions are really good..

Many of the issues we did find were due to problems in the medm files themselves.  medm tolerates text widgets that are too small to actually hold the text.  It also will display an enum value as a string even if the datatype of the widget is "decimal" rather than string.  I have fixed these problems in the medm files..

.. a few remaining issues ..

- When a widget has color mode "alarm" in medm the colors are NO_ALARM=green, MINOR=yellow, MAJOR=red.  In CSS we get NO_ALARM=black, MINOR=orange, MAJOR=red.  This leads to displays that are very difficult to read, particularly when NO_ALARM text is on a dark background.  Could we make the alarm colors in the autoconverted files match medm?

- Text graphics widgets with Right Justify are not right justified in CSS.  It looks like Left and Center justify are OK.  The same is true for text update widgets.

- Slider widgets do not convert well at all.  Note that the CSS widget is tiny, with unreadable text.

- When a related display with multiple displays has a label (e.g. "Multiple displays" in attached medm screen) that label does not appear in CSS.

If these problems were fixed I think that the conversion tool would be really excellent. The autoconverted screens would be sufficient in nearly all cases, with no need for tweaking.

kasemir, 2018-01-31:
Example MEDM display:

[image]

Original BOY conversion that lacks in 'right' alignment, slider, ..:

[image]

Conversion with updated code:

[image]

MarkRivers, 2018-01-31:
Hi Kay, 

That's great, thanks!  

How hard would it be to make the MINOR color be yellow instead of orange?  I think the orange is too similar to the red.

kasemir, 2018-01-31:
The colors for the Minor, Major, Invalid, Disconnected states are defined in a color.def file.
That's outside of the *.opi file itself.

In this screenshot I created a color.def that uses Minor=Yellow, configured that in the preferences (, restarted CSS):

[image]

MarkRivers, 2018-01-31:
OK, thanks.  color.def does not define the color for NO_ALARM, you made that green in the converted widget itself?

kasemir, 2018-01-31:
Yes, for no alarm it uses the 'foreground' color.

.. which I made rgb(0, 255, 0) i.e. green.
Maybe I should have made that 'OK', so then that's also configurable via the color.def

MarkRivers, 2018-02-06:
This request seems to have been merged.  However, when I download the nightly build from today (Feb. 6) the resulting directory when I unzip has a date of Jan. 8, 2018 and does not contain the fix.  Is there something wrong with the nightly builds?
```
@corvette CSS]# ls -l
total 176936
drwxrwxr-x 10 root root      4096 Jan  8 13:06 basic-epics-4.5.0
-rwxr-xr-x  1 root root 181175121 Feb  6 10:14 basic-epics-4.5.0-linux.gtk.x86_64.zip
[root@corvette CSS]# ls -l basic-epics-4.5.0
total 212
-rw-rw-r--  1 root root 96198 Jan  8 13:06 artifacts.xml
drwxrwxr-x  5 root root   108 Jan  8 13:06 configuration
-rwxr-xr-x  1 root root 80385 Jan  8 12:59 css
-rw-rw-r--  1 root root   495 Jan  8 13:06 css.ini
drwxrwxr-x  3 root root    24 Jan  8 13:06 diirt
drwxrwxr-x  2 root root     6 Jan  8 13:06 dropins
drwxrwxr-x 36 root root  4096 Jan  8 13:06 features
drwxrwxr-x  2 root root     6 Jan  8 13:06 logs
drwxrwxr-x  4 root root    76 Jan  8 13:06 p2
drwxrwxr-x  5 root root 24576 Jan  8 13:06 plugins
drwxrwxr-x  2 root root    32 Jan  8 13:06 readme
```

kasemir, 2018-02-07:
> -rwxr-xr-x  1 root root 181175121 Feb  6 10:14 basic-epics-4.5.0-linux.gtk.x86_64.zip
Not sure. Could this be a web caching issue?
When I look on the web server,  these are the 'nightly' snapshots for 2018:

$ ls -l nightly_2018_*/basic-epics-4.5.0-linux.gtk.x86_64.zip 
-rw-rw-r-- 1 kasemir kasemir 181179749 Jan  8 21:42 nightly_2018_01_08/basic-epics-4.5.0-linux.gtk.x86_64.zip
-rw-rw-r-- 1 kasemir kasemir 181181889 Jan  9 17:49 nightly_2018_01_09/basic-epics-4.5.0-linux.gtk.x86_64.zip
-rw-rw-r-- 1 kasemir kasemir 181182943 Jan 10 21:43 nightly_2018_01_10/basic-epics-4.5.0-linux.gtk.x86_64.zip
-rw-rw-r-- 1 kasemir kasemir 181185175 Jan 11 16:49 nightly_2018_01_11/basic-epics-4.5.0-linux.gtk.x86_64.zip
-rw-rw-r-- 1 kasemir kasemir 181185251 Jan 12 11:49 nightly_2018_01_12/basic-epics-4.5.0-linux.gtk.x86_64.zip
-rw-rw-r-- 1 kasemir kasemir 181185389 Jan 15 13:11 nightly_2018_01_15/basic-epics-4.5.0-linux.gtk.x86_64.zip
-rw-rw-r-- 1 kasemir kasemir 181185099 Jan 16 10:49 nightly_2018_01_16/basic-epics-4.5.0-linux.gtk.x86_64.zip
-rw-rw-r-- 1 kasemir kasemir 180235965 Jan 17 13:52 nightly_2018_01_17/basic-epics-4.5.0-linux.gtk.x86_64.zip
-rw-rw-r-- 1 kasemir kasemir 180236945 Jan 18 12:51 nightly_2018_01_18/basic-epics-4.5.0-linux.gtk.x86_64.zip
-rw-rw-r-- 1 kasemir kasemir 180236820 Jan 19 15:50 nightly_2018_01_19/basic-epics-4.5.0-linux.gtk.x86_64.zip
-rw-rw-r-- 1 kasemir kasemir 180242037 Jan 22 13:54 nightly_2018_01_22/basic-epics-4.5.0-linux.gtk.x86_64.zip
-rw-rw-r-- 1 kasemir kasemir 180241752 Jan 23 12:51 nightly_2018_01_23/basic-epics-4.5.0-linux.gtk.x86_64.zip
-rw-rw-r-- 1 kasemir kasemir 180458387 Jan 24 13:49 nightly_2018_01_24/basic-epics-4.5.0-linux.gtk.x86_64.zip
-rw-rw-r-- 1 kasemir kasemir 180464423 Jan 25 16:50 nightly_2018_01_25/basic-epics-4.5.0-linux.gtk.x86_64.zip
-rw-rw-r-- 1 kasemir kasemir 180464748 Jan 26 11:48 nightly_2018_01_26/basic-epics-4.5.0-linux.gtk.x86_64.zip
-rw-rw-r-- 1 kasemir kasemir 180497523 Jan 29 21:46 nightly_2018_01_29/basic-epics-4.5.0-linux.gtk.x86_64.zip
-rw-rw-r-- 1 kasemir kasemir 181264369 Jan 30 22:00 nightly_2018_01_30/basic-epics-4.5.0-linux.gtk.x86_64.zip
-rw-rw-r-- 1 kasemir kasemir 181267070 Jan 31 13:37 nightly_2018_01_31/basic-epics-4.5.0-linux.gtk.x86_64.zip
-rw-rw-r-- 1 kasemir kasemir 180140405 Feb  1 13:32 nightly_2018_02_01/basic-epics-4.5.0-linux.gtk.x86_64.zip
-rw-rw-r-- 1 kasemir kasemir 180141881 Feb  2 15:50 nightly_2018_02_02/basic-epics-4.5.0-linux.gtk.x86_64.zip
-rw-rw-r-- 1 kasemir kasemir 180141963 Feb  5 11:50 nightly_2018_02_05/basic-epics-4.5.0-linux.gtk.x86_64.zip
-rw-rw-r-- 1 kasemir kasemir 180141750 Feb  6 21:43 nightly_2018_02_06/basic-epics-4.5.0-linux.gtk.x86_64.zip

What I don't understand: None matches the exact file size that you see.

When I download the latest nightly, the size matches the one built last night.

$ ll -d basic-epics-4.5.0*
drwxrwxr-x 10 ky9 wheel      4096 Feb  7 08:50 basic-epics-4.5.0
-rw-r--r--  1 ky9 wheel 180141750 Feb  7 08:48 basic-epics-4.5.0-linux.gtk.x86_64.zip

It does include the command-line access to the EDM converter, https://github.com/ControlSystemStudio/cs-studio/issues/2385,
so it's based on sources newer than the Jan. 8 that you see:

[ky9@diane Downloads]$ basic-epics-4.5.0/css -nosplash -application org.csstudio.opibuilder.converter.edl
Feb 07, 2018 8:51:20 AM org.csstudio.diirt.util.Activator start
INFO: Starting diirt.util
************************
** EDL File Converter **
************************

MarkRivers, 2018-02-07:
My apologies, I think I downloaded the wrong file, sns-css rather than basic-epics.  But I was unzipping basic-epics, so it was an old version.  I have the new version now.

MarkRivers, 2018-02-07:
The conversions with the new version look really good.  Here is the test screen, modified so there are backgrounds to judge the text justification.

This the medm version.
[image]

This is the CSS version.
[image]

There does appear to be a problem with text justification on embedded displays.  

This is the medm version.
[image]

This is the CSS version.
[image]

Note that in the medm version all of the text is correctly right justified.  The left hand part of the display is an embedded display, NDPluginBase.adl.

In CSS the right hand side of the screen the text is correctly right justified.  However, on the left hand side (embedded display of NDPluginBase.opi) the text is not correctly right justified.  It is also smaller than it should be.  Perhaps this is because CSS adds an additional border around the embedded display which makes the text shrink?  That would be OK, but the lack of justification is not good.

If I open NDPluginBase.opi as a stand-alone display it looks fine.
[image]

Perhaps there is something I am not understanding about how embedded displays work in CSS?  I assume that it is loading the embedded display at run-time, not using some version it found during the conversion processes?

kasemir, 2018-02-07:
I assume it's related to the fonts and the embedded display resize.
The *.opi file should be created with fixed, pixel-based font size, and the embedded display without the resize option.

Can you email me the file, the 'host' and the 'embedded' adl file?

MarkRivers, 2018-02-07:
I just manually edited NDStats.opi.  It was sufficient to change the border_style in the linkingContainer widget  from 3 to 0.

Now it looks perfect!
[image]

kasemir, 2018-02-08:
Can you email the 'host' and the 'embedded' adl file?
I'll see if the resizing or the border style can be created such that no patching is required.

kasemir, 2018-02-08:
Just noticed you did email them. Thanks, will look at them.

MarkRivers, 2018-02-08:
I think all you need to do is change the converter so that the border_style in the linkingContainer widget is 0, not 3.

kasemir, 2018-02-08:
There were two issues.
The linking container was created to resize the embedded content to fit the container.
.. and it was created with a border.
The widget was created with the same size as in MEDM, but then the border reduced the actual space available for the content. The content was shrunk, and in the process the fonts/positions got disturbed.

With the update, the linking container is created without border, and without resizing.

MarkRivers, 2018-02-09:
I have used today's nightly build and the embedded displays all look great, thanks!

Another question.  I notice that "git diff" shows me this for each converted file:
-  <boy_version>5.1.0.201801081728</boy_version>
+  <boy_version>5.1.0.201802091122</boy_version>

This means that each time I re-do the conversion with the latest release of CSS the opi files will appear to change, even if the actual contents have not.  This does not play well with version control.

Is this information generated by the converter, or is it always added when opi files are written?  If the former, can then we get rid of it?  If the latter, then this should also be a problem if one simply reads an opi file into CSS and writes it out again, unchanged?  Version control would say it had changed?

kasemir, 2018-02-09:
It's not in the converter, it's in BOY.
I've created a pull request to only use the version number without 'qualifier' i.e. build date.

kasemir, 2018-04-13:
Next issue: this plot of waveform over array index:
```
"cartesian plot" {
        object {
                x=10
                y=92
                width=480
                height=230
        }
        plotcom {
                title="$(P)$(R)$(PLT)"
                xlabel="Time point"
                ylabel="Value"
                clr=14
                bclr=2
        }
        style="line"
        erase_oldest="plot last n pts"
        trace[0] {
                ydata="$(P)$(R)$(PLT)"
                data_clr=27
                yaxis=0
        }
        x_axis {
                rangeStyle="auto-scale"
        }
        y1_axis {
                rangeStyle="auto-scale"
        }
        y2_axis {
                rangeStyle="auto-scale"
        }
}
```

Converter stops on `yaxis=0`, and plot of waveform over index currently needs a 'buffer size'

PR #2388 (2018-02-02): Medm cvt 2387
Fixes #2387
closed 2018-06-12 by berryma4