ControlSystemStudio/cs-studio #2394: Save & Restore: incorrect delta between live value and stored set point

claudio-rosati, 2018-02-13:
Using a simple integer value as PV value, Save & Restore in CS Studio erroneously shows a non-zero delta value even if stored set point and live set point are equal, or a wrong value. See attached screen shots:

[image]
[image]

In other cases (e.g. doubles) the delta is displayed correctly.

@georgweiss an I missing something?

kasemir, 2018-02-13:
No idea about the specifics of the save/restore tool, but your PV is indicated as never processed (UDF_ALARM, zero seconds since epoch for time stamp). So the tool could consider this as PV-has-no-valid-value. Do you still see the error when you process the PV?

georgweiss, 2018-02-13:
I'm not experienced enough in Epics to say for sure, but attached screen shot indicates that the PV does indeed have a value.
[image]