ControlSystemStudio/cs-studio #2400: Problems with ChannelFinder caused by java.lang.ClassNotFoundException

claudio-rosati, 2018-03-08:
I've not tested Channel since the last 6 months (more or less). Now we need it and I have the following exception (my product includes `com.sun.jersey 1.17.0.v201503170330`, `com.sun.jersey.bundle 1.19.0`, `com.sun.jersey.contribs.jersey-multipart 1.19.0`, `com.sun.jersey.core 1.19.0`, and `javax.ws.rs.jsr311-api 1.1.1`):

```
2018-03-08 16:13:40.912 SEVERE [Thread 1] gov.bnl.channelfinder.api.Activator (configuredClient) - Could not retrieve configured client for ChannelFinder
org.eclipse.core.runtime.CoreException: Plug-in "org.csstudio.utility.channelfinder" was unable to instantiate class "org.csstudio.utility.channelfinder.ChannelFinderClientFromPreferences".
	at org.eclipse.core.internal.registry.osgi.RegistryStrategyOSGI.throwException(RegistryStrategyOSGI.java:194)
	at org.eclipse.core.internal.registry.osgi.RegistryStrategyOSGI.createExecutableExtension(RegistryStrategyOSGI.java:186)
	at org.eclipse.core.internal.registry.ExtensionRegistry.createExecutableExtension(ExtensionRegistry.java:905)
	at org.eclipse.core.internal.registry.ConfigurationElement.createExecutableExtension(ConfigurationElement.java:243)
	at org.eclipse.core.internal.registry.ConfigurationElementHandle.createExecutableExtension(ConfigurationElementHandle.java:55)
	at gov.bnl.channelfinder.api.Activator.configuredClient(Activator.java:61)
	at gov.bnl.channelfinder.api.Activator.start(Activator.java:29)
	at org.eclipse.osgi.internal.framework.BundleContextImpl$3.run(BundleContextImpl.java:774)
	at org.eclipse.osgi.internal.framework.BundleContextImpl$3.run(BundleContextImpl.java:1)
	at java.security.AccessController.doPrivileged(Native Method)
	at org.eclipse.osgi.internal.framework.BundleContextImpl.startActivator(BundleContextImpl.java:767)
	at org.eclipse.osgi.internal.framework.BundleContextImpl.start(BundleContextImpl.java:724)
	at org.eclipse.osgi.internal.framework.EquinoxBundle.startWorker0(EquinoxBundle.java:932)
	at org.eclipse.osgi.internal.framework.EquinoxBundle$EquinoxModule.startWorker(EquinoxBundle.java:309)
	at org.eclipse.osgi.container.Module.doStart(Module.java:581)
	at org.eclipse.osgi.container.Module.start(Module.java:449)
	at org.eclipse.osgi.framework.util.SecureAction.start(SecureAction.java:470)
	at org.eclipse.osgi.internal.hooks.EclipseLazyStarter.postFindLocalClass(EclipseLazyStarter.java:107)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.findLocalClass(ClasspathManager.java:529)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.findLocalClass(ModuleClassLoader.java:325)
	at org.eclipse.osgi.internal.loader.BundleLoader.findLocalClass(BundleLoader.java:345)
	at org.eclipse.osgi.internal.loader.sources.SingleSourcePackage.loadClass(SingleSourcePackage.java:36)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClassInternal(BundleLoader.java:419)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:372)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:364)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.loadClass(ModuleClassLoader.java:161)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	at java.lang.Class.getDeclaredConstructors0(Native Method)
	at java.lang.Class.privateGetDeclaredConstructors(Class.java:2671)
	at java.lang.Class.getConstructor0(Class.java:3075)
	at java.lang.Class.newInstance(Class.java:412)
	at org.eclipse.core.internal.registry.osgi.RegistryStrategyOSGI.createExecutableExtension(RegistryStrategyOSGI.java:184)
	at org.eclipse.core.internal.registry.ExtensionRegistry.createExecutableExtension(ExtensionRegistry.java:905)
	at org.eclipse.core.internal.registry.ConfigurationElement.createExecutableExtension(ConfigurationElement.java:243)
	at org.eclipse.core.internal.registry.ConfigurationElementHandle.createExecutableExtension(ConfigurationElementHandle.java:55)
	at org.eclipse.ui.internal.registry.ViewDescriptor.createView(ViewDescriptor.java:58)
	at org.eclipse.ui.internal.ViewReference.createPart(ViewReference.java:101)
	at org.eclipse.ui.internal.e4.compatibility.CompatibilityPart.createPart(CompatibilityPart.java:278)
	at org.eclipse.ui.internal.e4.compatibility.CompatibilityPart.create(CompatibilityPart.java:316)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:55)
	at org.eclipse.e4.core.internal.di.InjectorImpl.processAnnotated(InjectorImpl.java:966)
	at org.eclipse.e4.core.internal.di.InjectorImpl.processAnnotated(InjectorImpl.java:931)
	at org.eclipse.e4.core.internal.di.InjectorImpl.inject(InjectorImpl.java:151)
	at org.eclipse.e4.core.internal.di.InjectorImpl.internalMake(InjectorImpl.java:375)
	at org.eclipse.e4.core.internal.di.InjectorImpl.make(InjectorImpl.java:294)
	at org.eclipse.e4.core.contexts.ContextInjectionFactory.make(ContextInjectionFactory.java:162)
	at org.eclipse.e4.ui.internal.workbench.ReflectionContributionFactory.createFromBundle(ReflectionContributionFactory.java:105)
	at org.eclipse.e4.ui.internal.workbench.ReflectionContributionFactory.doCreate(ReflectionContributionFactory.java:74)
	at org.eclipse.e4.ui.internal.workbench.ReflectionContributionFactory.create(ReflectionContributionFactory.java:56)
	at org.eclipse.e4.ui.workbench.renderers.swt.ContributedPartRenderer.createWidget(ContributedPartRenderer.java:129)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.createWidget(PartRenderingEngine.java:975)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeCreateGui(PartRenderingEngine.java:651)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$1.run(PartRenderingEngine.java:536)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:42)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.createGui(PartRenderingEngine.java:520)
	at org.eclipse.e4.ui.workbench.renderers.swt.ElementReferenceRenderer.createWidget(ElementReferenceRenderer.java:70)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.createWidget(PartRenderingEngine.java:975)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeCreateGui(PartRenderingEngine.java:651)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeCreateGui(PartRenderingEngine.java:757)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.access$0(PartRenderingEngine.java:728)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$2.run(PartRenderingEngine.java:722)
[... 88 lines cut ...]
	at gov.bnl.channelfinder.api.ChannelFinderClientImpl$CFCBuilder.create(ChannelFinderClientImpl.java:262)
	at org.csstudio.utility.channelfinder.ChannelFinderClientFromPreferences.reloadConfiguration(ChannelFinderClientFromPreferences.java:52)
	at org.csstudio.utility.channelfinder.ChannelFinderClientFromPreferences.<init>(ChannelFinderClientFromPreferences.java:36)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at java.lang.Class.newInstance(Class.java:442)
	at org.eclipse.core.internal.registry.osgi.RegistryStrategyOSGI.createExecutableExtension(RegistryStrategyOSGI.java:184)
	... 145 more
Caused by: java.lang.ClassNotFoundException: com.sun.ws.rs.ext.RuntimeDelegateImpl cannot be found by javax.ws.rs.jsr311-api_1.1.1
	at org.eclipse.osgi.internal.loader.BundleLoader.findClassInternal(BundleLoader.java:461)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:372)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:364)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.loadClass(ModuleClassLoader.java:161)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	at java.lang.Class.forName0(Native Method)
	at java.lang.Class.forName(Class.java:264)
	at javax.ws.rs.ext.FactoryFinder.newInstance(FactoryFinder.java:62)
	at javax.ws.rs.ext.FactoryFinder.find(FactoryFinder.java:155)
	at javax.ws.rs.ext.RuntimeDelegate.findDelegate(RuntimeDelegate.java:105)
	... 158 more
```

Can you help me?

claudio-rosati, 2018-03-15:
No one can help me on that?

shroffk, 2018-03-15:
Hello,
does your product contain '''javax.ws.rs.jsr311-api_1.1.1'''

claudio-rosati, 2018-03-16:
Yes @shroffk it is.
The `javax.ws.rs.jsr311-api_1.1.1.jar` file is inside `org.csstudio.ess.product/repository/target/repository/plugins/`, and  `org.csstudio.ess.product.core.feature` includes the `javax.ws.rs.jsr311-api (0.0.0)` plugin. When I double-click it I see the following:

[image]

claudio-rosati, 2018-03-20:
@shroffk 
That's completely crazy!!! 
I was able to temporarily solve the problem, and the way I succeeded scares me.

Inside the `plugin`  folder of my product (on MacOS) I have the following jar files:

```
com.sun.jersey_1.17.0.v201503170330.jar
com.sun.jersey.bundle_1.19.0.jar
com.sun.jersey.contribs.jersey-multipart_1.19.0.jar
com.sun.jersey.core_1.19.0.jar
```

That is what I've done:

- I've removed the `.metadata` folder from the workspace, run CS-Studio and open the channel viewer: The problem occurred.
- Then I've removed the `.metadata` folder from the workspace and removed the `com.sun.jersey_1.17.0.v201503170330.jar` file. Nothing changed: the problem was still there.
- Then I've removed the `.metadata` folder from the workspace,  put back `com.sun.jersey_1.17.0.v201503170330.jar` and removed the other 3 files (`com.sun.jersey*_1.19.0.jar`). Restarting CS-Studio nothing is displayed in the Channel Viewer because the plugin cannot start.
- I've put back the removed files (`com.sun.jersey*_1.19.0.jar`) _et voilà_ restarting CS-Studio no exceptions and I can do a query (not working because I cannot access the Channel Finder service from home).

Successive run of CS-Studio continue to work. So it seems I should be happy but... if I remove the CS-Studio app from my Mac and extract a fresh new one from the zip file produced by the build phase, I run again in the same problem. So, again, I have to run it once without the `com.sun.jersey*_1.19.0.jar` files, put them back, and then everything run fine again.

Have you an idea of what can I do to remove this strange behaviour?

claudio-rosati, 2018-03-20:
P.S.
inside `org.csstudio.ess.product.core.feature` I've tried to scrambling the following four plugins, putting them all at the beginning or at the end... without any success.

```xml
   <plugin
         id="com.sun.jersey"
         download-size="0"
         install-size="0"
         version="0.0.0"
         unpack="false"/>

   <plugin
         id="com.sun.jersey.core"
         download-size="0"
         install-size="0"
         version="0.0.0"
         unpack="false"/>

   <plugin
         id="com.sun.jersey.bundle"
         download-size="0"
         install-size="0"
         version="0.0.0"
         unpack="false"/>

   <plugin
         id="com.sun.jersey.contribs.jersey-multipart"
         download-size="0"
         install-size="0"
         version="0.0.0"
         unpack="false"/>

```

claudio-rosati, 2018-03-20:
Tested on CentOS, the problems always exists, and never goes away.

kasemir, 2018-03-20:
Just a thought since the symptoms change when you mess with the workspace/.metadata and add/remove *.jar files:
Delete your $HOME/.eclipse folder.
Maybe it contains outdated information for those plugin *.jar files.

shroffk, 2018-03-20:
Ok, I think @kasemir 's solution is a good first check.
If that does not work then I will recommend adding the jars manually to the classpath. This sounds like more of the osgi and SPI not working with each other.

claudio-rosati, 2018-03-21:
@kasemir
Not working, sorry.

@shroffk 
What do you means with "adding the JARS to the classpath" in an eclipse-based application?

kasemir, 2018-03-21:
Well, I really have no idea about the channel finder in general nor this specific piece of code, but one last for what it's worth:

If the .eclipse cleanup doesn't help, that suggests this is a real problem and not a case of temporary plugin class loader confusion.
Since Kunal mentioned SPI, if this is a case of `ServiceLoader.load` called in plugin A not being able to locate a class in plugin B, you can either make plugin A depend on plugin B (which might not be easy if plugin A is a 3rd party jar that you use 'as is') or add the Eclipse buddy class loading hints to the manifest (as we do for some in https://github.com/ControlSystemStudio/maven-osgi-bundles/blob/master/repository/pom.xml, see also http://help.eclipse.org/kepler/index.jsp?topic=%2Forg.eclipse.platform.doc.isv%2Freference%2Fmisc%2Fbuddy_loading.html)

shroffk, 2018-03-21:
Hello,
The jersey libraries required are converted to osgi bundles.

@claudio-rosati are you building the master? can you test my 4.5.1 product and let me know if it works on centos (I have tested on windows and debian 7,8,9)
http://cs-studio.sourceforge.net/nsls2/nsls2.html

looking at my product configuration I noticed that I simply add the channel and channelfinder features. I do not add anything additional in the nsls2 core feature.

https://github.com/ControlSystemStudio/org.csstudio.nsls2.product/blob/master/repository/css-nsls2.product

claudio-rosati, 2018-03-21:
Thank you @shroffk, looking at your product I've understood that I had a messy in my product and ess features, including twice or more the same puglins. Just using the product like your made things working.

closed 2018-03-21 by claudio-rosati

shroffk, 2018-03-21:
yay!!!

claudio-rosati, 2018-03-21:
@shroffk 
Only a question: now I found a Run menu (never seen before) with only one entry – External Tools.
What is it for?