ControlSystemStudio/cs-studio #2401: Update to Eclipse 4.7 Oxygen

willrogers, 2018-03-09:
We want to update CS-Studio to build and run using Java 9. A prerequisite for that is to build against Eclipse Oxygen.

berryma4, 2018-03-09:
let's update to the new diirt release and then branch first.

willrogers, 2018-03-09:
Fine, there's no rush.

kasemir, 2018-03-09:
Eric, you make it sound as if changing diirt is easier than updating to Java 9 ;-)

berryma4, 2018-04-05:
master is open for updating to Oxygen.

berryma4, 2018-04-05:
I merged ControlSystemStudio/cs-studio-thirdparty, did you have a branch for ControlSystemStudio/cs-studio?

berryma4, 2018-04-06:
I made a pull request with the new urls.
We were able to remove efx, because it is now in the Eclipse Release Train (the oxygen url).

berryma4, 2018-04-06:
I've also updated the common product urls, and added the java 9 vmarg
```java
--add-modules ALL-SYSTEM
```

berryma4, 2018-04-06:
Also, with java 9, we should be able to use GTK3:
http://openjdk.java.net/jeps/283

berryma4, 2018-04-16:
tried upgrading to e(fx)clipse 3.0.0
I had issues with the diirt jars/MANFEST resolving javafx.*
I tried adding extraClasspathElements for javafx to the tycho compiler, but this didn't work.
Out of frustration, rolling back to e(fx)clipse 2.4.0
I still removed:
all javafx imports, org.eclipse.fx.javafx from all requires (not needed anymore)

claudio-rosati, 2018-08-06:
About e(fx)clipse... and JDK 11/OpenJFX 11

https://tomsondev.bestsolution.at/2018/08/04/supporting-openjfx-11-from-jdk11-onwards-in-efxclipse/

closed 2021-04-13 by willrogers