ControlSystemStudio/cs-studio #2408: Bad dependancy between Save & Restore and Logbook

claudio-rosati, 2018-03-22:
At ESS I was requested to temporarily remove any reference to Logbook from CS-Studio because the current (temporary) solution is web-based. People in charge of the current commission don't want having menu entries that can make CS-Studio user think it is possible to start logging from CS-Studio.

I've removed the following from my product:

```xml
	<feature id="org.csstudio.applications.logbook.feature"/>
	<feature id="org.csstudio.applications.logbook.fault.feature"/>
	<feature id="org.csstudio.logbook.viewer.feature"/>
	<feature id="org.csstudio.olog.feature"/>
```

and most of the Logbook entries disappeared. There is still a single entry (CS-Studio -> Utilities -> Create Log Entry) that to my understanding is a leftover from the Save & Restore plugin.

When this menu entry is selected, the following is displayed:

[image]

where nothing can be done because no logbook is available.

That said, is it possible if not remove this menu entry, at least having it disabled if no logbooks are defined?

P.S.
I can do it myself if addressed to the right piece of code ;-)

shroffk, 2018-04-19:
> leftover from the Save & Restore plugin

That menu entry is supplied by the "org.csstudio.logbook.ui" plugin.
Can you check if this plugin exists in your product...if it is present then the next step would be to discover how this plugin in pulled in.

claudio-rosati, 2018-04-20:
Hello Kunal,

the only place in my product where I can find the `org.csstudio.logbook.ui` string is inside my `plugin_customization.ini` file:

```
#
# Logbook preferences
#
org.csstudio.logbook.ui/Autenticate.user=true
org.csstudio.logbook.ui/Default.logbook=Operations
org.csstudio.logbook.ui/Default.level=Info
org.csstudio.logbook.ui/Show.history=true
org.csstudio.logbook.ui/Result.size=50
org.csstudio.logbook.ui/Log.url.formatt=http://localhost/logbook/index.html#{logId}_1
org.csstudio.logbook.ui/pvRegex="^.*PV.*?:(.*?)$"
```

shroffk, 2018-04-20:
hmm... that is strange I don't understand how this action could be there if the o.c.logbook.ui plugin is not in the final product.
I will search and see if the menu action is added anywhere else.