ControlSystemStudio/cs-studio #2418: "DataSourceProvider 'beast://' did not create a valid datasource."

MichaelRitzert, 2018-04-05:
I'm trying to display our system's alarm state on an OPI using the beast://
diirt datasource. However, whatever I try, I always receive this error:

```
2018-04-03 23:42:19 ERROR: Error from pv connection layer: java.lang.IllegalStateException: DataSourceProvider 'beast://' did not create a valid datasource.
```

and the PV stays "disconnected". There's no further related output to be found.
I have tried beast://pv-name, beast://area-name, etc. as described in
https://htmlpreview.github.io/?https://github.com/ControlSystemStudio/cs-studio/blob/master/applications/alarm/alarm-plugins/org.csstudio.alarm.diirt.datasource.info/html/beast-datasource.html

I'm using a custom-built CSS product built against the 4.4 branch.

shroffk, 2018-04-19:
- Are you able to view the alarm tree?
- Is there any more error info? a stacktrace etc?

MichaelRitzert, 2018-05-02:
Yes, I can view the alarm tree, interact with it, etc.
And unfortunately, this is the only piece of information I get. I also checked the terminal from which I started CSS.

shroffk, 2018-05-23:
I have feeling this might be a configuration/preferences issue.
In edit preferences goto
CSS Core -> Data Sources
There should be beast.xml preference file
Ensure that it is configured to match your alarm tree root 

here is the NSLS2 example, NSLS2_OPR is the root name
```
<?xml version="1.0" encoding="UTF-8"?>
<beast version="1">
  <!-- JMS connection configuration =================================== 
    Example to configure the dataSource option -->
  <dataSourceOptions configName="NSLS2_OPR"/>
</beast>
```

MichaelRitzert, 2018-05-28:
Thanks for the suggestion. This probably was part of the problem — but not all of it: I created the file, changed the configName as required, and restarted. However I still get the very same error.
The file is correctly listed in the dialog.
[image]

shroffk, 2018-05-29:
Hi Michael, 
Almost there, you need to move the file to a location like below...
```C:\git\preferences\accl\configuration\diirt\datasources\beast\beast.xml```

so under your diirt folder there needs to be a datasource/beast folder with the above file

MichaelRitzert, 2018-05-31:
That's it, thanks!
Do you have any ideas, how I could package this file in a jar? Can I use the platform:/plugin/… syntax?

MichaelRitzert, 2018-06-02:
I tried to use a platform:/ location that happens to contain XML files that certainly are not valid for this use case, but I think they should still show up in the dialog. This gives me a null pointer exception after URIUtil.toFile:

2018-06-02 09:01:32.079 FINE [Thread 1] org.csstudio.diirt.util.core.preferences.DIIRTPreferences (resolvePlatformPath) - About to resolve platform path [path: platform:/plugin/org.csstudio.b2vxd.configuration.validators.pxdlimits/resources].
2018-06-02 09:01:32.080 FINE [Thread 1] org.csstudio.diirt.util.core.preferences.DIIRTPreferences (resolvePlatformPath) - Resolved URL
    before: platform:/plugin/org.csstudio.b2vxd.configuration.validators.pxdlimits/resources
     after: jar:file:/home/ritzert/css-belle2/plugins/org.csstudio.b2vxd.configuration.validators.pxdlimits_0.0.1.201711151110.jar!/resources/
2018-06-02 09:01:32.080 FINE [Thread 1] org.csstudio.diirt.util.core.preferences.DIIRTPreferences (resolvePlatformPath) - Escaped URL
    before: jar:file:/home/ritzert/css-belle2/plugins/org.csstudio.b2vxd.configuration.validators.pxdlimits_0.0.1.201711151110.jar!/resources/
     after: jar:file:/home/ritzert/css-belle2/plugins/org.csstudio.b2vxd.configuration.validators.pxdlimits_0.0.1.201711151110.jar%21/resources/
2018-06-02 09:01:32.080 FINE [Thread 1] org.csstudio.diirt.util.core.preferences.DIIRTPreferences (resolvePlatformPath) - Resolved URI
    before: jar:file:/home/ritzert/css-belle2/plugins/org.csstudio.b2vxd.configuration.validators.pxdlimits_0.0.1.201711151110.jar%21/resources/
     after: jar:file:/home/ritzert/css-belle2/plugins/org.csstudio.b2vxd.configuration.validators.pxdlimits_0.0.1.201711151110.jar%21/resources/
2018-06-02 09:01:32.080 WARNING [Thread 1] org.csstudio.diirt.util.core.preferences.DIIRTPreferences (resolveAndVerifyDIIRTPath) - Unable to resolve the given DIIRT path [platform:/plugin/org.csstudio.b2vxd.configuration.validators.pxdlimits/resources].
java.lang.NullPointerException
	at org.csstudio.diirt.util.core.preferences.DIIRTPreferences.resolvePlatformPath(DIIRTPreferences.java:198)

https://github.com/ControlSystemStudio/cs-studio/blob/master/core/diirt/diirt-plugins/org.csstudio.diirt.util.core.preferences/src/org/csstudio/diirt/util/core/preferences/DIIRTPreferences.java#L198

shroffk, 2018-06-05:
Why are you trying to package this file in a jar?
I package these with my product
https://github.com/cs-studio-nsls2/org.csstudio.nsls2.product/tree/master/features/org.csstudio.nsls2.product.feature/rootfiles

MichaelRitzert, 2018-06-08:
And then, how do you set the path? Use a relative path? I can't find the place in your https://github.com/cs-studio-nsls2/org.csstudio.nsls2.product/blob/master/features/org.csstudio.nsls2.product.feature/rootfiles/configuration/plugin_customization.ini .

If I can get it in a jar, I have a fixed name everywhere.

shroffk, 2018-08-02:
Ah!!
sorry, unfortunately the datasource configuration can't be accesses via eclipse resource paths.
in the plugin_customization.ini I put the location of the datasource configuration folder, and I usually package this folder with cs-studio itself. This way everyone gets it...not ideal but the most workable.