ControlSystemStudio/cs-studio #2422: EPICS Addr. List seems to be ignored by BEAST alarm-server

beenje, 2018-04-18:
I installed a BEAST alarm-server and configured the epics/addr_list to the IP of the channel access gateway required to access the PVs.
This is what I have in my plugin_customization.ini:
```
org.csstudio.platform.libs.epics/use_pure_java=true
org.csstudio.platform.libs.epics/support_dbe_property=true
org.csstudio.platform.libs.epics/monitor=VALUE
org.csstudio.platform.libs.epics/addr_list="172.16.14.24"
```

When I start the server, I can see in the logs it's taken into account:
```
Apr 17 17:08:01 alarm-02.tn.esss.lu.se systemd[1]: Started Alarm Server TS2.
Apr 17 17:08:01 alarm-02.tn.esss.lu.se systemd[1]: Starting Alarm Server TS2...
Apr 17 17:08:01 alarm-02.tn.esss.lu.se sh[8407]: 2018-04-17 17:08:01.791 INFO [Thread 1] org.csstudio.alarm.beast.server.Application (start) - Alarm Server 4.5.0.201711010050 started for 'TS2' configuration
Apr 17 17:08:01 alarm-02.tn.esss.lu.se sh[8407]: Alarm Server 4.5.0.201711010050
Apr 17 17:08:01 alarm-02.tn.esss.lu.se sh[8407]: Configuration Root: TS2
Apr 17 17:08:01 alarm-02.tn.esss.lu.se sh[8407]: Database URL:       jdbc:mysql://alarm-02.tn.esss.lu.se/alarm
Apr 17 17:08:01 alarm-02.tn.esss.lu.se sh[8407]: JMS URL:            failover:(tcp://alarm-02.tn.esss.lu.se:61616)
Apr 17 17:08:01 alarm-02.tn.esss.lu.se sh[8407]: JMS Server Topic:   TS2_SERVER
Apr 17 17:08:01 alarm-02.tn.esss.lu.se sh[8407]: JMS Client Topic:   TS2_CLIENT
Apr 17 17:08:01 alarm-02.tn.esss.lu.se sh[8407]: JMS Talk Topic:     TS2_TALK
Apr 17 17:08:01 alarm-02.tn.esss.lu.se sh[8407]: JMS Global Topic:   GLOBAL_SERVER
Apr 17 17:08:01 alarm-02.tn.esss.lu.se sh[8407]: EPICS Addr. List:   "172.16.14.24"
```

After a few seconds, I get some `pvConnectionTimeout` for all the PVs. See full log:
[alarm-server.log](https://github.com/ControlSystemStudio/cs-studio/files/1922869/alarm-server.log)

I did a tcpdump while starting the server and I only see requests to the local network broadcast address:
```
17:08:02.347009 IP 172.16.107.23.51719 > 172.16.107.255.ca-1: UDP, length 1016
17:08:02.347351 IP 172.16.107.23.51719 > 172.16.107.255.ca-1: UDP, length 992
17:08:02.347409 IP 172.16.107.23.51719 > 172.16.107.255.ca-1: UDP, length 72
```
[tcpdump.log](https://github.com/ControlSystemStudio/cs-studio/files/1922880/tcpdump.log)

There is no traffic going to the gateway IP (172.16.14.24)...

I tried to define `org.csstudio.platform.libs.epics/addr_list` both with "" and without.

Am I defining the `addr_list` incorrectly or is it not taken into account?

kasemir, 2018-04-18:
> This is what I have in my plugin_customization.ini:
```
org.csstudio.platform.libs.epics/use_pure_java=true
org.csstudio.platform.libs.epics/support_dbe_property=true
org.csstudio.platform.libs.epics/monitor=VALUE
org.csstudio.platform.libs.epics/addr_list="172.16.14.24" 
```

Those used to be the valid CA settings from the beginning of CSS for about a decade.
Then there was a time when 'diirt' wanted those settings to be in a separate file, and there was a preference setting to specify that file,
while vtype.pv used these:
```
org.csstudio.vtype.pv/use_pure_java=true
org.csstudio.vtype.pv/addr_list=127.0.0.1 some_host:5066
org.csstudio.vtype.pv/max_array_bytes=100000000
```

Right now we all use these preferences:
```
org.csstudio.diirt.util.core.preferences/diirt.ca.pure.java=true
org.csstudio.diirt.util.core.preferences/diirt.ca.addr.list=127.0.0.1 some_host:5066
org.csstudio.diirt.util.core.preferences/diirt.ca.auto.addr.list=true
org.csstudio.diirt.util.core.preferences/diirt.ca.max.array.size=100000000
org.csstudio.diirt.util.core.preferences/diirt.ca.beacon.period=15.0
org.csstudio.diirt.util.core.preferences/diirt.ca.connection.timeout=30.0
org.csstudio.diirt.util.core.preferences/diirt.ca.monitor.mask=VALUE
org.csstudio.diirt.util.core.preferences/diirt.ca.custom.mask=5
org.csstudio.diirt.util.core.preferences/diirt.ca.dbe.property.supported=false
org.csstudio.diirt.util.core.preferences/diirt.ca.honor.zero.precision=true
org.csstudio.diirt.util.core.preferences/diirt.ca.repeater.port=5065
org.csstudio.diirt.util.core.preferences/diirt.ca.server.port=5064
org.csstudio.diirt.util.core.preferences/diirt.ca.value.rtyp.monitor=false
org.csstudio.diirt.util.core.preferences/diirt.ca.variable.length.array=auto
```

I'm sure it will continue to improve.

beenje, 2018-04-18:
It works better with the proper preferences :-) Thanks!

I think this issue should be used to fix some things:
1. I took the settings from the GitHub repository: https://github.com/ControlSystemStudio/cs-studio/blob/master/applications/alarm/alarm-plugins/org.csstudio.alarm.beast.server/plugin_customization.ini
It'd be nice if it was updated with the current valid settings.
Same goes for the documentation http://cs-studio.sourceforge.net/docbook/ch14.html

2. In the log file, the "EPICS Addr. List:" displayed is linked to `org.csstudio.platform.libs.epics/addr_list` value but is not the one used. It's very confusing. The value used should be displayed.

kasemir, 2018-04-18:
The examples have been updated, the misleading log message has been removed.

PR #2424 (2018-04-19): Alarm server: Update example settings for CA
Fixes #2422
closed 2018-04-19 by berryma4