ControlSystemStudio/cs-studio #2423: Alarm Server: Failed to initialize PV type 'pva' from org.csstudio.vtype.pv

beenje, 2018-04-18:
At ESS, we compiled CS-Studio and the alarm-server ourselves.
When starting the alarm-server, we get the following exceptions:
```
Apr 18 09:47:55 alarm-02.tn.esss.lu.se sh[9212]: 2018-04-18 09:47:55.399 SEVERE [Thread 1] org.csstudio.vtype.pv.PVPlugin (start) - Failed to initialize PV type 'pva' from org.csstudio.vtype.pv
Apr 18 09:47:55 alarm-02.tn.esss.lu.se sh[9212]: org.eclipse.core.runtime.CoreException: Plug-in "org.csstudio.vtype.pv" was unable to instantiate class "org.csstudio.vtype.pv.pva.PVA_PVFactory".
Apr 18 09:47:55 alarm-02.tn.esss.lu.se sh[9212]: at org.eclipse.core.internal.registry.osgi.RegistryStrategyOSGI.throwException(RegistryStrategyOSGI.java:194)
...
Apr 18 09:47:55 alarm-02.tn.esss.lu.se sh[9212]: Caused by: java.lang.ClassNotFoundException: org.epics.pvaccess.client.ChannelRequester cannot be found by org.csstudio.vtype.pv_4.5.0.201803211120
...
```
Full log: 
[alarm-server-exceptions.log](https://github.com/ControlSystemStudio/cs-studio/files/1922986/alarm-server-exceptions.log)

It tried a build from SNS: https://ics-web.sns.ornl.gov/css/nightly/
And no exception is raised.

So I guess that something is missing in our build pipeline.

Has someone any idea where we should start looking?

kasemir, 2018-04-18:
For what it's worth, the vtype.pv supports pva (and ca, and mqtt) but that's only been tried with displays.
In displays, using pva://.., I do see the value, alarm border, units, precision of the PV reflected when using PVA, so I assume that the alarm server would also get the alarm severity via pva://.., but I've never tried that. Would be good for you to test and let me know what else might be missing.

As for getting the PVA PVFactory to find the org.epics.pvaccess classes, check the plugin dependencies listed in your *.project file.
The SNS alarm server ends up with plugins/org.epics.jca_2.4.1.jar (the updated JCA bundle which now includes CAJ), org.epics.pvdata_6.1.1.jar and org.epics.pvaccess_5.1.1.jar (the latest PVA code).

Earlier, the alarm server product would have this in the plugins/ dir:
org.csstudio.platform.libs.epics_4.3.0 (which included the jca and caj jar),
org.epics.pvdata_4.0.3.jar and org.epics.pvaccess_4.0.4.jar.

Does your alarm server product have any plugins/*pvdata* or plugins/*pvaccess* jar?

claudio-rosati, 2018-04-19:
Thank you Kay.
org.epics.pvdata and org.epics.pvaccess dependencies were missing. I've added them and just waiting for the build running.

beenje, 2018-04-27:
Tested OK with the new ESS release 4.6.1.0 pushed by Claudio:

```
Apr 27 14:47:44 alarm-02.tn.esss.lu.se systemd[1]: Starting Alarm Server TS2...
Apr 27 14:47:45 alarm-02.tn.esss.lu.se sh[31926]: 2018-04-27 14:47:45.789 INFO [Thread 1] org.csstudio.alarm.beast.server.Application (start) - Alarm Server 4.5.0.201804270944 started for 'TS2' configuration
Apr 27 14:47:45 alarm-02.tn.esss.lu.se sh[31926]: Alarm Server 4.5.0.201804270944
Apr 27 14:47:45 alarm-02.tn.esss.lu.se sh[31926]: Configuration Root: TS2
Apr 27 14:47:45 alarm-02.tn.esss.lu.se sh[31926]: Database URL:       jdbc:mysql://alarm-02.tn.esss.lu.se/alarm
Apr 27 14:47:45 alarm-02.tn.esss.lu.se sh[31926]: JMS URL:            failover:(tcp://alarm-02.tn.esss.lu.se:61616)
Apr 27 14:47:45 alarm-02.tn.esss.lu.se sh[31926]: JMS Server Topic:   TS2_SERVER
Apr 27 14:47:45 alarm-02.tn.esss.lu.se sh[31926]: JMS Client Topic:   TS2_CLIENT
Apr 27 14:47:45 alarm-02.tn.esss.lu.se sh[31926]: JMS Talk Topic:     TS2_TALK
Apr 27 14:47:45 alarm-02.tn.esss.lu.se sh[31926]: JMS Global Topic:   GLOBAL_SERVER
Apr 27 14:47:45 alarm-02.tn.esss.lu.se sh[31926]: 2018-04-27 14:47:45.809 INFO [Thread 20] org.csstudio.alarm.beast.server.SeverityPVHandler (run) - Not using any heartbeat PV
Apr 27 14:47:45 alarm-02.tn.esss.lu.se sh[31926]: 2018-04-27 14:47:45.812 CONFIG [Thread 1] org.csstudio.vtype.pv.PVPlugin (start) - PV prefix pva provided by pva in org.csstudio.vtype.pv
Apr 27 14:47:45 alarm-02.tn.esss.lu.se sh[31926]: 2018-04-27 14:47:45.815 CONFIG [Thread 1] org.csstudio.vtype.pv.PVPlugin (start) - PV prefix ca provided by ca in org.csstudio.vtype.pv
Apr 27 14:47:45 alarm-02.tn.esss.lu.se sh[31926]: 2018-04-27 14:47:45.815 CONFIG [Thread 1] org.csstudio.vtype.pv.PVPlugin (start) - PV prefix ca provided by epics in org.csstudio.vtype.pv
Apr 27 14:47:45 alarm-02.tn.esss.lu.se sh[31926]: 2018-04-27 14:47:45.816 CONFIG [Thread 1] org.csstudio.vtype.pv.PVPlugin (start) - PV prefix loc provided by loc in org.csstudio.vtype.pv
Apr 27 14:47:45 alarm-02.tn.esss.lu.se sh[31926]: 2018-04-27 14:47:45.817 CONFIG [Thread 1] org.csstudio.vtype.pv.PVPlugin (start) - PV prefix sim provided by sim in org.csstudio.vtype.pv
Apr 27 14:47:45 alarm-02.tn.esss.lu.se sh[31926]: 2018-04-27 14:47:45.817 CONFIG [Thread 1] org.csstudio.vtype.pv.PVPlugin (start) - PV prefix mqtt provided by mqtt in org.csstudio.vtype.pv
Apr 27 14:47:45 alarm-02.tn.esss.lu.se sh[31926]: osgi> 2018-04-27 14:47:45.819 CONFIG [Thread 1] org.csstudio.vtype.pv.PVPlugin (start) - Default PV type ca
```

closed 2018-04-27 by beenje