ControlSystemStudio/cs-studio #2429: ChannelViewer: can't tell if search has been completed

mark0n, 2018-05-10:
When searching with ChannelViewer it is hard (impossible?) to tell if it is still busy searching or if the result is empty. In both cases the list is empty. Please make it obvious to the user if it is busy or not. Some ideas:

- show "no results" if ChannelFinder returned an empty set of PVs
- use the progress bar/busy widget in the lower right corner of the window to indicate that the search is in progress

shroffk, 2018-06-05:
This is an annoying problem which is due to the fact that swt/eclipse only supports the spinning wheel indicator with the use of Runnables and not Callables...
With the new phoebus application, we will ensure that either the progress bar or the Wait Cursor will convey this information.

shroffk, 2018-06-08:
[image]