ControlSystemStudio/cs-studio #2430: Allow BOY Action Button to open Display Builder file

mark0n, 2018-05-10:
It seems like it is not possible to link from an Action Button in BOY to a Display Builder screen (open in run mode). I tried an "Open OPI" action but that only allows me to select *.opi files. I also tried "Open File" but that opened the *.bob file in edit mode...
Note: All my files are in my workspace (not pulled from web server).

I'm running the FRIB version of CS-Studio 4.5.6 on Debian jessie.

kasemir, 2018-05-10:
The suggested upgrade path from BOY to Display Builder at this time is to open the existing screens in the display builder runtime. Most of them should work OK. For those that don't, maybe it's possible to update the display builder to better handle that specific widget/property when loading legacy *.opi files. There will be cases involving scripts or plots that do require opening the *.opi in the display builder editor, adjusting it to the new set of widgets/properties, then save as a *.bob file.
Once all displays "work", switch the beam line from BOY to the Display Builder.

The mode of running both BOY and Display Builder in parallel within the same instance of CSS, where action buttons clicked within one runtime opens the next display in the _other_ runtime is currently not directly supported. If you really do need this, you could use a script. I could provide an example for that next week.

mark0n, 2018-05-10:
Converting all FRIB OPIs to Display Builder at once is a big effort and will potentially delay the switch to Display Builder. Cross-linking between different groups makes this even more difficult (for example our LLRF interlock screen has a "PLC interlock" LED with an action button linking to the PLC group's details screen that shows which of the various inputs caused it to trip off).
Also, some system owners might feel more comfortable with switching to Display Builder while others tend to be more conservative...

We might be able to keep the number of links into the other run-time low. So a script might actually be feasible.

kasemir, 2018-05-11:
Try this:

```
# Action for BOY *.opi Action Button
# that opens a display in the Display Builder runtime

# Filename to open, path within workspace
file = "/Display Builder/01_main.bob"
# Macros (or None)
macros = "<NAME>value1</NAME><OTHER>another value</OTHER>"
        
from org.csstudio.openfile import DisplayUtil
DisplayUtil.getInstance().openDisplay(file, macros)
```

mark0n, 2018-05-11:
Thanks, opening the screen works as expected. However, when I try to shove all macros from BOY into Display Builder I'm running into trouble due to different string formats: `display.getPropertyValues('macros').toString()` returns "{Parent Macros} {LCID=LCID_10361, FOO1=BAR1, FOO2=BAR2}" whereas the method you mentioned expects an XML string. Is there a simple way to convert to Display Builder format (short of converting the strings with plain Python)?

kasemir, 2018-05-14:
When writing scripts, you can call pretty much any of the Java code, but that of course means you need to look at the Java code to figure out what's there.
```
print(display.__class__.__name__)
```
shows that it's an `DisplayEditpart`, and drilling down into its source shows that the macros are of type `MacrosInput`. This code would dump all the macros and their value:

```
map = display.getPropertyValue('macros').getMacrosMap()
for name in map:
    print("%s = %s" % (name, map.get(name)))
```

This could be used in most cases to convert the BOY macros of a display or a widget that happens to have a 'macros' property to the string used in `DisplayUtil.getInstance().openDisplay(file, macros)`:

```
map = display.getPropertyValue('macros').getMacrosMap()
display_builder_macro_setting="".join(
   [ "<%s>%s</%s>" % (name, map.get(name), name) for name in map ]
   )
```

I say most cases because it doesn't escape '<' etc. inside the value. If the value of the macro contained something that looks like XML, it can fail. But in practice, it's probably good enough.

mark0n, 2018-05-14:
That's good stuff :-)

As a summary, here is the script I'm using to open a Display Builder screen (`myfile.bob` from an BOY Action Button:
```
file = "/myproject/mydirectory/myfile.bob"

map = display.getPropertyValue('macros').getMacrosMap()
displayBuilderMacros="".join(
  [ "<%s>%s</%s>" % (name, map.get(name), name) for name in map ]
)

from org.csstudio.openfile import DisplayUtil
DisplayUtil.getInstance().openDisplay(file, displayBuilderMacros)
```
Thank you so much!

closed 2018-05-14 by mark0n