ControlSystemStudio/cs-studio #2437: Phoebus Integration

shroffk, 2018-06-05:
Create a plugin which provides a service which allows for the launching of cs-studio/phoebus.

shroffk, 2018-06-05:
@kasemir 
I have two options when it comes to including launch actions for various phoebus applications from cs-studio.

Option 1.
Add the launch action in the associated application plugin itself...i.e in the o.c.diag.probe
This would add a dependency to the o.c.phoebus.integration plugin and would need additional preferences to allow users to enable/disable including phoebus actions.

Option 2.
Create a new stand alone plugin which registers the actions. 
This would mean that the app launch action would not be in the same plugin with the old cs-studio applications...but it would be easy for people to control weather to include these actions in their product.

Which option should I add...I am leaning towards Option2. since it would be the lease disruptive although not the nicest way.

kasemir, 2018-06-05:
Yes, 2, for the reasons you mention.
That way, you can decide if you want to include 'probe' or 'launch-probe-in-phoebus', 'PV Tree' or 'launch-PV-Tree-in-phoebus', 'XXX' or 'launch-XXX-in-phoebus' in your site-specific product.
I also see value in not having to change the existing XXX code at all for this purpose.

shroffk, 2018-10-31:
Integrate (menu actions)
 - [ ] probe
 - [ ] databrowser
 - [ ] displaybuilder
 - [ ] alarm applications
 - [ ] channel applications
 - [ ] logbook applications

Integrate (context menu actions)
 - [ ] probe
 - [ ] databrowser
 - [ ] displaybuilder
 - [ ] alarm applications
 - [ ] channel applications
 - [ ] logbook applications