ControlSystemStudio/cs-studio #2445: Using Eclipse Photon

claudio-rosati, 2018-07-18:
@kasemir @shroffk @willrogers @berryma4 

Here my initial report of using Eclipse Photon with CS-Studio

I've Installed Eclipse Photon, starting a new workspace. I've selected Java 8 instead of Java 10 in the Installed JREs preferences, and modified accordingly the Compiler preferences to 1.8.

The Target Platform I've created (following Ch. 4 of Control System Studio Guide) is plenty of errors because some plugins (Eclipse) are not found.

The Run Configuration has to be fixed because some plugins have many versions, and the good one has to be selected. I'm unable to solve some missing constraints for `org.csstudio.rap.core` and `org.csstudio.rap.ui.util`.

Moreover, the default Run Configuration has the `--add-modules ALL-MODULE-PATH` argument and must be removed.

The application starts with some exceptions (mostly related to the missing rap constraints), but it is a good starting point I think.

kasemir, 2018-07-18:
I tend to delete all *rap* plugins from my workspace. Some *rap* plugins implement the same classes as corresponding *rcp* plugins (ex. org.csstudio.opibuilder.rap, org.csstudio.opibuilder.rcp). You can't have both, must select either rcp or rap. But when the IDE auto-creates a new launch configuration, it tends to include both. Easier to get started if you remove all *rap* plugins from the workspace.

claudio-rosati, 2018-07-19:
I've now imported only my project (ESS product) and the Display Builder one.
No more warning... but the following error when I try to open a .bob file

```
!ENTRY org.eclipse.e4.ui.workbench 4 0 2018-07-19 16:39:21.198
!MESSAGE Unable to create class 'org.eclipse.ui.internal.e4.compatibility.CompatibilityEditor' from bundle '532'
!STACK 0
org.eclipse.e4.core.di.InjectionException: java.lang.NoClassDefFoundError: javafx/embed/swt/FXCanvas
	at org.eclipse.e4.core.internal.di.InjectorImpl.internalMake(InjectorImpl.java:423)
	at org.eclipse.e4.core.internal.di.InjectorImpl.make(InjectorImpl.java:331)
	at org.eclipse.e4.core.contexts.ContextInjectionFactory.make(ContextInjectionFactory.java:190)
	at org.eclipse.e4.ui.internal.workbench.ReflectionContributionFactory.createFromBundle(ReflectionContributionFactory.java:105)
	at org.eclipse.e4.ui.internal.workbench.ReflectionContributionFactory.doCreate(ReflectionContributionFactory.java:74)
	at org.eclipse.e4.ui.internal.workbench.ReflectionContributionFactory.create(ReflectionContributionFactory.java:56)
	at org.eclipse.e4.ui.workbench.renderers.swt.ContributedPartRenderer.createWidget(ContributedPartRenderer.java:129)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.createWidget(PartRenderingEngine.java:1012)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeCreateGui(PartRenderingEngine.java:672)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeCreateGui(PartRenderingEngine.java:778)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.access$0(PartRenderingEngine.java:749)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$2.run(PartRenderingEngine.java:743)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:42)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.createGui(PartRenderingEngine.java:727)
	at org.eclipse.e4.ui.internal.workbench.PartServiceImpl$1.handleEvent(PartServiceImpl.java:104)
	at org.eclipse.e4.ui.services.internal.events.UIEventHandler$1.run(UIEventHandler.java:40)
	at org.eclipse.swt.widgets.Synchronizer.syncExec(Synchronizer.java:233)
	at org.eclipse.ui.internal.UISynchronizer.syncExec(UISynchronizer.java:144)
	at org.eclipse.swt.widgets.Display.syncExec(Display.java:4860)
	at org.eclipse.e4.ui.internal.workbench.swt.E4Application$1.syncExec(E4Application.java:212)
	at org.eclipse.e4.ui.services.internal.events.UIEventHandler.handleEvent(UIEventHandler.java:36)
	at org.eclipse.equinox.internal.event.EventHandlerWrapper.handleEvent(EventHandlerWrapper.java:201)
	at org.eclipse.equinox.internal.event.EventHandlerTracker.dispatchEvent(EventHandlerTracker.java:196)
	at org.eclipse.equinox.internal.event.EventHandlerTracker.dispatchEvent(EventHandlerTracker.java:1)
	at org.eclipse.osgi.framework.eventmgr.EventManager.dispatchEvent(EventManager.java:230)
	at org.eclipse.osgi.framework.eventmgr.ListenerQueue.dispatchEventSynchronous(ListenerQueue.java:148)
	at org.eclipse.equinox.internal.event.EventAdminImpl.dispatchEvent(EventAdminImpl.java:135)
	at org.eclipse.equinox.internal.event.EventAdminImpl.sendEvent(EventAdminImpl.java:78)
	at org.eclipse.equinox.internal.event.EventComponent.sendEvent(EventComponent.java:39)
	at org.eclipse.e4.ui.services.internal.events.EventBroker.send(EventBroker.java:52)
	at org.eclipse.e4.ui.internal.workbench.UIEventPublisher.notifyChanged(UIEventPublisher.java:60)
	at org.eclipse.emf.common.notify.impl.BasicNotifierImpl.eNotify(BasicNotifierImpl.java:374)
	at org.eclipse.e4.ui.model.application.ui.impl.ElementContainerImpl.setSelectedElement(ElementContainerImpl.java:174)
	at org.eclipse.e4.ui.internal.workbench.ModelServiceImpl.showElementInWindow(ModelServiceImpl.java:634)
	at org.eclipse.e4.ui.internal.workbench.ModelServiceImpl.bringToTop(ModelServiceImpl.java:598)
	at org.eclipse.e4.ui.internal.workbench.PartServiceImpl.delegateBringToTop(PartServiceImpl.java:788)
	at org.eclipse.e4.ui.internal.workbench.PartServiceImpl.bringToTop(PartServiceImpl.java:401)
	at org.eclipse.e4.ui.internal.workbench.PartServiceImpl.showPart(PartServiceImpl.java:1238)
	at org.eclipse.ui.internal.WorkbenchPage.busyOpenEditor(WorkbenchPage.java:3277)
	at org.eclipse.ui.internal.WorkbenchPage.access$26(WorkbenchPage.java:3192)
	at org.eclipse.ui.internal.WorkbenchPage$10.run(WorkbenchPage.java:3174)
	at org.eclipse.swt.custom.BusyIndicator.showWhile(BusyIndicator.java:71)
	at org.eclipse.ui.internal.WorkbenchPage.openEditor(WorkbenchPage.java:3169)
	at org.eclipse.ui.internal.WorkbenchPage.openEditor(WorkbenchPage.java:3133)
	at org.eclipse.ui.internal.WorkbenchPage.openEditor(WorkbenchPage.java:3123)
	at org.eclipse.ui.ide.IDE.openEditor(IDE.java:563)
	at org.eclipse.ui.ide.IDE.openEditor(IDE.java:518)
	at org.eclipse.ui.actions.OpenFileAction.openFile(OpenFileAction.java:100)
	at org.eclipse.ui.actions.OpenSystemEditorAction.run(OpenSystemEditorAction.java:97)
	at org.eclipse.ui.views.navigator.OpenActionGroup.runDefaultAction(OpenActionGroup.java:137)
	at org.eclipse.ui.views.navigator.MainActionGroup.runDefaultAction(MainActionGroup.java:325)
	at org.eclipse.ui.views.navigator.ResourceNavigator.handleOpen(ResourceNavigator.java:761)
	at org.eclipse.ui.views.navigator.ResourceNavigator$2.open(ResourceNavigator.java:477)
	at org.eclipse.ui.OpenAndLinkWithEditorHelper$InternalListener.open(OpenAndLinkWithEditorHelper.java:46)
	at org.eclipse.jface.viewers.StructuredViewer$2.run(StructuredViewer.java:850)
[... cut ...]
Caused by: java.lang.ClassNotFoundException: javafx.embed.swt.FXCanvas cannot be found by org.eclipse.fx.ui.workbench3_2.4.0.201605100504
	at org.eclipse.osgi.internal.loader.BundleLoader.findClassInternal(BundleLoader.java:508)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:419)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:411)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.loadClass(ModuleClassLoader.java:150)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 109 more
```

claudio-rosati, 2018-07-19:
Again, "frustrating" doesn't express completely my feeling!

willrogers, 2018-07-19:
I also cannot find JavaFX libraries. I am not sure what the current state of master is - is it aimed at Java 8 or Java 9+?

shroffk, 2018-07-19:
What version of the efxclipse is being used? is it loaded successfully?

kasemir, 2018-07-19:
All cs-studio 'masters' should be Java 8. All phoebus 'masters' are java 9+.

willrogers, 2018-07-19:
For me, it seems to work with maven but not when running from Eclipse.

claudio-rosati, 2018-07-20:
The same is for me.

claudio-rosati, 2018-11-06:
Eventually I was successful to use Photon IDE (2018-09) to build CS-Studio with JDK 8.

I had only a small issue with the new JDK 8 (1.0.8_191, tested it only on my MacOS X): by default when the installed Java SE 8 is detected (or updated, being my workspace comes from the Oxygen version of the IDE) it doesn't contain the `<path-to-jvms>/jdk1.8.0_191.jdk/Contents/Home/jre/lib/jfxswt.jar` JAR file, and must be added manually, otherwise FXCanvas is not resolved.

Probably my [previous comment](https://github.com/ControlSystemStudio/cs-studio/issues/2445#issuecomment-406300374) was related to the same problem.

So, from my point of view, now I can use Photon to build the CS-Studio based on JDK 8, and this issue can be closed.

shroffk, 2020-07-23:
This has been completed.

closed 2020-07-23 by shroffk