ControlSystemStudio/cs-studio #2446: BOY Combo widget not updating in windows

berryma4, 2018-07-19:
I'm looking to confirm a bug on 4.5.x, pvmanager as datasource.
In windows:
1. create an empty opi
2. add a combo widget
3. add items; as, asd, asdf
4. add pvname: loc://test
5. open probe: loc://test
6. run opi, and select as, or asd, or asdf 
7. notice the pv does not update, or the display on the widget

In linux, it works fine. Although I noticed if one item is a number, the pv still updates, but doesn't display on the widget. changing the pv name to: loc://test<VString> fixes the issue. This must be a new feature, as I have screens that "work" with numbers in 4.4.x.

kasemir, 2018-07-19:
What if  you define the PV as `loc://test("as")` or an enum?

In your example, it's somewhat unclear if the PV is meant to be a number, so the combo should map enum items to a number, or if the PV is meant to be a string (which it isn't).

kasemir, 2018-07-19:
When I use `loc://test("as")` to assert that we're dealing with a string PV, it works fine with BOY/vtype.PV as well as Display Builder.

kasemir, 2018-07-19:
Same with `loc://test<VEnum>(0, "as", "asd", "asdf")` where the PV is an enum. Can then also use "Items from PV"

berryma4, 2018-07-19:
So, updating:
I'm looking to confirm a bug on 4.5.x, pvmanager as datasource.
In windows:
1. create an empty opi
2. add a combo widget
3. add items; as, asd, asdf
4. add pvname: loc://test or loc://test("as")
5. open probe: loc://test
6. run opi, and select as, or asd, or asdf 
7. notice the pv does not update (stays as default "as"), or the display on the widget

In linux, it works fine. Although I noticed if one item is a number, the pv still updates, but doesn't display on the widget. changing the pv name to: loc://test<VString> fixes the issue. This must be a new feature, as I have screens that "work" with numbers in 4.4.x.

hulei, 2018-07-20:
CSS Version: cs-studio-4.5.7-win32.win32.x86_64.zip
pvname:  loc://test  or  loc://test("as")
items are :
[image]
test result:
strings updates OK!  
when I choose the number 9, dispeared in 2 seconds. 
And old version :basic-epics-4.1.1 winodws 64bit , the same thing.