ControlSystemStudio/cs-studio #2448: SVG images not displayed in BOY

claudio-rosati, 2018-07-25:
Hello all,
I need help on this issue that, I've verified, appeared only since we moved from Neon to Oxygen (It was reported to me only now because BOY is not used everyday).

When opening am OPI containing an Image widget loading an SVG file, it is not displayed, and the following is on the log file:

```
!ENTRY org.eclipse.ui 4 0 2018-07-25 12:11:51.189
!MESSAGE Unhandled event loop exception
!STACK 0
java.lang.LinkageError: loader constraint violation: when resolving method "org.apache.batik.bridge.BridgeContext.getCSSEngineForElement(Lorg/w3c/dom/Element;)Lorg/apache/batik/css/engine/CSSEngine;" the class loader (instance of org/eclipse/osgi/internal/loader/EquinoxClassLoader) of the current class, org/csstudio/utility/batik/SVGHandler, and the class loader (instance of org/eclipse/osgi/internal/loader/EquinoxClassLoader) for the method's defining class, org/apache/batik/bridge/BridgeContext, have different Class objects for the type org/apache/batik/css/engine/CSSEngine used in the signature
	at org.csstudio.utility.batik.SVGHandler.buildElementsToUpdateList(SVGHandler.java:786)
	at org.csstudio.utility.batik.SVGHandler.<init>(SVGHandler.java:188)
	at org.csstudio.swt.widgets.symbol.SVGSymbolImage.loadDocument(SVGSymbolImage.java:313)
	at org.csstudio.swt.widgets.symbol.SVGSymbolImage.access$2(SVGSymbolImage.java:300)
	at org.csstudio.swt.widgets.symbol.SVGSymbolImage$2.runWithInputStream(SVGSymbolImage.java:280)
	at org.csstudio.swt.widgets.util.AbstractInputStreamRunnable.run(AbstractInputStreamRunnable.java:32)
	at org.eclipse.swt.widgets.RunnableLock.run(RunnableLock.java:37)
	at org.eclipse.swt.widgets.Synchronizer.runAsyncMessages(Synchronizer.java:182)
	at org.eclipse.swt.widgets.Display.runAsyncMessages(Display.java:4034)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3701)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1150)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1039)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:153)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:680)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:594)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:148)
	at org.csstudio.utility.product.Workbench.runWorkbench(Workbench.java:99)
	at org.csstudio.startup.application.Application.startApplication(Application.java:265)
	at org.csstudio.startup.application.Application.start(Application.java:119)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:196)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:388)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:243)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:653)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:590)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1499)
```

Looking at the application's plugins I've discovered the following:

[image]

As you can see 3 batik plugins exist in both version 1.7 and 1.8: the 1.8 come from Oxygen.

Manually removing  the 1.8 plugins makes SVGs being displayed again, without any errors in the log files.

Have you an idea on how to get rid of the problem? Or, alternatively, how to avoid Tycho packaging the 1.8 files?

claudio-rosati, 2018-07-25:
Temporarily I'll go to have a post-build process that will remove the 1.8 files from the distribution tar/zip files.

kasemir, 2018-07-25:
I don't use SVGs, and have no idea about Batik, but as for general class loader errors..

Do you know which plugin contains org.apache.batik.anim.dom.SAXSVGDocumentFactory, and which plugin contains org.apache.batik.css.engine.CSSEngine?
I assume what's happening is this: SAXSVGDocumentFactory is in plugin A, and CSSEngine is in plugin B. You do have both plugin A and B in your product. They are tycho-generated wrappers of the original batik jar files. Unpack the plugins, and check their MANIFEST.MF. Are they basically empty?
What you need is either for A/MANIFEST.MF to include "Require-Bundle: B", or some buddy class loading instructions. Otherwise the code running inside plugin A (SAXSVGDocumentFactory) won't be able to locate the code in plugin B (CSSEngine).

claudio-rosati, 2018-07-25:
What I think is happening is the following: Eclipse RCP (I don't know which plugin) start earlier registering an engine via its own version of batik (1.8 for Oxygen, 1.9.1 for Photon). Then when BOY needs to display SVGs through a factory class tries to obtain the engine, but at that point a class version mismatch is found, or the expected engine cannot be found and it fails.

kasemir, 2018-07-25:
Not sure it's that complicated. The error message is quite straight forward: 
```
java.lang.ClassNotFoundException:
org.apache.batik.css.engine.CSSEngine
cannot be found by
org.apache.xmlgraphics.batik-anim_1.10.0
```

The batik-anim plugin (1.10.0) cannot find CSSEngine. To find it, you need two things:
1) Some plugin XX in your product that contains the CSSEngine class.
2) A way for the batik-anim plugin to get the XX plugin into its class loader. For example, check the batik-anim plugin MANIFEST.MF for "Require-Bundle: XX"

You wrote earlier how you replaced pom content that included instructions:
```
<artifact>
  <id>org.codehaus.cargo:empty-jar:1.4.13</id>
  <instructions>
    ...<Require-Bundle>org.apache.batik.ext.awt;bundle-version="1.7.0"</Require-Bundle>
```
with simple `<artifact>` wrappers. The resulting plugins might lack the necessary "Require-Bundle" entries, so check the content of the generated plugins.

claudio-rosati, 2018-07-25:
I'll try tomorrow to play with the org.reficio Maven plugin inside `maven-osgi-bundles` to add re requirement.
Thank you

claudio-rosati, 2018-07-26:
I'm sorry to say that it is that complicated.

First I've modified the artifact in the `maven-osgi-bundels` pom file:

```
<artifact>
  <id>org.apache.xmlgraphics:batik-anim:${batik.version}</id>
<instructions>
<Require-Bundle>org.apache.xmlgraphics.batik-css;bundle-version="${batik.version}"</Require-Bundle>
</instructions>
</artifact>
```

and mow its manifest sounds good:

```
Manifest-Version: 1.0
Bnd-LastModified: 1532588623024
Bundle-ManifestVersion: 2
Bundle-Name: org.apache.xmlgraphics.batik-anim
Bundle-SymbolicName: org.apache.xmlgraphics.batik-anim
Bundle-Version: 1.10.0
Created-By: 1.8.0_172 (Oracle Corporation)
Export-Package: org.apache.batik.anim;uses:="org.apache.batik.anim.dom
 ,org.apache.batik.anim.timing,org.apache.batik.anim.values,org.apache
 .batik.ext.awt.geom,org.apache.batik.util,org.w3c.dom",org.apache.bat
 ik.anim.dom;uses:="org.apache.batik.anim.values,org.apache.batik.css.
 dom,org.apache.batik.css.engine,org.apache.batik.css.engine.value,org
 .apache.batik.css.parser,org.apache.batik.dom,org.apache.batik.dom.ev
 ents,org.apache.batik.dom.svg,org.apache.batik.dom.util,org.apache.ba
 tik.dom.xbl,org.apache.batik.i18n,org.apache.batik.parser,org.apache.
 batik.util,org.w3c.dom,org.w3c.dom.css,org.w3c.dom.events,org.w3c.dom
 .stylesheets,org.w3c.dom.svg,org.w3c.dom.views,org.xml.sax",org.apach
 e.batik.anim.dom.resources,org.apache.batik.anim.resources,org.apache
 .batik.anim.timing;uses:="org.apache.batik.anim,org.apache.batik.i18n
 ,org.apache.batik.parser,org.apache.batik.util,org.w3c.dom,org.w3c.do
 m.events",org.apache.batik.anim.values;uses:="org.apache.batik.anim.d
 om,org.apache.batik.dom.svg"
Import-Package: org.apache.batik.anim;resolution:=optional,org.apache.
 batik.anim.dom;resolution:=optional,org.apache.batik.anim.timing;reso
 lution:=optional,org.apache.batik.anim.values;resolution:=optional,or
 g.apache.batik.css.dom;resolution:=optional,org.apache.batik.css.engi
 ne;resolution:=optional,org.apache.batik.css.engine.value;resolution:
 =optional,org.apache.batik.css.engine.value.svg;resolution:=optional,
 org.apache.batik.css.parser;resolution:=optional,org.apache.batik.dom
 ;resolution:=optional,org.apache.batik.dom.events;resolution:=optiona
 l,org.apache.batik.dom.svg;resolution:=optional,org.apache.batik.dom.
 svg12;resolution:=optional,org.apache.batik.dom.util;resolution:=opti
 onal,org.apache.batik.dom.xbl;resolution:=optional,org.apache.batik.e
 xt.awt.geom;resolution:=optional,org.apache.batik.i18n;resolution:=op
 tional,org.apache.batik.parser;resolution:=optional,org.apache.batik.
 util;resolution:=optional,org.apache.batik.w3c.dom.events;resolution:
 =optional,org.w3c.css.sac;resolution:=optional,org.w3c.dom;resolution
 :=optional,org.w3c.dom.css;resolution:=optional,org.w3c.dom.events;re
 solution:=optional,org.w3c.dom.smil;resolution:=optional,org.w3c.dom.
 stylesheets;resolution:=optional,org.w3c.dom.svg;resolution:=optional
 ,org.w3c.dom.views;resolution:=optional,org.xml.sax;resolution:=optio
 nal
Require-Bundle: org.apache.xmlgraphics.batik-css;bundle-version="1.10"
Require-Capability: osgi.ee;filter:="(&(osgi.ee=JavaSE)(version=1.6))"
Tool: Bnd-3.3.0.201609221906
```

The first problem I've now is that when my product is complete, the `org.apache.xmlgraphics.batik-anim_1.10.0.jar` on its plugins folder is the one from Maven Central, not from `maven-osgi-bundels`, so it doesn't have the Require-Bundle line.

So I copied the right one and started the application. Still not working with the same problem:

```
!ENTRY org.eclipse.ui.workbench 4 2 2018-07-26 10:10:26.137
!MESSAGE Problems occurred when invoking code from plug-in: "org.eclipse.ui.workbench".
!STACK 0
java.lang.NoClassDefFoundError: org/apache/batik/css/engine/CSSEngine
	at org.apache.batik.anim.dom.SAXSVGDocumentFactory.<init>(SAXSVGDocumentFactory.java:111)
	at org.csstudio.utility.batik.SVGUtils.loadSVG(SVGUtils.java:30)
	at org.csstudio.opibuilder.tools.thumbnails.ImageIconDecorator.decorateImage(ImageIconDecorator.java:53)
	at org.eclipse.ui.internal.decorators.FullDecoratorDefinition.decorateImage(FullDecoratorDefinition.java:113)
	...
Caused by: java.lang.ClassNotFoundException: org.apache.batik.css.engine.CSSEngine cannot be found by org.apache.xmlgraphics.batik-anim_1.10.0
	at org.eclipse.osgi.internal.loader.BundleLoader.findClassInternal(BundleLoader.java:484)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:395)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:387)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.loadClass(ModuleClassLoader.java:150)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	at org.apache.batik.anim.dom.SAXSVGDocumentFactory.<init>(SAXSVGDocumentFactory.java:111)
	at org.csstudio.utility.batik.SVGUtils.loadSVG(SVGUtils.java:30)
	... 143 more
```

I think that, nevertheless batik changed the packaging name (`org.apache.xmlgraphics.batik-xxx` instead of `org.apache.batik.xxx`) the class and package names are the same. So the workbench loads at the very beginning the ver. 1.8 of the engine. Then, when the ver. 1.10 of the library is used by `org.csstudio.utility.batik.SVGUtils.loadSVG()` it will not found the right class because another version was already loaded. I also think that the error message is a bit misleading.

Anyway, the error is still here!

kasemir, 2018-07-26:
The plugin that you created with "<artifact>..<instructions><Require-Bundle>.." looks good, but then you noticed that you get another plugin packaged into the product.

> So I copied the right one and started the application.

Can you confirm that this worked? Simply copying a plugin.jar file might not have any effect because P2 wants to control the installation of plugins. You may be able to delete the unwanted `plugins/*.jar` file and then add the desired one into the `dropins/` folder. In any case, at runtime, the menu Help, About, Installation Details could be used to verify which plugin you really have in the product.

Finally, you're right: If code in the batik-anim plugin cannot find the CSSEngine via its own classloader, but there is a way to inject it from your code that has access to both batik-anim and the CSSEngine, that should work.

claudio-rosati, 2018-07-26:
Having to deploy a solution, the simple one was replacing the following lin in my deployment script
```
(cd org.csstudio.ess.product; time mvn $MVNOPT --settings ../ess-css-extra/maven/settings.xml clean verify) | tee 7_org.csstudio.ess.product.log
```
with the following ones
```
( \
  cd org.csstudio.ess.product; \
  time mvn $MVNOPT --settings ../ess-css-extra/maven/settings.xml clean verify; \
  cd repository/target/products; \
  export CSSVER=$(cat cs-studio-ess/linux/gtk/x86_64/cs-studio/ess-version.txt); \
  echo "Fixing BATIK dependencies for CS-Studio ${CSSVER}..."; \
  zip -d cs-studio-ess-${CSSVER}-win32.win32.x86_64.zip "plugins/org.apache.batik*1.8*.jar"; \
  zip -d cs-studio-ess-${CSSVER}-macosx.cocoa.x86_64.zip "ESS CS-Studio.app/Contents/Eclipse/plugins/org.apache.batik*1.8*.jar"; \
  mkdir cs-studio-ess-${CSSVER}-linux.gtk.x86_64; \
  cd cs-studio-ess-${CSSVER}-linux.gtk.x86_64; \
  tar -zxf ../cs-studio-ess-${CSSVER}-linux.gtk.x86_64.tar.gz; \
  find ./cs-studio/plugins -type f -iname 'org.apache.batik*1.8*.jar' -delete; \
  rm -f ../cs-studio-ess-${CSSVER}-linux.gtk.x86_64.tar.gz; \
  tar -zcf ../cs-studio-ess-${CSSVER}-linux.gtk.x86_64.tar.gz .; \
  cd ..; \
  rm -fr cs-studio-ess-${CSSVER}-linux.gtk.x86_64 \
) | tee 7_org.csstudio.ess.product.log
```

I've removed the 3 extra 1.8 batik jar files from the plugin directory. It's a patch but it works, waiting for a better solution.

utzeln, 2018-08-24:
We had the same issue - i.e. exception thrown when editing/running OPI files with SVG symbols:
`2018-08-22T18:36:57.357+02 SEVERE [Thread 1] org.csstudio.logging.PluginLogListener (logging) - Unhandled event loop exception
java.lang.LinkageError: loader constraint violation: when resolving method "org.apache.batik.bridge.BridgeContext.getCSSEngineForElement(Lorg/w3c/dom/Element;)Lorg/apache/batik/css/engine/CSSEngine;" the class loader (instance of org/eclipse/osgi/internal/loader/EquinoxClassLoader) of the current class, org/csstudio/utility/batik/SVGHandler, and the class loader (instance of org/eclipse/osgi/internal/loader/EquinoxClassLoader) for the method's defining class, org/apache/batik/bridge/BridgeContext, have different Class objects for the type org/apache/batik/css/engine/CSSEngine used in the signature
     at org.csstudio.utility.batik.SVGHandler.buildElementsToUpdateList(SVGHandler.java:786)
     at org.csstudio.utility.batik.SVGHandler.<init>(SVGHandler.java:188)
     at org.csstudio.swt.widgets.symbol.SVGSymbolImage.loadDocument(SVGSymbolImage.java:313)
`
And yes Eclipse Oxygen comes with batik 1.7 and 1.8 which is a problem for our fragment 
cs-studio/core/utility/utility-plugins/org.csstudio.utility.batik

as its META-INF/MANIFEST.MF does not restrict the version of the required batik bundles.

What is working for us is limit the version as having bundle-version="1.7.0" means in fact 1.7 version or higher:
`Require-Bundle: org.apache.batik.bridge;bundle-version="[1.7,1.8)",
 org.apache.batik.dom;bundle-version="[1.7,1.8)",
 org.apache.batik.dom.svg;bundle-version="[1.7,1.8)",
 org.apache.batik.transcoder;bundle-version="[1.7,1.8)",
 org.apache.batik.util;bundle-version="[1.7,1.8)",
 org.apache.batik.css;bundle-version="[1.7,1.8)",
 org.apache.batik.xml;bundle-version="[1.7,1.8)",
 org.apache.batik.ext.awt;bundle-version="[1.7,1.8)",`

Let me know if it works for you and we will make the pull request.

willrogers, 2018-09-03:
Wow - I just made a completely new release of CS-Studio and realised only later that these symbol widgets were not working. This should be a high priority fix! I will try to test the suggestion.

willrogers, 2018-09-03:
I made this change and it does appear to have fixed the problem. I will build a new product to double-check, then I will submit a pull request.

willrogers, 2018-09-03:
This works much better - the widgets load - but I still get the following exception:

```
Exception in thread "Thread-198" java.lang.LinkageError: loader constraint violation: loader (instance of <bootloader>) previously initiated loading for a different type with name "org/w3c/dom/events/Event"
	at java.lang.ClassLoader.findBootstrapClass(Native Method)
	at java.lang.ClassLoader.findBootstrapClassOrNull(ClassLoader.java:1015)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:413)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClassInternal(BundleLoader.java:479)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:395)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:387)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.loadClass(ModuleClassLoader.java:150)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClass(ClassLoader.java:763)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.defineClass(ModuleClassLoader.java:276)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.defineClass(ClasspathManager.java:655)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.findClassImpl(ClasspathManager.java:578)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.findLocalClassImpl(ClasspathManager.java:538)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.findLocalClass(ClasspathManager.java:525)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.findLocalClass(ModuleClassLoader.java:328)
	at org.eclipse.osgi.internal.loader.BundleLoader.findLocalClass(BundleLoader.java:368)
	at org.eclipse.osgi.internal.loader.sources.SingleSourcePackage.loadClass(SingleSourcePackage.java:36)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClassInternal(BundleLoader.java:430)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:395)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:387)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.loadClass(ModuleClassLoader.java:150)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClass(ClassLoader.java:763)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.defineClass(ModuleClassLoader.java:276)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.defineClass(ClasspathManager.java:655)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.findClassImpl(ClasspathManager.java:578)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.findLocalClassImpl(ClasspathManager.java:538)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.findLocalClass(ClasspathManager.java:525)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.findLocalClass(ModuleClassLoader.java:328)
	at org.eclipse.osgi.internal.loader.BundleLoader.findLocalClass(BundleLoader.java:368)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClassInternal(BundleLoader.java:446)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:395)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:387)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.loadClass(ModuleClassLoader.java:150)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	at org.apache.batik.dom.svg.SVGDOMImplementation$1.createEvent(SVGDOMImplementation.java:228)
	at org.apache.batik.dom.events.DocumentEventSupport.createEvent(Unknown Source)
	at org.apache.batik.dom.AbstractDocument.createEvent(Unknown Source)
	at org.apache.batik.bridge.BaseScriptingEnvironment.dispatchSVGLoad(BaseScriptingEnvironment.java:568)
	at org.apache.batik.bridge.BaseScriptingEnvironment.dispatchSVGLoad(BaseScriptingEnvironment.java:563)
	at org.apache.batik.bridge.BaseScriptingEnvironment.dispatchSVGLoad(BaseScriptingEnvironment.java:563)
	at org.apache.batik.bridge.BaseScriptingEnvironment.dispatchSVGLoad(BaseScriptingEnvironment.java:563)
	at org.apache.batik.bridge.BaseScriptingEnvironment.dispatchSVGLoadEvent(BaseScriptingEnvironment.java:550)
	at org.apache.batik.bridge.UpdateManager.dispatchSVGLoadEvent(UpdateManager.java:239)
	at org.apache.batik.bridge.UpdateManager.dispatchSVGLoadEvent(UpdateManager.java:220)
	at org.csstudio.utility.batik.SVGHandler$2.run(SVGHandler.java:457)
	at java.lang.Thread.run(Thread.java:748)
```

closed 2019-08-23 by berryma4