ControlSystemStudio/cs-studio #2449: Usage question loc:// from parent to macro from child opi

alvcarmona, 2018-07-26:
I'm a beginer using CSStudio. And I have a question about usage:
I have a opi with some text inputs and a button, the button opens another opi and I want to check some PVs with a name that is the composition of the tex input. I assign the text input to a loc://a, loc://b and loc://c variables and the PVs I want to check are something like  "loc://a:loc://b:loc://c" but I don't know how to make this composition in the PV property of a text update. Can I assign a loc:// value from the parent opi to a macro of the children? How? Is there a better way to do it?
Thanks

kasemir, 2018-07-26:
Good question.
Things like that never seemed to come up in the last 30 years of edd/dm, medm, dm2k, edm.
So BOY respectively the Display Builder don't have a built-in mechanism for that, but you can use a script to accomplish it. The thing about scripts, however, is that if you have to ask how to do it, you probably shouldn't use them, because you'll not be able to maintain it as details of the underlying API change.

Anyway, here's what you can do:

Create a display with one or more text inputs using PVs like `loc://pv_name1("sim://")`, `loc://pv_name2("sine")`, as you already have.

Create an ActionButton with an action to write the value 1 to the PV `loc://open$(DID)(0)`. That's similar to what you have, but instead of using the button to directly open the other display (and then wondering how to get the macros defined based on the PV values), you now have a display where you can edit the sections of you PV name, and a button that writes to an "..open.." PV.

Finally, add a script to the button:
The script has PVs  `loc://open$(DID)(0)` (trigger), `loc://pv_name1("sim://")` (not triggering), `loc://pv_name2("sine")` (not triggering).

So pushing the button writes 1 to the "..open.." PV which triggers the script, and that script also receives the other two PVs.

The script could in fact be attached to anything (display, any other widget). It's triggered by the `loc://open$(DID)(0)` PV, but since that in turn is written by the button, I would attach the script to the button, since that's where I'd be looking for it when I later need to update the display.

Content of that script:

```
# Running with BOY or Display Builder?
display_builder = 'getVersion' in dir(widget)
if display_builder:
    from org.csstudio.display.builder.runtime.script import PVUtil, ScriptUtil
    from java.util import LinkedHashMap
    if PVUtil.getLong(pvs[0]) > 0:
        # Get values of one or more PVs and turn them into macros
        # Could set several macros like this:
        # macros.put("PV1", PVUtil.getString(pvs[1]))
        # macros.put("PV2", PVUtil.getString(pvs[2]))
        # but in here want to combine values of several PVs into one macro
        macros = LinkedHashMap()
        macros.put("PV", PVUtil.getString(pvs[1]) + PVUtil.getString(pvs[2]))
        # Open display with those macros
        ScriptUtil.openDisplay(widget, "sub.opi", "TAB", macros)
else:
    from org.csstudio.opibuilder.scriptUtil import PVUtil, ScriptUtil
    from org.csstudio.opibuilder.util import MacrosInput
    from java.util import LinkedHashMap
    if PVUtil.getLong(pvs[0]) > 0:
        # Get values of one or more PVs and turn them into macros
        macros = LinkedHashMap()
        macros.put("PV", PVUtil.getString(pvs[1]) + PVUtil.getString(pvs[2]))
        # Open display with those macros
        ScriptUtil.openOPI(widget, "sub.opi", 0, MacrosInput(macros, True))   
```

claudio-rosati, 2018-07-27:
Kay, I don't understand why 

```
# Running with BOY or Display Builder?
display_builder = 'getVersion' in dir(widget)
if display_builder:
  ...
else:
  ...
```

is able to discriminate between BOY and Display Builder.

kasemir, 2018-07-27:
Both scripting engines receive a `widget` variable.
You can check what type you get via
```
print(widget.__class__.__name__)
```

If you look at the class hieratchy, you'll find that the display builder widgets are all derived from `Widget`, while BOY widgets are all based on `AbstractWidgetEditPart`.

One way to distinguish them is by looking for the `getVersion()` method, which the Display Builder provides by BOY doesn't. That's why the Display Builder examples in `script_util/portable.py` suggest this:
```
# To write a portable script, check for the display builder's widget type:
display_builder = 'getVersion' in dir(widget)

if display_builder:
  # Code specific to display builder.
  # In rare cases you need to further check for RCP or Phoebus,
  # read script_util/portable.py
else:
  # Code specific to BOY
```

alvcarmona, 2018-07-27:
@kasemir thanks for the reply. What would be the name of the macro in the child opi for the loc://pv_name1 of the parent? Im checking $(pv_name1) but is not working.

kasemir, 2018-07-27:
The name of the macro is whatever you decide for the name to be.
You can call it "Fred":
```
macros.put("Fred", PVUtil.getString(pvs[1]))
```

Again, if you have to ask how to do scripts, maybe that's a hint..

kasemir, 2018-07-27:
In the initial example, the macro passed into the newly opened display was "PV":
```
 macros.put("PV", ...)
```
So the newly opened "sub.opi" would receive a macro "$(PV)".

alvcarmona, 2018-07-27:
Thanks  @kasemir now I can do what I wanted.

closed 2018-07-27 by alvcarmona