ControlSystemStudio/cs-studio #2470: OPI won't start in standalone window

Amadrath, 2018-08-08:
When I try to start an opi in standalone mode, I get the following error:
`2018-08-08 16:22:25.179 SCHWERWIEGEND [Thread 1] org.csstudio.logging.PluginLogListener (logging) - Unhandled event loop exception
org.eclipse.swt.SWTException: Widget is disposed`
And the Standalone OPI Summary shows:
`org.eclipse.core.runtime.CoreException: Plug-in "org.csstudio.opibuilder" was unable to instantiate class "org.csstudio.opibuilder.runmode.OPIShellSummary".`
I have tried this on an SNS and NSLS flavours. Has anybody a clue?

kasemir, 2018-08-08:
Try to open the menu CS-Studio, Debugging, Logging Configuration.
If that also fails, you're using some JRE that lacks JavaFX.
Do you have an Oracle JDK that includes JavaFX?
Check Help, About, Installation Details, Configuration for *java*.

This will work:
```
java.vendor=Oracle Corporation
java.version=1.8.0_121
java.vm.name=Java HotSpot(TM) 64-Bit Server VM
java.vm.specification.name=Java Virtual Machine Specification
java.vm.specification.vendor=Oracle Corporation
java.vm.specification.version=1.8
java.vm.vendor=Oracle Corporation
java.vm.version=25.121-b13
```

If it's OpenJDK, or a RedHat installation of oracle-jre without jfx, it won't suffice.

Amadrath, 2018-08-09:
The missing JavaFX was the problem. After installing 'openjfx' (for 'openjdk-8') the standalone windows opened.

kasemir, 2018-08-09:
Good. That's by design. Several of the newer components use JavaFX for graphics, so they won't work unless that's available.

closed 2018-08-09 by kasemir

willrogers, 2018-08-09:
I'm interested in how you installed 'openjfx', as that's not always very easy.

Amadrath, 2018-08-09:
Simply: apt install openjfx

willrogers, 2018-08-09:
Thanks. It looks like Debian is ahead of RedHat.