ControlSystemStudio/cs-studio #2475: Context menu on Data Browser (value) axes

skyfrench, 2018-08-13:
Configuring axes can take several clicks and can be unintuitive.

Propose a context menu for the value axes to make it easier to change axis properties:

e.g.
[image]

kasemir, 2018-08-13:
The older SWT RTPlot has an unfortunate configuration dialog that opens up access to too much detail.
For example, it allows changing the horizontal axis from a time axis to a log numeric axis, which doesn't make any sense for the Data browser. LIkewise, it would allow changing a numerix axis in an XYPlot widget to a time axis.

The newer JavaFX RTPlot has a simpler configuration dialog which is very close to what you suggest for an axis dialog:

[image]

So I suggest moving up to Data Browser 3 and thus the JFX RTPlot. Of course if you prefer you can do whatever you want with Data Browser 2 and the SWT RTPlot.

willrogers, 2018-08-13:
> So I suggest moving up to Data Browser 3 and thus the JFX RTPlot

On the one hand I would like to, on the other it's non-trivial to integrate another repo into our build chain and I'd like to get some improvements to our users.

willrogers, 2018-08-13:
In fact, is Eric hosting a p2 repo for Display Builder and DB3?