ControlSystemStudio/cs-studio #2483: Data Browser archive data request type preference

skyfrench, 2018-08-14:
It would be nice to allow a user to specify a Data Browser default archive data request method in Preferences.

kasemir, 2018-08-14:
Not sure.
The data browser defaults to 'optimized', and what that means is up to the archive data source.
For the RDB, it will fetch N (default is N=800) min/max/average samples, unless there are fewer than 800 raw samples in the requested start..end range, in which case you get the raw samples.
This means you can look at 1 year, 1 month, 1 week of data and get the best possible overview, i.e. outline of min/max value range as well as average, while reducing the network traffic as well as client memory usage. As you zoom in, you eventually get the raw data.

Why would you want to switch to 'raw'?

1) The 'optimized' implementation is broken for your archive data source.  -> Fix it.
2) The user always wants to see the raw data. Yes, of course. But then the user will soon complain that fetching the data takes longer, redrawing the plot takes longer, and CSS aborts with out-of-memory. Not a practical approach.
3) There are edge cases where the plot for a channel simply looks odd when requesting optimized data. This is typically with sparse data, for time ranges around the threshold of automatically switching between optimized and raw data. Manually switching to raw makes the plot look better. Sometimes increasing the number of `plot_bins` for optimized data helps to avoid the situation. In theory, it should be similar to the screen width in pixels: In a plot that's 1000 pixels wide, showing more than 1000 data points doesn't improve the display. Makes more sense to show 1000 min/max/average samples from the 'optimized' request. But that's the theory, which in practice differs from reality. Not sure what to do about it, but again simply always using 'raw' doesn't seem like a good solution.

berryma4, 2018-08-14:
What might be a nice feature, would be to have an option in the preferences to set the plot_bins from the pixels available. This way, we wouldn't have to have a different config for different monitor resolutions.

willrogers, 2018-08-14:
OK, I'm behind this suggestion. We have a real problem, let's explore what the correct answer is. It might be that you cover it in case three.

We have a magnet at 0A. We turn it on, and it rapidly moves to 10A. In the final bin are values 7, 8, 9, and 10; the average is 8.5 A. It stays at 10A for multiple bins, then we turn it off. The next bin with values is 10 and 9; average 9.5A.

What we see is trace at 8.5A for the whole period where the magnet has actually been at 10A. This will induce (and has induced) alarm in the control room. Switching to 'raw' data gives the correct view and usually results in the user vowing always to switch to raw data.

Where in the chain do you think the problem is here?

kasemir, 2018-08-14:
Yes, this is the edge case.

Most of the time, the data doesn't change. You have less than `plot_bins` values. The data does, however, change rapidly within a 'bin'.

For the RDB, our stored procedure in Oracle would notice that that the total sample count between start and end is less than `plot_bins`, so you get the raw data. But as you zoom out, there can indeed be a case where the total sample count exceeds `plot_bins`, so you end up with flat lines until you reach to the sudden flurry of changes which are all within one 'bin'.

Is this with the archive appliance? Does it not switch based on the total count of samples between start and end?

Also, even if you have 7, 8, 9, 10 within a bin, don't you still get an optimized `{ min=7, max = 10, avg=8.5 }` reading, so the outline shows how the value spiked to 10?

An easy solution might be to just set the `plot_bins` to 4000. Somewhat related to Eric's suggestion, you don't need the exact pixel width of the screen. Having 4000 samples for a 1000 pixel screen is fine, and asserts that you keep getting enough raw samples to avoid your magnet example problem. The point is that we really don't need 100,000 samples for a 1000 pixel screen. Reducing 100,000 raw samples to 4000 optimized samples is what's really helpful.

berryma4, 2018-08-14:
If switching to raw fixes your 'issue', that would mean bin per pixel would 'fix' your issue. Although, I would suggest showing the min/max/std, because they could still be missing information.

kasemir, 2018-08-14:
In theory bin per pixel seems right, but in practice exact pin per pixel doesn't work for me because the pixels and bins aren't aligned. If you use 2 or 3 bins per pixel, then raw data that fits the screen is drawn as raw data, and only once you move into having 2 or 3 data points per pixel, i.e. you can no longer really separate them, will it switch to bins, and thus you don't notice any loss. Quite the opposite: You now see outline vs. average, which is better than just a fat vertical line.

berryma4, 2018-08-14:
Would this work better?
Have an option in the preferences to set the plot_bins *3 from the pixels available. This way, we wouldn't have to have a different config for different monitor resolutions.

(By better, I mean, less people thinking they need to 'switch to raw'.)

kasemir, 2018-08-14:
Yes, I'll try to implement this for DB3, and if that works, we can back port to DB2:
```
# Number of binned samples to request for optimized archive access.
# Negative values scale the display width,
# i.e. -3 means: 3 times Display pixel width.
plot_bins=-3
```

kasemir, 2018-08-14:
Seems easy enough to implement and I think will be helpful. See DB3 example linked above.

kasemir, 2018-08-14:
Will create pull request for DB2

willrogers, 2018-08-14:
So I don't think that this is exactly the same problem, but I was playing around with a similar PV and I got a plot like the below. I find it hard to interpret. Is this clear when you know what you're looking for?

[image]

willrogers, 2018-08-14:
Lots of samples, although I'm not sure how to interpret. The first period has lots of similar-but-different samples:

[image]

Then we get a lot that are all the same. Presumably, this is just repeating the last bin with data, though I'd have to dig some more:

[image]

Finally, a lot that are all the same at a lower value:

[image]

willrogers, 2018-08-14:
If we zoom in on the left half of the plot, it's updating at 10Hz, so the binning is working correctly.
For most of the right half of the plot, it's just showing the contents of one bin, the one shown in the second image of samples.

kasemir, 2018-08-14:
The sample view suggests to me that the plot shows exactly what the archiver appliance provides in min/max/avg samples. Most bins hold ~5000 raw samples, so you have quite some reduction in samples, good for faster drawing and less memory usage.

The problem is that the samples look odd. Is the AA providing nonsense for optimized data? Does it not make sense compared to the raw data? Could be either a problem in the AA web service, or the 'pbraw' client lib.

willrogers, 2018-08-14:
This is exactly the same situation. What happened (at about 9am on 10th August) in the raw data is that the current dropped from 90 A to 0 A over a period of 2 seconds, after which there are no more events until this morning. What's being plotted in that gap is the problem.

The web view has done a much better job:

[image]

willrogers, 2018-08-14:
The web view has done this by just getting the last sample from each 3-minute period:

http://cs03r-cs-serv-55.pri.diamond.ac.uk:8080/retrieval/data/getData.qw?pv=lastSample_180(SR13A-PC-QUADF-04:I)&from=2018-08-07T16:15:00.000Z&to=2018-08-14T16:16:52.727Z

willrogers, 2018-08-14:
Meanwhile, you're right, the AA is returning multiple versions of that last bin after there are no more events. This doesn't look right.

```
306 { "secs": 1533886578, "val": [90.2809985327711,0.0023051445738371435,90.27474975585938,90.28955078125,6652.0], "n    anos": 0, "severity":0, "status":0 },
307 { "secs": 1533887334, "val": [90.28097586122523,0.002312926572795892,90.27503204345703,90.28968048095703,6639.0],     "nanos": 0, "severity":0, "status":0 },
308 { "secs": 1533888090, "val": [90.12221041852315,3.3658977152393765,-0.0038155915681272745,90.28824615478516,5679.    0], "nanos": 0, "severity":0, "status":0 },
309 { "secs": 1533888846, "val": [90.12221041852315,3.3658977152393765,-0.0038155915681272745,90.28824615478516,5679.    0], "nanos": 0, "severity":0, "status":0 },
310 { "secs": 1533889602, "val": [90.12221041852315,3.3658977152393765,-0.0038155915681272745,90.28824615478516,5679.    0], "nanos": 0, "severity":0, "status":0 },
311 { "secs": 1533890358, "val": [90.12221041852315,3.3658977152393765,-0.0038155915681272745,90.28824615478516,5679.    0], "nanos": 0, "severity":0, "status":0 },
312 { "secs": 1533891114, "val": [90.12221041852315,3.3658977152393765,-0.0038155915681272745,90.28824615478516,5679.    0], "nanos": 0, "severity":0, "status":0 },
```

berryma4, 2018-08-14:
This is great! Thank you!

kasemir, 2018-08-14:
The total data chain is something like this:
1) Raw data in disk files
2) AA code
3) AA web service
4) protobuf-java.jar
5) org.csstudio.archive.reader.appliance.ApplianceArchiveReader
6) Data Browser's ArchiveFetchJob
7) Data Browser's Sample inspector, Plot

At the end we find the erroneously duplicated optimized samples.
Do you see them returned from the AA web service, using curl, wget or a web browser?

berryma4, 2018-08-14:
Maybe something isn't right with the AA. I was just looking into this, which also shows some erroneous data.

all data is limited to +/-90 deg range, but plot shows data outside this range

[image]

willrogers, 2018-08-14:
The json above comes from wget, so I think the problem is coming from early on in your sequence.

willrogers, 2018-08-17:
OK, so we've established that there is a problem with the data. When there are no points in a bin, the summary from the previous bin is being duplicated. In the above examples, we keep plotting the result from the final bin where there was data.

Now, the question is _**what should the binning algorithm do**_? I find that scipy returns NaNs for empty bins: https://docs.scipy.org/doc/scipy/reference/generated/scipy.stats.binned_statistic.html. But what we need to plot in this case is a line from the last sample in the last non-empty bin.

What do the other archivers do in this case?

kasemir, 2018-08-17:
The SciPy idea doesn't apply to us because it assumes you always sample the data, and no sample means there's a gap in the data. But we like to use 'monitor', so no new samples mean no change, the last value is still the current value of the PV. So our plot continues from the last sample, unless the last sample is NaN to indicate that we have  agap.

For us, there's a difference between no new samples and NaN.

When a bin has an 'Archive off' type sample, NaN should be returned, so the trace ends at that bin. There's a gap until we again have a numeric value.

If there is no data, nothing should be returned for that bin. That means the last valid numeric value still applies. I cannot find real data for that right now, but it would look like this:

[image]

The last bin contained several raw samples, so we have a min/max/average sample.
Then nothing, until finally some new sample. The line continues from the last average to the new sample. The outline continues from the last min/max to the new sample.

Simply repeating the last min/max/avg is wrong, because it suggests there was data, and it continued to fluctuate between min/max all the time. So that's a bug in the AA.

For an implementation example, check org.csstudio.archive.reader.rdb.AveragedValueIterator. It implements the binning if there's no server-side method like a stored procedure.

kasemir, 2018-08-17:
Sorry, wrong picture. This sketch illustrates a last optimized min/max/avg sample, then no new samples, until finally one new sample:

[image]

willrogers, 2018-08-17:
So even if we fix the bug, we have a problem.

In the example I provided, the last bin contained the magnet ramping down to zero at the end. However, the average for the last bin was 90, almost the maximum value. In the gap, there are no new samples but the correct magnet value is 0, but if we plot from the _last bin average_ it will look like there was a current of 90 A for the whole of that time, even if the area does reach down to zero.

kasemir, 2018-08-17:
Yes, you have an unfortunate edge case where the value varies a lot for a little while, then stays constant. Maybe this type of channel should better be archived with 'scan': Save a sample every N seconds, only suppressing them if the value stays the same. That way, you don't get the flurry of samples within a bin. Or do prepare a plot for your magnets that indeed uses "Raw".
Or come up with a mechanism that perfectly handles every possible type of data, showing every detail, without ever running out of memory.

willrogers, 2018-08-17:
> For what it's worth, the channel archiver's CGI data server optimizes by turning N samples in a bin into initial value, min, max, final. When drawn with a continuous line, you also get the basic idea of a min/max outline, but then continue from the final value. Could pick that idea back up.

This is exactly what we need (possibly average as well). Does CS-Studio support plotting such a summary?

kasemir, 2018-08-17:
Yes. How you implement ArchiveReader.getOptimizedValues() is up to the data source.
You can return just the raw data as a series of VType, with will be drawn as a line.
Or optimized data as a series of VStatistics, which will be drawn as a line for the average with outline for min/max, another faint line for the std. dev.

Or N - to - { initial,min,max,final } as a series of { VType, VType, VType, VType }, which will look good on the plot and does what you now like to to. It has the disavantage that when you export the data, you no longer know that the quadruple represents initial, min, max, average. So next you'll get emails from your users to explain those numbers. You could ask the maintainers of the VStatistics to add a 'finalValue()' method, so then we could plot it and preserve the information about min, max.

rjwills28, 2020-11-13:
Will and myself have been looking into this a little more recently and we have a proposed solution that could improve the data visualisation in these cases. Most of the changes would be made on the Archiver Appliance side. We would add a new `PostProcessor` (based on the current `Optimized` postprocessor that is used). The new `PostProcessor` (called `OptimizedWithLastSample`) would keep track of the last value that gets added to each bin before the statistics are computed on that bin. If a bin contains no samples, the current behaviour is to copy the statistics from the previous bin that had samples in it. Instead we would take the LAST VALUE recorded for the previous bin that had samples in it and use this value as the mean, min and max of the empty bin with the number of samples = 0 accordingly. 

For example:
- Bin with samples:
 Bin 40000: nSamples = 10, mean = 30, min = 0, max = 50, last value = 20
- Next bin has no samples:
Bin 40001: nSamples = 0, mean = 20, min = 20, max = 20, last value = 20
- Next bin has no samples:
Bin 40002: nSamples = 0, mean = 20, min = 20, max = 20, last value = 20

This will mean that during periods where the PV does change, the plot will always correctly show the last value reported.

We plan to put forward this solution to the epicsarchiverap group. Note that this implementation requires no change to the VStatistics or the databrowser code.

kasemir, 2020-11-13:
> If a bin contains no samples, the current behaviour is to copy the statistics from the previous bin

That clearly seems wrong, basically making up bad data.

> Instead we would take the LAST VALUE recorded for the previous bin that had samples in it and use this value as the mean, min and max of the empty bin with the number of samples = 0 accordingly.

Yes, that would be better, and correct since the last value is still valid until we receive an updated sample.

willrogers, 2021-03-01:
We have a solution that works for us.

closed 2021-03-01 by willrogers