ControlSystemStudio/cs-studio #2490: Data Browser Trace context menu

skyfrench, 2018-08-16:
It can take quite a few clicks to modify a trace (change its colour, axis etc) which can be frustrating.
Related to [#2475](https://github.com/ControlSystemStudio/cs-studio/issues/2475) it would be nice to implement a context menu for a trace.

Something like this:
[image]

kasemir, 2018-08-16:
There is already a context menu on the plot (save screenshot, print, email, send to logbook, ..). How would you switch between the plot's context menu and one specific to the trace?
I'm also not sure about the use case.
For the most part, this is a control system tool. Operators look at it. Maybe zoom & pan to find something. They shouldn't spend time changing the color, hiding a trace (why was the trace there in the first place??).

The runtime plot config dialog of the JFX version concentrates on those options that operators might need to change at runtime:

[image]

Changing the axis range, grid, log or autoscale are all about inspecting the data.

skyfrench, 2018-08-16:
The proposal isn't to add any functionality - but instead to make the existing functionality which is used a little easier to get to. It can take a number of clicks to get the Properties tab to display the traces settings. The menu would just make it possible to change some of the commonly modified trace properties without having to get the traces tab back in view each time. 

In terms of switching between the two context menus: clicking on (within a tolerance) a trace would bring up the trace menu; clicking elsewhere on the plot would bring up the existing context menu.

kasemir, 2018-08-16:
Can you click on a hidden trace to un-hide ;-)