ControlSystemStudio/cs-studio #2492: Download Page is Down

dhidas, 2018-08-16:
from here:
http://controlsystemstudio.org/controlsystemstudio/current.html

I get links to the different versions such as:
http://download.controlsystemstudio.org/updates/4.5

All seem to give an error message like:
404 Not Found

Code: NoSuchKey
Message: The specified key does not exist.
Key: updates/4.5
RequestId: FE48712496A83EDC
HostId: PfQTWRKQWKFYtFCXj1M2H/TzuU9OOxmSCbnjFA+0jvM3TrieCYYe02ZguxV77fWYo4k46GL8PkA=

berryma4, 2018-08-16:
We should probably make our webpage more clear.
You will need to click the "Artifacts" tab, then click the "Download" button.
The link ends up being:
http://download.controlsystemstudio.org/release/4.5/

The update links:
http://download.controlsystemstudio.org/updates/4.5
are Eclipse p2 sites, not for people consumption, but for entering into Eclipse.

willrogers, 2018-08-16:
I've said this before, I think the actual downloads should be the default.

willrogers, 2018-08-16:
https://github.com/ControlSystemStudio/cs-studio/issues/2443

berryma4, 2018-08-16:
I've update the download page to go to the artifacts page first, and changed the name to binaries.

closed 2018-08-16 by berryma4