ControlSystemStudio/cs-studio #2500: How to open databrowser plot

willrogers, 2018-08-22:
We have been making fixes to the databrowser, and one thing that often comes up in user feedback is that opening a plot is not as intuitive as it might be. In particular, you often end up with a new plot when what you would prefer is for a trace to be added to an existing plot. 

Some ideas:

* if you double-click on a search result in the Databrowser perspective, the trace could be added to the current plot on a new axis
* if you drag from a search result, it could still offer you the dialog to choose options
* if you use the context menu, the trace could be added to the current plot on the new axis

If this wasn't what you wanted, you can always open a new plot first using the toolbar button.

Other ideas:

* we could have a different context menu item to open a new plot
* would it be possible to drag OPI widgets into a databrowser plot?
* there's no current way to get the PV context menu from a probe
* should a databrowser axis disappear by default if no traces are plotted on it?

Any thoughts?

cc @skyfrench

kasemir, 2018-08-22:
The data browser already accepts dropped strings as PV names, even allowing more than one. I can drag the next line from the web browser into a data browser:

sim://sine sim://ramp

The result is two popup dialogs, one for each PV name, in case I want to pick a specific axis or configure the sample rate. In most cases just "OK" suffices.

Drag & drop makes it easy to pick a specific data browser for the new PVs.

From a context menu, like the context menu of a display widget, I don't see a good way to select the target data browser.

So I guess the most promising step would be to add 'drag' support to BOY, so you can drag PVs from displays into a data browser. I assume that won't be too hard to add to BOY, and the data browser is already ready to handle it.

From the search view you can already drag PVs into an existing plot. Adding double-click to add a PV sounds OK, but might again have to determine which target to use when there's more than one data browser open.

kasemir, 2018-08-22:
> should a databrowser axis disappear by default if no traces are plotted on it?

In many cases, yes, but what about the user who carefully configured an axis with label, color, limits, .., now moves some traces around, and the axis vanishes.
Likewise: Should the data browser close when you delete all traces?
I'd rather keep the plot as well as the axes around and let the user remove them as desired.

kasemir, 2018-08-22:
Added 'drag' to the display builder: Widgets with "pv_name" now allow dragging that PV name from the runtime widget, to be dropped into the data browser or other tools that accept PV names as text.

claudio-rosati, 2018-08-23:
I've tried to D&D from Display Builder meters example into the Data Browser... I like it a lot.
Thank you Kay!

kasemir, 2018-08-23:
Missed some detail.. right now the scroll bars (if the display content exceeds the window size) cannot be used because it's starting to 'drag' the scroll bar. That's a side effect of SWT/JFX DnD hell..
In principle, the 'drag' should start on each widget, i.e. make the JFX node for each widget a JFX drag source. That works for the phoebus version, is straight forward, and also allowed me to drag the PV names from the RCP display builder into an outside tool or into an SWT text editor.
... But it results in a crash when trying to drag from the display builder (JFX hosted in SWT FXCanvas) into the data browser (also JFX in FXCanvas). One FXCanvas translating the JFX Drag events in to SWT, then the other FXCanvas translating the SWT drag events into JFX resulted in null pointer exceptions, warnings about unexpected drag exit.. I gave up and use SWT drag on the display builder's FXCanvas, so now it tries to drag the scroll bars as well. Need to fix that.

kasemir, 2018-08-23:
Fixed the display builder scroll bar conflict, https://github.com/kasemir/org.csstudio.display.builder/commit/0537c7f5a458785ece338860abace297fa5cdadd

kasemir, 2018-08-27:
DragNdrop in display builder needs another fix because trying to move a slider now starts a 'drag' operation, just like trying to scroll a table that's so large that it has scroll bars.