ControlSystemStudio/cs-studio #2501: Data Browser 'Sample Inspector'

skyfrench, 2018-08-23:
We've had some feedback that it'd be very useful if the cross-hair cursor gave more detail and precision on the selected samples. Perhaps adding a pop up window on click (a 'Sample Inspector') to display the sample detail would help with this, without crowding up the existing cross-hair display?

kasemir, 2018-08-23:
I'm OK with a sample inspector, but unclear how to best open/control it.
Open on click will be confusing because the cross-hair is an add-on mode.
The cross hair can be on or off.
Meanwhile, unaffected by the cross hair, you can be in zoom, pan, or just plain mouse pointer mode.
If you're in zoom or pan mode, the click is used to start the, well, zoom or pan. Opening a sample inspector when you're trying to zoom or pan would be odd.

Adding to the context menu is somewhat difficult because the same underlying plot is used for the data browser, display builder XY plot etc., which each add their own application specific context menu. Best would be a sample inspector that's wholly handled within the shared plot code.

What would technically work: If the cross hair is on, and the mouse is in plain pointer mode, then a click opens a popup with sample detail. Should each click open another one, which remain open until closed?

skyfrench, 2018-08-24:
> What would technically work: If the cross hair is on, and the mouse is in plain pointer mode, then a click opens a popup with sample detail. Should each click open another one, which remain open until closed?

Our proposal was exactly that; left-click opens a sample detail pop-up (and at the moment each click opens a new one, the user decides when to close them).

kasemir, 2018-08-24:
> Our proposal was exactly that; left-click opens a sample detail pop

.. if the mouse_mode is NONE (plain pointer, not PAN, ZOOM_*).. OK, go for it.