ControlSystemStudio/cs-studio #2505: Using ADDR_LIST

DanielCuneo, 2018-08-24:
Getting started with Phoebus and using an addr_list like this:
128.3.131.255, like I would with EPICS Base.
This did not provide access to a test PV I'm developing with.

I set this value in a setting.ini file which I load via command line arguments (Windows 7).

However, using the exact IP address did:
128.3.131.56

I did not find documentation on how the addr_list works in detail.
Perhaps I missed it ?? TIA

kasemir, 2018-08-27:
Interesting. Are you sure that your netmask is set correctly, i.e. 128.3.131.255 is indeed the correct broadcast address?

I have this in my settings.ini which I load via the `-settings  /path/to/my/settings.ini` command line option:
```
org.phoebus.pv.ca/addr_list=10.111.36.255 160.91.228.17 
org.phoebus.pv.ca/auto_addr_list=false
```

So I'm using a broadcast address for the local subnet, plus the specific IP address for a gateway to other systems. The local subnet IOCs are found via the broadcast address.

This is on a computer with the following network configuration:
```
enp3s0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.111.36.130  netmask 255.255.255.0  broadcast 10.111.36.255
        inet6 fe80::ec4:7aff:fe06:37d4  prefixlen 64  scopeid 0x20<link>
```

So 10.111.36.255 is indeed the correct subnet broadcast.  To test, when I change the broadcast address to the erroneous 10.111.38.255 then my local IOCs are no longer found.

DanielCuneo, 2018-08-27:
Thank Kay for responding. Here's the output from `ifconfig` on the machine hosting the IOC:
`
enp2s0    Link encap:Ethernet  HWaddr 0c:c4:7a:81:4c:d4  
          inet addr:128.3.130.64  Bcast:128.3.131.255  Mask:255.255.252.0
`
I'll try adding the setting; `org.phoebus.pv.ca/auto_addr_list=false` and see what happens.
Sounds like it's not a bug just configuration goof. I'll post back with results.

Thanks for you help.

kasemir, 2018-08-27:
OK, you show the bcast address on the machine hosting the IOC.
What about the machine running the client (phoebus)?
For windows, try `ipconfig`. Is it also on the same subnet, with the same netmask and bcast address?

Or is it on a different subnet? In that case, the routers between your client computer and the IOC host may not pass bcast traffic, which would explain why you need to use the absolute IP.

DanielCuneo, 2018-08-27:
So I look dumb again...

W7 client is on a different subnet: `128.3.130.128`
I'll try adjusting the subnet mask to broadcast, otherwise it's a router issue.

Cheers

kasemir, 2018-08-28:
Well, now I'm all confused. First you implied that your IOC is on a host with IP 128.3.131.56, because using the exact IP address you were able to connect.
Then you wrote that this is the machine hosting the IOC:  inet addr:128.3.130.64 Bcast:128.3.131.255 Mask:255.255.252.0. That's a machine on the same subnet, but a different host. 128.3.130.64 vs.  128.3.131.56. Now you write "different subnet: 128.3.130.128", but based on your previous comments,  128.3.130.128 would be a host address, not a network address.

Draw a map of your subnets, or try some online subnet mask calculator like 
http://www.subnet-calculator.com/subnet.php?net_class=B

Given the example with IP 128.3.130.64 and netmask 255.255.252.0, your subnet would include every IP in the range of 128.3.128.1 - 128.3.131.254, with a broadcast address of 128.3.131.255.
Assert that all computers on the same subnet use the same net mask and thus are truly in agreement that they're on the same subnet.
Now the bcast address should work. If not, run wireshark on the host that's running the IOC, and check if it receives the bcast search requests, i.e. messages to UDP 5064, sent from the client computer that's trying to connect via channel access. If the IOC doesn't receive the UDP search requests, look for firewalls, switches, routers that block them.

DanielCuneo, 2018-08-28:
To clear things up:

Linux IOC host IP address:
128.3.130.64
255.255.252.0 (10 bit subnet)

W7 Phoebus client IP address:
128.3.128.20 (should be covered in the 10 bit address of the subnet)
255.255.255.0  (wrong netmask)

The problem is the subnet mask on the W7 machine.
Thank you for the lesson. 

Last question, what does the option: `org.phoebus.pv.ca/auto_addr_list=false` do ?
Cheers

kasemir, 2018-08-28:
Ah, wrong subnet mask! Yes, I've had that problem, can take a long time to figure out.

The auto_addr_list is one of the fundament channel access settings, see https://epics.anl.gov/base/R3-14/12-docs/CAref.html .

By default, the client uses the local broadcast address, i.e. what you see in the ifconfig or ipconfig output. So for IOCs on the same subnet, you don't have to do anything.
For IOCs on other subnets, you add the broadcast addresses of the other subnets to the addr list (and then it might still not work because routers block the broadcasts, so you need a gateway).

I've just set auto_addr_list=false in the example so I can be sure that what I add to the addr_list is all that it's using, not silently adding the bcast of the local network interface.

closed 2018-09-11 by kasemir