ControlSystemStudio/cs-studio #2506: Alarm system no more working after MySQL drivers updated to ver. 8.

claudio-rosati, 2018-08-29:
After the MySQL drivers update the following exception happened when opening the Alarm perspective:

```
Loading class `com.mysql.jdbc.Driver'. This is deprecated. The new driver class is `com.mysql.cj.jdbc.Driver'. The driver is automatically registered via the SPI and manual loading of the driver class is generally unnecessary.
2018-08-29 11:53:57.381 SEVERE [Thread 20] org.csstudio.alarm.beast.ui.clientmodel.AlarmClientModel (readConfiguration) - Cannot connect to RDB
java.lang.Exception: Cannot connect to database, No suitable driver found for jdbc:mysql://172.16.107.12/alarm
        at org.csstudio.alarm.beast.client.AlarmConfiguration.<init>(AlarmConfiguration.java:98)
        at org.csstudio.alarm.beast.client.AlarmConfiguration.<init>(AlarmConfiguration.java:125)
        at org.csstudio.alarm.beast.ui.clientmodel.AlarmClientModel$1.<init>(AlarmClientModel.java:398)
        at org.csstudio.alarm.beast.ui.clientmodel.AlarmClientModel.readConfiguration(AlarmClientModel.java:398)
        at org.csstudio.alarm.beast.ui.clientmodel.ReadConfigJob.run(ReadConfigJob.java:49)
        at org.eclipse.core.internal.jobs.Worker.run(Worker.java:56)
Caused by: java.sql.SQLException: No suitable driver found for jdbc:mysql://172.16.107.12/alarm
        at java.sql.DriverManager.getConnection(DriverManager.java:689)
        at java.sql.DriverManager.getConnection(DriverManager.java:208)
        at org.csstudio.platform.utility.rdb.internal.MySQL_RDB.connect(MySQL_RDB.java:63)
        at org.csstudio.platform.utility.rdb.RDBUtil.<init>(RDBUtil.java:179)
        at org.csstudio.platform.utility.rdb.RDBUtil.connect(RDBUtil.java:143)
        at org.csstudio.alarm.beast.client.AlarmConfiguration.<init>(AlarmConfiguration.java:94)
        ... 5 more
```

kasemir, 2018-08-29:
I can confirm.
Looks like the driver name changed from com.mysql.jdbc.Driver to com.mysql.cj.jdbc.Driver, and the new name is not exported by platform.libs.jdbc.

kasemir, 2018-08-29:
About to create pull requests that fix this.

MySQL sure has changed. Not only the package names, but also some URL detail.
For my MySQL version (5.5.60-MariaDB), using the updated JDBC driver, I can no longer connect as just `jdbc:mysql://localhost/ALARM`.

Need to use `jdbc:mysql://localhost/ALARM?useUnicode=true&useJDBCCompliantTimezoneShift=true&useLegacyDatetimeCode=false&serverTimezone=UTC`, see https://stackoverflow.com/questions/26515700/mysql-jdbc-driver-5-1-33-time-zone-issue

closed 2018-08-29 by berryma4

kasemir, 2018-08-30:
Claudio, is the previously used URL like `jdbc:mysql://localhost/ALARM` sufficient with your setup, or do you get some time zone related error on connect, so you must use a URL like `jdbc:mysql://localhost/ALARM?serverTimezone=UTC`

claudio-rosati, 2018-08-31:
Thank you Kay. That was the following problem and I found the answer googling about MySQL URLs and properties. Thank you again for your prompt answer.

PR #2509 (2018-08-29): Mysql 2506
Fixes #2506