ControlSystemStudio/cs-studio #2530: jdk11 and Eclipse Photon

willrogers, 2018-10-05:
A new ticket to describe the transition to the jdk11 and Eclipse Photon. We have started this work and find quite a few difficulties.

We should update to jdk11 as it is the next long-term support release, even if OpenJDK don't support LTS releases explicitly. I think we need to update to Photon in order to be able to do this, so we should do it at the same time.

The current problem I am stuck on is `javax.xml.bind`. This is required by `org.csstudio.diirt.util.core.preferences`. The bundle `javax.xml.bind` is available in Orbit, but I can't start the bundle because it can't find the `javax.xml` package. I don't understand this problem.

```
osgi> start 3731
org.osgi.framework.BundleException: Could not resolve module: org.csstudio.diirt.util [3731]
  Unresolved requirement: Require-Bundle: org.csstudio.diirt.util.core.preferences; bundle-version="1.0.2"
    -> Bundle-SymbolicName: org.csstudio.diirt.util.core.preferences; bundle-version="1.0.2.201810050804"; singleton:="true"
       org.csstudio.diirt.util.core.preferences [3732]
         Unresolved requirement: Require-Bundle: javax.xml.bind
           -> Bundle-SymbolicName: javax.xml.bind; bundle-version="2.2.0.v201105210648"
              javax.xml.bind [3679]
                Unresolved requirement: Import-Package: javax.xml; version="[1.3.0,2.0.0)"
```

@ifmalcuaz has just started at Diamond and is working on this problem as well.

@claudio-rosati @shroffk @kasemir @berryma4 do you have any ideas?

willrogers, 2018-10-05:
I've made a little bit more progress. Pinning the latest `jaxb` versions  gets a slightly different exception:

```
2018-10-05 11:40:20.842 SEVERE [Thread 1] org.csstudio.diirt.util.DiirtStartup (preWindowOpen) - Implementation of JAXB-API has not been found on module path or classpath.
javax.xml.bind.JAXBException: Implementation of JAXB-API has not been found on module path or classpath.
 - with linked exception:
[java.lang.ClassNotFoundException: com.sun.xml.bind.v2.ContextFactory]
	at javax.xml.bind.ContextFinder.newInstance(ContextFinder.java:269)
	at javax.xml.bind.ContextFinder.find(ContextFinder.java:412)
	at javax.xml.bind.JAXBContext.newInstance(JAXBContext.java:721)
	at javax.xml.bind.JAXBContext.newInstance(JAXBContext.java:662)
	at org.csstudio.diirt.util.core.preferences.pojo.DataSources.toFile(DataSources.java:171)
	at org.csstudio.diirt.util.core.preferences.DIIRTPreferences.toFiles(DIIRTPreferences.java:733)
	at org.csstudio.diirt.util.core.preferences.DIIRTPreferences.toFiles(DIIRTPreferences.java:773)
	at org.csstudio.diirt.util.DiirtStartup.preWindowOpen(DiirtStartup.java:46)
```

However, `com.sun.xml.bind.v2.ContextFactory` is in `org.glassfish.jaxb.runtime_2.4.0.b180830_0438` and `org.glassfish.jaxb.runtime_2.4.0.b180830_0438` is loaded, so I'm puzzled as to why this exception is happening.

claudio-rosati, 2018-10-05:
Hello @willrogers,

what you described at first it seems very strange o me. Here is the manifest for the `org.csstudio.diirt.util.core.preferences` plugin:

```java
Manifest-Version: 1.0
Bundle-ManifestVersion: 2
Bundle-Name: DIIRT Core Preferences
Bundle-SymbolicName: org.csstudio.diirt.util.core.preferences;singleton:=true
Bundle-Version: 1.0.2.qualifier
Bundle-Activator: org.csstudio.diirt.util.core.preferences.Activator
Bundle-Vendor: European Spallation Source ERIC
Require-Bundle: org.apache.commons.io;bundle-version="2.5.0",
 org.apache.commons.lang3;bundle-version="3.1.0",
 org.eclipse.core.runtime,
 org.epics.jca;bundle-version="2.3.6"
Bundle-RequiredExecutionEnvironment: JavaSE-1.8
Export-Package: org.csstudio.diirt.util.core.preferences;uses:="org.osgi.framework,org.eclipse.osgi.util,org.eclipse.core.runtime.preferences",
 org.csstudio.diirt.util.core.preferences.pojo;uses:="org.csstudio.diirt.util.core.preferences"
Bundle-ActivationPolicy: lazy
```

As you can see there is no (explicit) reference to `javax.xml.bind`. So I wonder where it come from.

willrogers, 2018-10-05:
I have added the dependency because `javax.xml.bind` was previously in the JDK, but now has been removed and needs to be found from elsewhere:

```
Manifest-Version: 1.0
Bundle-ManifestVersion: 2
Bundle-Name: DIIRT Core Preferences
Bundle-SymbolicName: org.csstudio.diirt.util.core.preferences;singleton:=true
Bundle-Version: 1.0.2.qualifier
Bundle-Activator: org.csstudio.diirt.util.core.preferences.Activator
Bundle-Vendor: European Spallation Source ERIC
Require-Bundle: org.apache.commons.io;bundle-version="2.5.0",
 org.apache.commons.lang3;bundle-version="3.1.0",
 org.eclipse.core.runtime,
 org.epics.jca;bundle-version="2.3.6",
 jaxb-api;bundle-version="2.4.0",
 org.glassfish.jaxb.runtime;bundle-version="2.4.0"
Bundle-RequiredExecutionEnvironment: JavaSE-1.8
Import-Package: com.sun.xml.bind.v2
Export-Package: org.csstudio.diirt.util.core.preferences;uses:="org.osgi.framework,org.eclipse.osgi.util,org.eclipse.core.runtime.preferences",
 org.csstudio.diirt.util.core.preferences.pojo;uses:="org.csstudio.diirt.util.core.preferences"
Bundle-ActivationPolicy: lazy
```

willrogers, 2018-10-05:
I think you misunderstand me. I need to **add** a dependency that provides JAXB, because it is no longer available in the JDK.

willrogers, 2018-10-05:
This bug seems similar, but I don't think is the same because I am trying to depend directly on the bundle that is causing the problem: https://bugs.eclipse.org/bugs/show_bug.cgi?id=534884

willrogers, 2018-10-05:
I have gone as far as providing an entirely different JAXB implementation (`org.eclipse.persistence.moxy`) and I see exactly the same problem. I really don't know how to solve this problem.

berryma4, 2018-10-05:
Is there a branch where we can see what is going on?
Are you adding ```--upgrade-module-path {path}```

willrogers, 2018-10-09:
Sorry, we found a bug in fast orbit feedback that has taken precedence. I hope to look at this again this week.

shroffk, 2018-10-10:
> Are you adding --upgrade-module-path {path}

This is not a good long term solutions, while the modules are temporarily available in the the jdk they are marked for removal. Thus this problem would resurface then.

@willrogers in addition to jaxb-api did you include the jaxb-impl(http://central.maven.org/maven2/javax/xml/jaxb-impl/)

willrogers, 2018-10-10:
I included `jaxb-api` and `org.glassfish.jaxb:jaxb-runtime`. The second should be a valid implementation but I couldn't make it work.

willrogers, 2018-10-10:
Now I'm running into problems with Javafx. I could at least build using a pre-built JVM including JavaFX, but I can't even build against the Maven version of javafx.

willrogers, 2018-10-11:
So, with the pre-built JVM, I can run and most things work. If I open the Logging Configuration view (JavaFX), nothing is rendered, but I get no exception. When I close the view, I get the following exception. I don't really know what to do about this one.

```
2018-10-11 09:38:09.885 SEVERE [Thread 1] org.csstudio.logging.PluginLogListener (logging) - Problems occurred when invoking code from plug-in: "org.eclipse.e4.ui.workbench.swt".
org.eclipse.e4.core.di.InjectionException: java.lang.NullPointerException
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:65)
	at org.eclipse.e4.core.internal.di.InjectorImpl.processAnnotated(InjectorImpl.java:1005)
	at org.eclipse.e4.core.internal.di.InjectorImpl.processAnnotated(InjectorImpl.java:970)
	at org.eclipse.e4.core.internal.di.InjectorImpl.disposed(InjectorImpl.java:463)
	at org.eclipse.e4.core.internal.di.Requestor.disposed(Requestor.java:156)
	at org.eclipse.e4.core.internal.contexts.ContextObjectSupplier$ContextInjectionListener.update(ContextObjectSupplier.java:78)
	at org.eclipse.e4.core.internal.contexts.TrackableComputationExt.update(TrackableComputationExt.java:111)
	at org.eclipse.e4.core.internal.contexts.TrackableComputationExt.handleInvalid(TrackableComputationExt.java:74)
	at org.eclipse.e4.core.internal.contexts.EclipseContext.dispose(EclipseContext.java:184)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.clearContext(PartRenderingEngine.java:1003)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeRemoveGui(PartRenderingEngine.java:976)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.access$1(PartRenderingEngine.java:883)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$4.run(PartRenderingEngine.java:878)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:42)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.removeGui(PartRenderingEngine.java:862)
	at org.eclipse.e4.ui.workbench.renderers.swt.ElementReferenceRenderer.disposeWidget(ElementReferenceRenderer.java:112)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeRemoveGui(PartRenderingEngine.java:955)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.access$1(PartRenderingEngine.java:883)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$4.run(PartRenderingEngine.java:878)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:42)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.removeGui(PartRenderingEngine.java:862)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.subscribeTopicToBeRendered(PartRenderingEngine.java:183)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:55)
	at org.eclipse.e4.core.di.internal.extensions.EventObjectSupplier$DIEventHandler.handleEvent(EventObjectSupplier.java:88)
	at org.eclipse.equinox.internal.event.EventHandlerWrapper.handleEvent(EventHandlerWrapper.java:201)
	at org.eclipse.equinox.internal.event.EventHandlerTracker.dispatchEvent(EventHandlerTracker.java:196)
	at org.eclipse.equinox.internal.event.EventHandlerTracker.dispatchEvent(EventHandlerTracker.java:1)
	at org.eclipse.osgi.framework.eventmgr.EventManager.dispatchEvent(EventManager.java:230)
	at org.eclipse.osgi.framework.eventmgr.ListenerQueue.dispatchEventSynchronous(ListenerQueue.java:148)
	at org.eclipse.equinox.internal.event.EventAdminImpl.dispatchEvent(EventAdminImpl.java:135)
	at org.eclipse.equinox.internal.event.EventAdminImpl.sendEvent(EventAdminImpl.java:78)
	at org.eclipse.equinox.internal.event.EventComponent.sendEvent(EventComponent.java:39)
	at org.eclipse.e4.ui.services.internal.events.EventBroker.send(EventBroker.java:52)
	at org.eclipse.e4.ui.internal.workbench.UIEventPublisher.notifyChanged(UIEventPublisher.java:60)
	at org.eclipse.emf.common.notify.impl.BasicNotifierImpl.eNotify(BasicNotifierImpl.java:374)
	at org.eclipse.e4.ui.model.application.ui.impl.UIElementImpl.setToBeRendered(UIElementImpl.java:304)
	at org.eclipse.e4.ui.internal.workbench.PartServiceImpl.hidePart(PartServiceImpl.java:1378)
	at org.eclipse.e4.ui.internal.workbench.PartServiceImpl.hidePart(PartServiceImpl.java:1308)
	at org.eclipse.e4.ui.workbench.renderers.swt.StackRenderer.closePart(StackRenderer.java:1250)
	at org.eclipse.e4.ui.workbench.renderers.swt.StackRenderer.access$0(StackRenderer.java:1236)
	at org.eclipse.e4.ui.workbench.renderers.swt.StackRenderer$2.mouseUp(StackRenderer.java:1076)
	at org.eclipse.swt.widgets.TypedListener.handleEvent(TypedListener.java:221)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:86)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:5686)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1370)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:4940)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:4518)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1170)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
[... cut ...]
	at org.eclipse.ui.internal.e4.compatibility.CompatibilityPart.destroy(CompatibilityPart.java:421)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:55)
	... 74 more
```

shroffk, 2018-10-11:
I think the exception makes sense... It is trying to close something that was never created.
The logging configuration relies on the efxclipse fx view, are any javafx views/components working?

shroffk, 2018-10-11:
I am guessing this is the latest version of the efxclipse version 3.X

willrogers, 2018-10-11:
The source code is here: https://github.com/eclipse/efxclipse-eclipse

Not so much as a readme.

willrogers, 2018-10-11:
There's not much on the website: https://www.eclipse.org/efxclipse/references.html

This blog post is very terse: https://www.eclipse.org/efxclipse/references.html

[I have a similar problem with OPIShellSummary.]

shroffk, 2018-10-11:
> I really struggle to find any documentation for efxclipse - do you know where it is?

No, I remember having the same experience...even  their apis had missing javadoc's
Anyway, I remember using their fxview example for debugging.

berryma4, 2018-10-14:
I should mention jython 2.7.2 should work under java 11 according to their
email list. But this version is not released yet. They don't have a date
yet.

kasemir, 2018-10-16:
Jython 2.7.1 works with Java 11, see https://github.com/shroffk/phoebus/issues/143#issuecomment-430293631

willrogers, 2018-11-06:
This proved to be more difficult than I had hoped. I won't be able put any more effort into this before Christmas at the earliest.

berryma4, 2019-01-02:
@willrogers 

I was able find a solution for this ... there are some important changes to the jaxb libraries that are hard to find information about. I found most information on the eclipse jaxb-ri github issue tickets and release notes. There are some oddities, like an issue ticket mentioned that versions with a b are for developers (ie. 2.4.0.b###). Release notes mentioned jaxb-core and jax-impl were merged.
So, I used:
jaxb-impl 2.3.1
jaxb-api 2.3.1

using 2.3.1, I needed to add -Djavax.xml.bind.context.factory=com.sun.xml.bind.v2.ContextFactory:

although, using the development 2.4.0.b### version, I was able to remove the property. So, we should be able to remove it when released.

After getting the versions straightened out, and many similar looking jars (jaxb-api, jaxb-impl, jaxb-core, jaxb-osgi, jaxb-runtime). I found jaxb-osgi had a depends on com.sun.source.tree in the jdk.compiler module, and the depends seemed endless, so I abandoned it. I read in an issue ticket that jaxb-runtime is for java se, while jaxb-impl for osgi. So, in the end the magic combination was jaxb-api and jaxb-impl.

There was still the classloading issue. I saw a clue from Hibernate:
https://github.com/hibernate/hibernate-orm/pull/2589/files
Which shows how you have to switch the ContextClassLoader, then set it back.

```java
final ClassLoader orig = Thread.currentThread().getContextClassLoader();
        Thread.currentThread().setContextClassLoader(JAXBContext.class.getClassLoader());
        try {
            JAXBContext context = JAXBContext.newInstance(DataSources.class.getPackageName(),
                    this.getClass().getClassLoader());
            Marshaller marshaller = context.createMarshaller();

            try (Writer writer = new FileWriter(new File(dsDir, DATASOURCES_FILE))) {

                writer.write("<?xml version=\"1.0\" encoding=\"UTF-8\" standalone=\"yes\" ?>\n");

                marshaller.setProperty(Marshaller.JAXB_FORMATTED_OUTPUT, true);
                marshaller.setProperty(Marshaller.JAXB_FRAGMENT, true);
                marshaller.marshal(this, writer);

            }
        } catch (JAXBException e) {
            throw new IOException(e);
        } finally {
            Thread.currentThread().setContextClassLoader(orig);
        }
``` 

lastly, istack-commons-runtime came up as a depends:
https://mvnrepository.com/artifact/com.sun.istack/istack-commons-runtime/3.0.7

And that's it!!!

I'll get a pull request in tomorrow for review.

willrogers, 2019-01-07:
That's great, thank you.

The other thing I was stuck on was getting JavaFX working inside Eclipse RCP using efxclipse. I really couldn't figure out what was going on and I couldn't find any useful documentation, but things may have improved in the past few months.

berryma4, 2019-01-11:
@willrogers so, it is definitely a "timing" thing for the "Workspace is closed" issue. If I turn on the osgi hook debug (-Defxclipse.osgi.hook.debug=true), I never get the error.

willrogers, 2019-01-22:
I am back full-time in a couple of weeks. I will try to put a bit more effort into this then. Hopefully we can get this pushed through.

willrogers, 2019-02-07:
This is a bit weird, but your branch won't compile because of the p2 location

```
http://downloads.efxclipse.bestsolution.at/p2-repos/openjfx-11/repository/repository
```

I can only find it at 

```
http://downloads.efxclipse.bestsolution.at/p2-repos/openjfx-11/repository
```

berryma4, 2019-02-07:
I think I need to push a commit. The efxclipse guy messed up a deployment,
but fixed it later. I'll push once I get in to work.

willrogers, 2019-02-07:
I've fixed it, I'll push once I've re-tested the build.

berryma4, 2019-02-07:
There might be a couple of changes.

berryma4, 2019-02-07:
I'll be at work in about an hour

willrogers, 2019-02-07:
The build succeeded on my machine but failed on Travis, I'm not sure why.

(Travis still returned a green tick though, I think that must be a mistake.)

willrogers, 2019-02-07:
It throws plenty of errors when I start it. From what you said yesterday perhaps this is expected and I should wait a couple of weeks for things to be fixed.

berryma4, 2019-02-07:
I pushed the commits, you should be all set

berryma4, 2019-02-07:
There were a couple of other changes you needed for it to work.

berryma4, 2019-02-07:
I had to update travis to use java11
https://github.com/ControlSystemStudio/org.csstudio.product/commit/ab268401c71ccc0cf6f7be485f26fdc0b988c89d

berryma4, 2019-02-07:
The job is timing out. I've emailed Travis for help. I noticed it is downloading a lot from maven, but I thought we were caching that directory.

The build should work for you though. Please post any issues. I should mention, when CS-Studio starts there are a lot of GTK3 warnings from Eclipse.

berryma4, 2019-02-08:
Sorry I just killed the build again, is an easy fix, I'm not sure why my ide didn't highlight it.

willrogers, 2019-02-08:
The changes I made:

```
diff --git a/applications/logbook/logbook-plugins/org.csstudio.logbook.olog.property.fault/src/org/csstudio/logbook/olog/property/fault/FaultConfigurationFactory.java b/applications/logbook/logbook-plugins/org.csstudio.logbook.olog.property.fault/src/org/csstudio/logbook/olog/property/fault/FaultConfigurationFactory.java
index d02167c140..474894d296 100644
--- a/applications/logbook/logbook-plugins/org.csstudio.logbook.olog.property.fault/src/org/csstudio/logbook/olog/property/fault/FaultConfigurationFactory.java
+++ b/applications/logbook/logbook-plugins/org.csstudio.logbook.olog.property.fault/src/org/csstudio/logbook/olog/property/fault/FaultConfigurationFactory.java
@@ -30,7 +30,7 @@ public class FaultConfigurationFactory {
         Thread.currentThread().setContextClassLoader(JAXBContext.class.getClassLoader());
         try {
             JAXBContext context = JAXBContext.newInstance(FaultConfiguration.class.getPackageName(),
-                    this.getClass().getClassLoader());
+                    FaultConfigurationFactory.class.getClassLoader());
             Unmarshaller um = context.createUnmarshaller();
             String file = "resources/default_fault_config.xml";
             URI filePath;
diff --git a/applications/pom.xml b/applications/pom.xml
index 7c2d093f9f..c6787329c6 100644
--- a/applications/pom.xml
+++ b/applications/pom.xml
@@ -53,7 +53,7 @@
       <properties>
         <eclipse-site>${eclipse.mirror.url}/releases/2018-12</eclipse-site>
         <eclipse-update-site>${eclipse.mirror.url}/eclipse/updates/4.10</eclipse-update-site>
-        <efx-site>${eclipse.mirror.url}/efxclipse/runtime-released/3.4.3/site</efx-site>
+        <efx-site>${eclipse.mirror.url}/efxclipse/runtime-nightly/site</efx-site>
       </properties>
     </profile>
     <profile>
diff --git a/core/pom.xml b/core/pom.xml
index 1290017901..ebe5908a8b 100644
--- a/core/pom.xml
+++ b/core/pom.xml
@@ -53,7 +53,7 @@
       <properties>
         <eclipse-site>${eclipse.mirror.url}/releases/2018-12</eclipse-site>
         <eclipse-update-site>${eclipse.mirror.url}/eclipse/updates/4.10</eclipse-update-site>
-        <efx-site>${eclipse.mirror.url}/efxclipse/runtime-released/3.4.3/site</efx-site>
+        <efx-site>${eclipse.mirror.url}/efxclipse/runtime-nightly/site</efx-site>
       </properties>
     </profile>
     <profile>
```

The first is I think the bug to which you were referring.

The others: why aren't you using the nightly efxclipse repo in the cs-studio repo?

After this it compiles and builds and starts and stops with no errors. As you might expect, the problems are related to JFX. If I open the logging configuration view (actually when I close it), I get the following exception:

```
2019-02-08 11:38:27.769 SEVERE [Thread 1] org.csstudio.logging.PluginLogListener (logging) - Problems occurred when invoking code from plug-in: "org.eclipse.e4.ui.workbench.swt".
org.eclipse.e4.core.di.InjectionException: java.lang.NullPointerException
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:68)
	at org.eclipse.e4.core.internal.di.InjectorImpl.processAnnotated(InjectorImpl.java:1001)
	at org.eclipse.e4.core.internal.di.InjectorImpl.processAnnotated(InjectorImpl.java:966)
	at org.eclipse.e4.core.internal.di.InjectorImpl.disposed(InjectorImpl.java:462)
	at org.eclipse.e4.core.internal.di.Requestor.disposed(Requestor.java:159)
	at org.eclipse.e4.core.internal.contexts.ContextObjectSupplier$ContextInjectionListener.update(ContextObjectSupplier.java:81)
	at org.eclipse.e4.core.internal.contexts.TrackableComputationExt.update(TrackableComputationExt.java:114)
	at org.eclipse.e4.core.internal.contexts.TrackableComputationExt.handleInvalid(TrackableComputationExt.java:77)
	at org.eclipse.e4.core.internal.contexts.EclipseContext.dispose(EclipseContext.java:187)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.clearContext(PartRenderingEngine.java:1006)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeRemoveGui(PartRenderingEngine.java:979)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.access$1(PartRenderingEngine.java:886)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$4.run(PartRenderingEngine.java:881)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:45)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.removeGui(PartRenderingEngine.java:865)
	at org.eclipse.e4.ui.workbench.renderers.swt.ElementReferenceRenderer.disposeWidget(ElementReferenceRenderer.java:115)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeRemoveGui(PartRenderingEngine.java:958)
[... 137 lines cut ...]
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:137)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:107)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:400)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
[... cut ...]
	at org.eclipse.ui.internal.e4.compatibility.CompatibilityPart.destroy(CompatibilityPart.java:424)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:58)
	... 75 more

```

berryma4, 2019-02-08:
Yes, I only had the nightly in the product build. This was because it's the runtime that needed the nightly, it really didn't matter what core and applications were compiled against, as long as the product pulls in the nightly.

Are you running your product, or the org.csstudio.product?

willrogers, 2019-02-08:
I'm running `org.csstudio.product`, I haven't tried to compile my product.

berryma4, 2019-02-08:
Does the jfx logging window come up fine? You just get the null pointer on
close? I did add a close in the logging UI code because of a change of
behavior in jfx 11, I wonder if efxclipse is now taking care of it.

willrogers, 2019-02-08:
No, the view is just blank.

berryma4, 2019-02-08:
Compiled, or from the ide (or both)? Could turn on the debugging for the swt classloader? It's in the product jvm properties in the product ini file, marked currently false.

willrogers, 2019-02-08:
Compiled, I basically never try to run from the IDE.

willrogers, 2019-02-08:
I get some more output, which typically looks like this:

```
FXClassLoader#findClassJavaFX11 - Loading class 'javafx.application.Platform' for org.eclipse.osgi.internal.loader.EquinoxClassLoader@6a0ae49b[org.csstudio.logging.ui:3.2.2.201902081109(id=91)]
FXClassLoader#findClassJavaFX11 - Using classloader jdk.internal.loader.Loader@5f68f882
FXClassLoader#findClassJavaFX11 - class javafx.application.Platform - ended
FXClassLoader#findClassJavaFX11 - started
FXClassLoader#findClassJavaFX11 - Loading class 'javafx.embed.swt.FXCanvas' for org.eclipse.osgi.internal.loader.EquinoxClassLoader@634e0da2[org.eclipse.fx.ui.workbench3:3.5.0.201902070600(id=274)]
FXClassLoader#findClassJavaFX11 - Using classloader jdk.internal.loader.Loader@5f68f882
FXClassLoader#findClassJavaFX11 - null - ended
FXClassLoader#findClassJavaFX11 - started
FXClassLoader#findClassJavaFX11 - Loading class 'jdk.internal.reflect.MethodAccessorImpl' for org.eclipse.osgi.internal.loader.EquinoxClassLoader@e3e3c3c[org.eclipse.e4.ui.bindings:0.12.300.v20180913-1535(id=193)]
FXClassLoader#findClassJavaFX11 - Using classloader jdk.internal.loader.Loader@5f68f882
FXClassLoader#findClassJavaFX11 - null - ended
```

I can't see anything that looks wrong, but I'm not sure what I'm looking for exactly.

berryma4, 2019-02-08:
Sorry I cleaned up a little too much:
https://github.com/ControlSystemStudio/org.csstudio.product/commit/4c19e1071223cec16d68d19635c41cc99d47a7db

JavaFX should be working for you now.

willrogers, 2019-02-11:
Amazing. This is all working as expected. It feels a bit more snappy on RHEL7 and the rendering problems I'm seeing in the latest versions of CS-Studio have gone as well. Perhaps this is because we are finally using GTK3.

Thank you for doing this work. What are the next steps? Presumably we want to merge these changes into master once we are happy that they are working for everyone.

berryma4, 2019-02-11:
If everything works for you too, I should clean up the commits on the
branch then do a pull request.

willrogers, 2019-02-22:
I have now adapted my product to build against these branches, and everything seems to be working well using the Tycho build.

I have one problem: when trying to launch using the IDE for development, it fails with a problem with `javax.annotation`. I don't know why because `javax.annotation` is in the target platform (the built product that is working).

```
!ENTRY org.eclipse.osgi 4 0 2019-02-22 09:24:35.371
!MESSAGE Application error
!STACK 1
org.eclipse.e4.core.di.InjectionException: java.lang.NoClassDefFoundError: javax/annotation/PostConstruct
	at org.eclipse.e4.core.internal.di.InjectorImpl.internalMake(InjectorImpl.java:422)
	at org.eclipse.e4.core.internal.di.InjectorImpl.make(InjectorImpl.java:333)
	at org.eclipse.e4.core.contexts.ContextInjectionFactory.make(ContextInjectionFactory.java:193)
	at org.eclipse.e4.ui.internal.workbench.swt.E4Application.createDefaultHeadlessContext(E4Application.java:498)
	at org.eclipse.e4.ui.internal.workbench.swt.E4Application.createDefaultContext(E4Application.java:512)
	at org.eclipse.e4.ui.internal.workbench.swt.E4Application.createE4Workbench(E4Application.java:211)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:586)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:339)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:563)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:151)
	at org.csstudio.utility.product.Workbench.runWorkbench(Workbench.java:99)
	at org.csstudio.startup.application.Application.startApplication(Application.java:265)
	at org.csstudio.startup.application.Application.start(Application.java:119)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:203)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:137)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:107)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:400)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:659)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:595)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1501)
	at org.eclipse.equinox.launcher.Main.main(Main.java:1474)
Caused by: java.lang.NoClassDefFoundError: javax/annotation/PostConstruct
	at org.eclipse.e4.core.internal.di.InjectorImpl.internalInject(InjectorImpl.java:139)
	at org.eclipse.e4.core.internal.di.InjectorImpl.internalMake(InjectorImpl.java:411)
	... 25 more
Caused by: java.lang.ClassNotFoundException: javax.annotation.PostConstruct cannot be found by org.eclipse.e4.core.di_1.7.100.v20180817-1215
	at org.eclipse.osgi.internal.loader.BundleLoader.findClassInternal(BundleLoader.java:460)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:422)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:414)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.loadClass(ModuleClassLoader.java:153)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:521)
	... 27 more

```

berryma4, 2019-02-22:
Make sure you have javax.annotation-api.

willrogers, 2019-02-22:
I can't see that in `maven-osgi-bundles` or `org.csstudio.product` but I can give it a go.

(There's `javax.activation-api` there).

willrogers, 2019-02-22:
Adding `javax.annotation-api` doesn't help.

berryma4, 2019-02-22:
Sorry, that's it. I really should have checked the name. But yes, make sure
that's selected in your run product. Also, adding Maven-osgi,
cs-studio-third-party, and eclipse, and efxclipse to your target, then add
required plugins. In the ide I have been running into an extra slf4j not
allowing the start up to get passed the splash screen. Deselecting the
extra plugin (version 15?), solves the issue.

berryma4, 2019-02-22:
Maybe we should wiki the instructions:
What you need to do to make this work:

  - install eclipse 2018-12
  - install efxclipse into the ide (be careful the wrong version is in eclipse, use the direct site version 3.5)
  - download the openjfx sdk, and put the path into preferences -> JavaFX, restart eclipse
  - pull in cs-studio your normal way

We should add what the "normal way" is. But I see it as putting the target together, clicking the run product from your product file, then going into the product run configuration and selecting add required plugins (then unselected the extra slj4).

willrogers, 2019-02-22:
I can't install efx into the IDE from the following site:

```
http://download.eclipse.org/efxclipse/runtime-nightly/site
```

I get a version conflict.

willrogers, 2019-04-09:
OK, the current state for me is:

* I can build the DLS product using the `issue_2530_EB` branches. It launches and it is snappy, I think because it is using GTK3.
* Ito working next to me has a strange compilation error that we can't figure out. I guess it must be something about his environment and as ever with Eclipse it's almost impossible to debug.
* I can't run from the IDE, because of the `javax.annotation` errors above.

The way that I set up the development environment is as follows. It's very simple and it has always worked for me in the past.

* Make a local Tycho build.
* Add that Tycho build as a target platform.
* Import DLS product as a Maven project.
* Run the DLS product from the run configurations menu.

Perhaps someone else could use their local build as a target platform and see if it works for them.

I am very keen to get these branches merged and make a 4.6 release at DLS.

cc @berryma4 @shroffk @ifmalcuaz

willrogers, 2019-04-09:
I've got exactly the same problem if I build the common product and import that into my IDE.

shroffk, 2019-04-09:
> I am very keen to get these branches merged and make a 4.6 release at DLS.

Me too.
Is building and running in the IDE a requirement for closing this PR.

willrogers, 2019-04-09:
Well, no let's get on and merge. Does everything work for you?

berryma4, 2019-04-10:
https://github.com/ControlSystemStudio/cs-studio-thirdparty/pull/75

Someone please okay this pull request

shroffk, 2019-04-10:
> Well, no let's get on and merge. Does everything work for you?

I can build on the command line on linux and windows.
The common product runs.
I can run the opi regression tests screens successfully

but...
I can't get the IDE set up
My CI server is complaining with various tycho errors (but this might be more of my responsibility)
I have not yet tested building the nsls2 product

[14 more replies]

closed 2019-04-26 by berryma4