ControlSystemStudio/cs-studio #2531: css crashes when converting an adl file

MarkRivers, 2018-10-05:
When I try to convert this adl file CSS crashes (I have added a .txt extension so Github would upload it).
[13IDD_LVP_T25.adl.txt](https://github.com/ControlSystemStudio/cs-studio/files/2450563/13IDD_LVP_T25.adl.txt)

This file contains the complete output of CSS when I try to convert that file:
[error.txt](https://github.com/ControlSystemStudio/cs-studio/files/2450590/error.txt)

This appears to be the fatal error:
```
Handling: text
org.jdom.IllegalNameException: The name "1" is not legal for JDOM/XML elements: XML names cannot begin with the character "1".
        at org.jdom.Element.setName(Element.java:207)
        at org.jdom.Element.<init>(Element.java:141)
        at org.jdom.Element.<init>(Element.java:153)
        at org.csstudio.opibuilder.properties.MacrosProperty.writeToXML(MacrosProperty.java:101)
        at org.csstudio.opibuilder.properties.ActionsProperty.writeToXML(ActionsProperty.java:146)
        at org.csstudio.opibuilder.persistence.XMLUtil.widgetToXMLElement(XMLUtil.java:96)
        at org.csstudio.opibuilder.persistence.XMLUtil.widgetToXMLElement(XMLUtil.java:104)
        at org.csstudio.opibuilder.persistence.XMLUtil.widgetToXMLString(XMLUtil.java:145)
        at org.csstudio.opibuilder.adl2boy.ADL2BOYApplication.convert(ADL2BOYApplication.java:111)
        at org.csstudio.opibuilder.adl2boy.ADL2BOYApplication.checkThenConvert(ADL2BOYApplication.java:103)
        at org.csstudio.opibuilder.adl2boy.ADL2BOYApplication.start(ADL2BOYApplication.java:66)
        at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:196)
        at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
        at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
        at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:388)
        at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:243)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.lang.reflect.Method.invoke(Method.java:498)
        at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:673)
        at org.eclipse.equinox.launcher.Main.basicRun(Main.java:610)
        at org.eclipse.equinox.launcher.Main.run(Main.java:1519)
        at org.eclipse.equinox.launcher.Main.main(Main.java:1492)
make: *** [opi/autoconvert/13IDD_LVP_T25.opi] Error 255
```

kasemir, 2018-10-05:
When run within the CSS GUI, I get this error dialog which explains the issue:

[image]

The ADL file contains sections like these:
```
"related display" {
...	display[0] {
		label="ROIs 1-8"
		name="mca_ROIs.adl"
		args="P=13IDD:,M=aim_adc1,1=1,2=2,3=3,4=4,5=5,6=6,7=7,8=8"
```

The BOY XML file stores macros in XML as
```
<NameOfMacro>The Value</NameOfMacro>
```
Macro names thus need to be valid XML tag names, which "1" is not.

In principle it would be possible to update the converter to add some prefix like "M" to all invalid macro names. That's somewhat easy for those sections that set macro values, but very involved when it comes to finding all the places that can potentially use macro names.

From the CSS standpoint the easiest solution is indeed updating the ADL files to use macro names that are not numbers, as indicated in the dialog.

kasemir, 2018-10-05:
Found a  fix: When reading macros, it patches numeric macro names by adding a prefix.
When evaluating macros, it does the same.
I'm undecided if that's a great fix, or a terrible hack.

closed 2018-10-08 by kasemir