ControlSystemStudio/cs-studio #2541: BOY jython path handling changed

kasemir, 2018-11-09:
This idea is including `some/path/script1.py`, and that script in turn includes other files from the same `some/path` .

In a workspace location, create sub-folder 'scripts' with two scripts:
```
# scripts/sub.py
def do_it(widget):
    widget.setPropertyValue("text", "Hello from python!")
```
and
```
# scripts/main.py
import sys
print "Path:\n" + "\n".join(sys.path)

from sub import do_it
do_it(widget)
```

In the parent directory, create a display with a Label.

Add a script to that label, calling `scripts/main.py`, triggered by `loc://init(1)`.

Execute that display, and label should show "Hello from python!"

Not sure when it changed, but this worked from the beginning of BOY, including versions built from 'master' at the end of July. Also works fine with the Display Builder.

But fails with BOY built right now:
```
ERROR: Error in scripts/main.py on Label.
..
ImportError: No module named sub
```

In the successful case, the path printout in `main.py` includes the scripts folder:
```
Path:
/home/ky9/Eclipse/CSS/plugins/org.python.jython_2.7.0.release.jar/Lib
/home/ky9/Eclipse/CSS/Lib
__classpath__
__pyclasspath__/
/home/ky9/Eclipse/runtime-sns-css.product/CSS/script_issue/scripts
```

With the current version of BOY, the path lacks the script directory:
```
Path:
/home/ky9/git/css_target/plugins/org.python.jython_2.7.1.release.jar/Lib
/home/ky9/Eclipse/4.7.3a/Lib
__classpath__
__pyclasspath__/
```

kasemir, 2018-11-09:
No obvious change in the BOY source code, but Jython was updated from 2.7.0 to 2.7.1 #2528 
Maybe this is new jython behavior, it no longer automatically searches for imports in the location of the current script?

Display builder has always added the path for the invoked script to the path, so never depended on this.

closed 2018-11-12 by berryma4

John-Holt-Tessella, 2019-05-15:
For reference, problem is in jython https://github.com/jythontools/jython/commit/c7a3ebac42ba2efb6f235c0ec4a72c048d925fb3

PR #2542 (2018-11-12): BOY: Add path to Jython scripts to search path
Fixes #2541
commit ab3e1702 (2018-11-12): BOY: Add path to Jython scripts to search path

Fixes #2541