ControlSystemStudio/cs-studio #2553: Archive Engine: Nanoseconds Incorrect

DominicOram, 2018-12-19:
At ISIS we've had some issues with the timestamp of the data being written by the archive engine:

[image]

Specifically it looks like the nanosecond field is being written to incorrectly (for more details see https://github.com/ISISComputingGroup/IBEX/issues/3779). We are using Archive Engine 4.0.0.201806141728 and mySQL 5.7

@kasemir have you seen this issue before?

kasemir, 2018-12-19:
Nadine at ITER had noticed this.
The Java Timestamp has nanosecond accuracy.

With Oracle, we simply use JDBC `statement.setTimestamp(..)` to set the full timestamp because the TIMESTAMP column in the database can handle it.

With Postgres and MySQL, the TIMESTAMP column is only good to seconds, and we set the nanoseconds to a separate nanosecs column. Nadine noticed, however, that something rounds the seconds, so we need to truncate the seconds before writing them.

On 2018-10-19 she emailed this diff that she had tested:

```
    private void completeAndBatchInsert(
            final PreparedStatement insert_xx, final RDBWriteChannel channel,
            final Timestamp stamp, final int severity,
            final Status status) throws Exception
    {
        // Set the stuff that's common to each type
        insert_xx.setInt(1, channel.getId());
        insert_xx.setTimestamp(2, stamp);
        insert_xx.setInt(3, severity);
        insert_xx.setInt(4, status.getId());
-       // MySQL nanosecs
-       if (rdb.getDialect() == Dialect.MySQL  ||  rdb.getDialect() == Dialect.PostgreSQL)
 
+       if (rdb.getDialect() == Dialect.Oracle)
+            insert_xx.setTimestamp(2, stamp);
+        else
+        {
+            // Truncate the time stamp
+            Timestamp truncated = Timestamp.from(stamp.toInstant());
+            truncated.setNanos(0);
+            insert_xx.setTimestamp(2, truncated);
+            // Set the nanos in a separate column
            insert_xx.setInt(6, stamp.getNanos());
        }
```

DominicOram, 2018-12-19:
Great, thanks. I'll test that my end and, assuming it works, make a PR for it.

kasemir, 2018-12-19:
Looking at your IBEX link, you had already concluded the issue is related to rounding instead of truncating the time stamp detail, "The best fix is probably to truncate it as we put the data into the archive", which is exactly what Nadine had also found. So this should work.

closed 2019-04-26 by berryma4