ControlSystemStudio/cs-studio #2555: use the comm.jar in python script

xlgforever, 2019-01-04:
i use the comm.jar in the eclipse java IDE to communicate with another device by UART. it works well. but when i use it in python script which is attached to a widget in opi file in CSS, i can't import the modules in the jar file.  i check the jython interpreter of PyDev, it shows that the comm.jar is in the system libs. and when i run jython.exe on the same computer, then i can import the modules. it seems that when i use jython in CSS, it can not find the jar files. what should i do?

kasemir, 2019-01-07:
The Jython path used by BOY  scripts is controlled by
```
org.csstudio.opibuilder/python_path
```

But reconsider talking to a UART from a display widget.
In an EPICS setup, IOCs do the work. IOCs talk to devices (via a UART or network or ...).
The display, well, simply displays the value of PVs and allows writing values to PVs.
The display doesn't _do_ anything.

If you talk to a UART from a script, while your script waits for a reply from the UART, the complete UI freezes. That's a great way to make friends.

Is the script that talks to your UART performing some useful control system functionality?
Then closing the display will stop that functionality. That's why all the work should be done by IOCs, so closing a display doesn't matter at al.
Conversely, opening multiple displays should be fine. In your case, would several users opening CSS to run your display result in N scripts talking to the device at the same time?

xlgforever, 2019-01-07:
> The Jython path used by BOY scripts is controlled by
>
> ```
> org.csstudio.opibuilder/python_path
> ```
> But reconsider talking to a UART from a display widget.
> In an EPICS setup, IOCs do the work. IOCs [...]

the reason that i use python script in CSS is that i want to monitor some PVs and when the PVs' value is smaller or bigger than some constants, the python script will convert these PVs' value into corresponding binary data according to some certain rules. And then it will send these data(very little data) to the device by UART. Maybe after seconds, it will receive some binary data from the device and convert the binary data into decimal numbers according to some rules. finally it will use these decimal numbers to wirte the PVs' value.

i have cost some time to implement the functionally successfully in python IDE by a python module named serial. i check the source code of the module, it uses different submodule to implement the communication functionality in different platform(by using the "os.name" function to select which submodule to use). when i use the serial module in python IDE, the os.name is "nt",and it become into "java" in CSS.

 So if i want to use the serial module in "java" platform, the source code of serial module shows that i should implement the JAVA Communication API. the comm.jar and the RXTXcomm.jar are the useful java serial communication API. i download and intall them and test them in eclipse java IDE, it works well. 

The most confusing question is that when i open "cmd" in the path of the "jython.jar" file, and then execute "java -jar jython.jar" to  open the jython, then excute "from gnu.io import *" , it works correctly(gnu.io is the package in the RXTXcomm.jar). but when i execute "from gnu.io import *" in python script of CSS, it shows "import error: no module named gnu". i am sure that the two jar files are added into the system lib of the "interpreter" of jython of the PyDev plug-in. and the jar files are in %JAVA_HOME%\jre\lib\ext(the eclipse JAVA IDE uses the two files in this folder according to my test).

i attempt to add the path of the jar files into the PYTHONPATH of CSS' preference, it is invalid. So it seems that the CSS cannot use the jar files in %JAVA_HOME%\jre\lib\ext even though i add the path into the interpreter's system lib of PyDev. I really wish that someone could help me!o(╥﹏╥)o

xlgforever, 2019-02-16:
> If you want to use python to communicate with devices, that's fine.
> Run that as a separate program, using C-Python, independent from the display, just like an IOC.
> Use the PCASPY library, https://pcaspy.readthedocs.io/en/latest/, so serve PVs, so now your python program is basically just like an IOC.
> 
> Then use the display to show those PVs.

thank you fof your advice. and I have learned about the PCASPY library. I have 2 questions. 

1. If i use the CSS, I could read and write the PVs in other IOCs of other computers, Just make sure they're on the same network. if i use the PCASPY library in python, can i do the same thing?

2.i notice that there are some command lines in tutorial of PCASPY library, which  begin with "$", such as
```
$ camonitor MTEST:RAND
MTEST:RAND                     2011-07-19 12:32:06.574775 0.646198
MTEST:RAND                     2011-07-19 12:32:07.574704 0.872313
MTEST:RAND                     2011-07-19 12:32:08.581681 0.171537
MTEST:RAND                     2011-07-19 12:32:09.581581 0.351235
```
when i learn about the library, i can not excute the command begined with"$"(i excute it in python on a windows PC), maybe because i am not familiar with Linux? 

look forward to your reply!

kasemir, 2019-02-18:
1. pcaspy PVs are on the network like PVs from IOCs

2. The '$' is the shell prompt. You don't type it. At the end of the line, you do need to press "Enter", even though it does not spell that out. See https://www.shellscript.sh/first.html

xlgforever, 2019-02-19:
> 1. pcaspy PVs are on the network like PVs from IOCs
> 2. The '$' is the shell prompt. You don't type it. At the end of the line, you do need to press "Enter", even though it does not spell that out. See https://www.shellscript.sh/first.html

oh, thank you. but i ask the author of pcaspy for how to use pcaspy to read/write PVs in other IOCs in the same LAN, he told me that pcaspy can only read/write the PVs in local IOC unless I explicitly use channel access to connect to other IOCs. accordin to my understand, pcaspy seems to only implement the ca server but no ca client(so i cannot explicitly connet to other IOCs?). Does this mean that PVS in other IOC in the same LAN cannot be accessed through the pcaspy module?

kasemir, 2019-02-19:
pcaspy is a CA server, that's the 'cas' part of the name.
It can serve PVs and react when some other client writes to these PVs.

Nothing stops you from concurrently using a CA client library like PyEpics (http://cars9.uchicago.edu/software/python/pyepics3/overview.html) in the same python code, and you can then use that to read/write PVs servered by other CA servers, which may include pcaspy or plain IOCs.

When you use both pcaspy and PyEpics you create both a CA server and a CA client, just like an IOC.
When doing this, read the PyEpics manual, especially note the CAThread and http://cars9.uchicago.edu/software/python/pyepics3/advanced.html#advanced-threads-label section.

xlgforever, 2019-02-25:
> pcaspy is a CA server, that's the 'cas' part of the name.
> It can serve PVs and react when some other client writes to these PVs.
> 
> Nothing stops you from concurrently using a CA client library like PyEpics (http://cars9.uchicago.edu/software/python/pyepics3/overview.html) in the same python code, and you can then use that to read/write PVs served by other CA servers, which may include pcaspy or plain IOCs.
> 
> When you use both pcaspy and PyEpics you create both a CA server and a CA client, just like an IOC.
> When doing this, read the PyEpics manual, especially note the CAThread and http://cars9.uchicago.edu/software/python/pyepics3/advanced.html#advanced-threads-label section.

thanks，now, i know that pyepics can r/w the PVs, if i want to display or control the PVs in python just like in CSS, is there any recommended GUI modules of python to use?

closed 2019-04-26 by berryma4