ControlSystemStudio/cs-studio #2556: RAP GEF p2 site

willrogers, 2019-01-22:
The rap gef url is a bit unusual. There seems to be a more standard one available:

```
-        <rap-gef-site>http://archive.eclipse.org/rt/rap/incubator/nightly/gef/20150327-1849</rap-gef-site>
+        <rap-gef-site>http://download.eclipse.org/rt/rap/incubator/3.0/gef/20150327-1849</rap-gef-site>
```

Could we swap?

@berryma4 @ralphlange

ralphlange, 2019-01-23:
Sure. The time stamps are the same, and `3.0` sounds a bit more standard and stable than `nightly` - the advantage might be purely psychological, though.

willrogers, 2019-01-23:
I'm pretty sure it's the same stuff and it will hang around, it's just that all the rest of our mirroring comes from `http://download.eclipse.org` so it's helpful to have this one as well.

berryma4, 2019-04-26:
we have removed rap from the build, closing.

closed 2019-04-26 by berryma4