ControlSystemStudio/cs-studio #2561: Install cs-studio

freeman892, 2019-03-03:
Hi,

I am wondering how one would install and run cs-studio under MacOS. 
Are additional steps required beyond downloading the software at 
http://download.controlsystemstudio.org/release/4.5/cs-studio-4.5.9-macosx.cocoa.x86_64.zip?

After several tries (different cs-studio version, lates Java runtime and JDK), I am always stuck with a crash directly after selecting the project dictionary.
("An error has occurred.")
I observed the same behavior under macOS and Linux, so it seems that I do something wrong.
But from the information on the website I can not figure out what.

Does somewhere exist a system requirement info?
I am using 
macOS 10.14.3 
Java 8 Update 201

Any ideas? 
Does somewhere exist a user guide for the installation, that you could maybe point me to?
(Sorry, for the stupid questions.)

For example if I try to run CS-Studio 4.5.9 I get "An error has occurred!"
The .css/.metadata/.log has the following content:
!SESSION 2019-03-03 13:23:23.625 -----------------------------------------------
eclipse.buildId=unknown
java.version=11.0.2
java.vendor=Oracle Corporation
BootLoader constants: OS=macosx, ARCH=x86_64, WS=cocoa, NL=de_DE
Framework arguments:  -keyring /Users/xxxx/.eclipse_keyring
Command-line arguments:  -os macosx -ws cocoa -arch x86_64 -keyring /Users/xxxx/.eclipse_keyring

!ENTRY org.eclipse.equinox.ds 1 0 2019-03-03 13:23:25.057
!MESSAGE Could not bind a reference of component org.csstudio.diirt.util.RegisterDatasource. The reference is: Reference[name = DataSourceProvider, interface = org.diirt.datasource.DataSourceProvider, policy = dynamic, cardinality = 0..n, target = null, bind = registerDatasource, unbind = deregisterDatasource]

!ENTRY org.eclipse.equinox.ds 1 0 2019-03-03 13:23:25.059
!MESSAGE Could not bind a reference of component org.csstudio.diirt.util.RegisterFormulaFunctionSet. The reference is: Reference[name = FormulaFunctionSet, interface = org.diirt.datasource.formula.FormulaFunctionSet, policy = dynamic, cardinality = 0..n, target = null, bind = registerFormulaFunctionSet, unbind = deregisterFormulaFunctionSet]

!ENTRY org.eclipse.equinox.ds 1 0 2019-03-03 13:23:25.061
!MESSAGE Could not bind a reference of component org.csstudio.diirt.util.RegisterService. The reference is: Reference[name = ServiceProvider, interface = org.diirt.service.ServiceProvider, policy = static, cardinality = 0..n, target = null, bind = registerService, unbind = null]

!ENTRY org.eclipse.core.resources 2 10035 2019-03-03 13:23:29.628
!MESSAGE The workspace exited with unsaved changes in the previous session; refreshing workspace to recover changes.

!ENTRY org.eclipse.osgi 4 0 2019-03-03 13:23:29.770
!MESSAGE Application error
!STACK 1
org.eclipse.e4.core.di.InjectionException: java.lang.NoClassDefFoundError: javax/annotation/PostConstruct
	at org.eclipse.e4.core.internal.di.InjectorImpl.internalMake(InjectorImpl.java:386)
	at org.eclipse.e4.core.internal.di.InjectorImpl.make(InjectorImpl.java:294)
	at org.eclipse.e4.core.contexts.ContextInjectionFactory.make(ContextInjectionFactory.java:162)
	at org.eclipse.e4.ui.internal.workbench.swt.E4Application.createDefaultHeadlessContext(E4Application.java:490)
	at org.eclipse.e4.ui.internal.workbench.swt.E4Application.createDefaultContext(E4Application.java:504)
	at org.eclipse.e4.ui.internal.workbench.swt.E4Application.createE4Workbench(E4Application.java:203)
	at org.eclipse.ui.internal.Workbench$5.run(Workbench.java:632)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:610)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:148)
	at org.csstudio.utility.product.Workbench.runWorkbench(Workbench.java:99)
	at org.csstudio.startup.application.Application.startApplication(Application.java:265)
	at org.csstudio.startup.application.Application.start(Application.java:119)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:196)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:388)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:243)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:673)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:610)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1519)
Caused by: java.lang.NoClassDefFoundError: javax/annotation/PostConstruct
	at org.eclipse.e4.core.internal.di.InjectorImpl.inject(InjectorImpl.java:151)
	at org.eclipse.e4.core.internal.di.InjectorImpl.internalMake(InjectorImpl.java:375)
	... 24 more
Caused by: java.lang.ClassNotFoundException: javax.annotation.PostConstruct cannot be found by org.eclipse.e4.core.di_1.6.1.v20160712-0927
	at org.eclipse.osgi.internal.loader.BundleLoader.findClassInternal(BundleLoader.java:410)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:372)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:364)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.loadClass(ModuleClassLoader.java:161)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:521)
	... 26 more

!ENTRY org.eclipse.e4.ui.workbench 4 0 2019-03-03 13:23:29.775
!MESSAGE FrameworkEvent ERROR
!STACK 0
java.lang.NoClassDefFoundError: javax/annotation/PreDestroy
	at org.eclipse.e4.core.internal.di.InjectorImpl.disposed(InjectorImpl.java:426)
	at org.eclipse.e4.core.internal.di.Requestor.disposed(Requestor.java:154)
	at org.eclipse.e4.core.internal.contexts.ContextObjectSupplier$ContextInjectionListener.update(ContextObjectSupplier.java:78)
	at org.eclipse.e4.core.internal.contexts.TrackableComputationExt.update(TrackableComputationExt.java:111)
	at org.eclipse.e4.core.internal.contexts.TrackableComputationExt.handleInvalid(TrackableComputationExt.java:74)
	at org.eclipse.e4.core.internal.contexts.EclipseContext.dispose(EclipseContext.java:176)
	at org.eclipse.e4.core.internal.contexts.osgi.EclipseContextOSGi.dispose(EclipseContextOSGi.java:106)
	at org.eclipse.e4.core.internal.contexts.osgi.EclipseContextOSGi.bundleChanged(EclipseContextOSGi.java:139)
	at org.eclipse.osgi.internal.framework.BundleContextImpl.dispatchEvent(BundleContextImpl.java:903)
	at org.eclipse.osgi.framework.eventmgr.EventManager.dispatchEvent(EventManager.java:230)
	at org.eclipse.osgi.framework.eventmgr.ListenerQueue.dispatchEventSynchronous(ListenerQueue.java:148)
	at org.eclipse.osgi.internal.framework.EquinoxEventPublisher.publishBundleEventPrivileged(EquinoxEventPublisher.java:213)
	at org.eclipse.osgi.internal.framework.EquinoxEventPublisher.publishBundleEvent(EquinoxEventPublisher.java:120)
	at org.eclipse.osgi.internal.framework.EquinoxEventPublisher.publishBundleEvent(EquinoxEventPublisher.java:112)
	at org.eclipse.osgi.internal.framework.EquinoxContainerAdaptor.publishModuleEvent(EquinoxContainerAdaptor.java:156)
	at org.eclipse.osgi.container.Module.publishEvent(Module.java:476)
	at org.eclipse.osgi.container.Module.doStop(Module.java:634)
	at org.eclipse.osgi.container.Module.stop(Module.java:498)
	at org.eclipse.osgi.container.SystemModule.stop(SystemModule.java:202)
	at org.eclipse.osgi.internal.framework.EquinoxBundle$SystemBundle$EquinoxSystemModule$1.run(EquinoxBundle.java:165)
	at java.base/java.lang.Thread.run(Thread.java:834)
Caused by: java.lang.ClassNotFoundException: javax.annotation.PreDestroy cannot be found by org.eclipse.e4.core.di_1.6.1.v20160712-0927
	at org.eclipse.osgi.internal.loader.BundleLoader.findClassInternal(BundleLoader.java:410)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:372)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:364)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.loadClass(ModuleClassLoader.java:161)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:521)
	... 21 more

kasemir, 2019-03-04:
Hi:

> I am using

> Java 8 Update 201

No, you are in fact using Java 11:

> java.version=11.0.2

.. and then it fails because Java FX is not found:

> org.eclipse.e4.core.di.InjectionException: java.lang.NoClassDefFoundError: javax/annotation/PostConstruct

Install the Oracle 8 JDK and assert that it's used.

See https://controlssoftware.sns.ornl.gov/css_rcp/issues.html

-Kay

freeman892, 2019-03-04:
Thank you! for pointing this out so swiftly.
After going a bit down the rabbit hole, I am still stuck. 
At first, I thought my previous installation of the java 8 JDK did not work properly.
While the "Java Control Panel" did show Java 8 this seems not to be reliable.
Indeed `javac -version` did still report java 11.
I followed [Install java 8 on Mojave](https://stackoverflow.com/questions/53772023/how-to-install-java-8-on-osx-macos-mojave-with-homebrew) but got the same.

So maybe the variable are not set properly (not a feature at all?).
First I did add 
`export JAVA_HOME=$(/usr/libexec/java_home -v 1.8)`
to my bash_profile by hand. 
Afterwards I got the following
`java -version`
`java version "1.8.0_202"`
and
`javac -version`
`javac 1.8.0_202`.
Still I have the same problem. cs-studio crashes after setting the project directory.
Especially, cs-studio still reports java 11!?

> !SESSION 2019-03-04 21:51:30.407 -----------------------------------------------
> eclipse.buildId=unknown
> 
> java.version=11.0.2
> java.vendor=Oracle Corporation
> BootLoader constants: OS=macosx, ARCH=x86_64, WS=cocoa, NL=de_DE
> ...
After this, I tried to use [jenv](http://www.jenv.be) to manage my java installations and environment variables.
Again `java -version` and `javac -version` report java 8.
Is there something else that I am missing?

Btw. 
`echo $JAVA_HOME`
`/usr/local/opt/jenv/versions/oracle64-1.8.0.202`

berryma4, 2019-03-04:
I believe you can use:
-vm "/your_path/2/Java/jdk1.8.0_07\bin\javaw"
To use a specific vm.

kasemir, 2019-03-04:
In principle, Mac should be the same as the others, the css.ini is just in a different place:
```
cd Css.app/Contents/Eclipse
vi css.ini
```
and add
```
-vm
/usr/local/opt/jenv/versions/oracle64-1.8.0.202/bin/java
```

Or run it like this:
```
Css.app/Contents/MacOS/css -vm /usr/local/opt/jenv/versions/oracle64-1.8.0.202/bin/java
```

But I have to admit that doesn't work for me, either.
I get "Error: could not find libjava.dylib" or "Alert: Failed to create the Java Virtual Machine".

kasemir, 2019-03-04:
This worked:
Edit `Css.app/Contents/Eclipse.app/Contents/Info.plist`.

It contains a section like this:
```
<key>Eclipse</key>
                <array>
                        <!-- to use a specific Java version (instead of the platform's default) uncomment one of the following options,
                                        or add a VM found via $/usr/libexec/java_home -V
                                <string>-vm</string><string>/System/Library/Java/JavaVirtualMachines/1.6.0.jdk/Contents/Commands/java</string>
                        -->
```

So in there, make it
```
 <key>Eclipse</key>
                <array>
                        <string>-vm</string><string>/usr/local/opt/jenv/versions/oracle64-1.8.0.202/bin/java</string>
```

freeman892, 2019-03-04:
Thank you both! 
The last comment did indeed work like a charm.
I added 
`<key>Eclipse</key>
                <array>
                        <string>-vm</string><string>/usr/local/opt/jenv/versions/oracle64-1.8.0.202/bin/java</string>`
to cs-studio.app/Contents/Info.plis, css now gets the proper java 8, and starts correctly. 
Thank you!

claudio-rosati, 2019-03-05:
This is my `.ini` file:

```
-startup
../Eclipse/plugins/org.eclipse.equinox.launcher_1.4.0.v20161219-1356.jar
--launcher.library
../Eclipse/plugins/org.eclipse.equinox.launcher.cocoa.macosx.x86_64_1.1.551.v20171108-1834
-data
@user.home/.ess-cs-studio
-share_link
/Users/claudiorosati/Projects/GitHubX/ess-symbols=/ess-symbols,/Users/claudiorosati/Projects/GitHubX/ess-templates=/ess-templates
-vm
/Library/Java/JavaVirtualMachines/jdk1.8.0_202.jdk/Contents/Home/jre/bin
-vmargs
-Xmx2g
-Xms256m
-XX:+UseParallelGC
-Dorg.eclipse.update.reconcile=false
-Dosgi.configuration.cascaded=true
-Dosgi.sharedConfiguration.area.readOnly=true
-Dosgi.install.area.readOnly=true
-Dosgi.requiredJavaVersion=1.8
-Dorg.osgi.framework.bundle.parent=ext
-Dosgi.framework.extensions=org.eclipse.fx.osgi
-Dpython.cachedir.skip=true
-Declipse.p2.unsignedPolicy=allow
-XstartOnFirstThread
-Dorg.eclipse.swt.internal.carbon.smallFonts
```

As you can see:

- `-vm …` must be placed just before the `-vmargs`,
- The path should point to the `bin` folder of the java distribution.

This is what Eclipse tell to force Eclipse itself to run on a specific Java version.

shroffk, 2019-04-23:
@freeman892 can this issue ticket be closed?

closed 2019-04-23 by freeman892