ControlSystemStudio/cs-studio #2565: Error reading help table of contents file

claudio-rosati, 2019-03-15:
Hello all,

I continue to have this error when I start CS-Studio:

```
2019-03-15 09:30:20.449 SEVERE [Thread 165] org.csstudio.logging.PluginLogListener (logging) - Error reading help table of contents file /"org.csstudio.archive.influxdb/tocChangelog.xml" (skipping file)
java.io.FileNotFoundException
	at org.eclipse.help.internal.toc.TocFileParser.parse(TocFileParser.java:44)
	at org.eclipse.help.internal.toc.TocFileProvider.getTocContributions(TocFileProvider.java:45)
	at org.eclipse.help.internal.toc.TocManager.getAndCacheTocContributions(TocManager.java:211)
	at org.eclipse.help.internal.toc.TocManager.getTocContributionsForToc(TocManager.java:199)
	at org.eclipse.help.internal.toc.TocManager.getRootTocContributions(TocManager.java:307)
	at org.eclipse.help.internal.toc.TocManager.getTocs(TocManager.java:68)
	at org.eclipse.help.internal.toc.TocManager.getTopic(TocManager.java:123)
	at org.eclipse.help.internal.toc.TocManager.getTopicPath(TocManager.java:138)
	at org.eclipse.help.internal.webapp.data.UrlUtil.getTopicPath(UrlUtil.java:237)
	at org.eclipse.help.internal.webapp.servlet.BreadcrumbsFilter.filter(BreadcrumbsFilter.java:78)
	at org.eclipse.help.internal.webapp.servlet.EclipseConnector.transfer(EclipseConnector.java:197)
	at org.eclipse.help.internal.webapp.servlet.ContentServlet.doGet(ContentServlet.java:47)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:687)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:790)
	at org.eclipse.equinox.http.registry.internal.ServletManager$ServletWrapper.service(ServletManager.java:180)
	at org.eclipse.equinox.http.servlet.internal.HttpServiceRuntimeImpl$LegacyServlet.service(HttpServiceRuntimeImpl.java:1223)
	at org.eclipse.equinox.http.servlet.internal.registration.EndpointRegistration.service(EndpointRegistration.java:148)
	at org.eclipse.equinox.http.servlet.internal.servlet.ResponseStateHandler.processRequest(ResponseStateHandler.java:62)
	at org.eclipse.equinox.http.servlet.internal.context.DispatchTargets.doDispatch(DispatchTargets.java:131)
	at org.eclipse.equinox.http.servlet.internal.servlet.ProxyServlet.service(ProxyServlet.java:74)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:790)
	at org.eclipse.equinox.http.jetty.internal.HttpServerManager$InternalHttpServiceServlet.service(HttpServerManager.java:284)
	at org.eclipse.jetty.servlet.ServletHolder.handle(ServletHolder.java:860)
	at org.eclipse.jetty.servlet.ServletHandler.doHandle(ServletHandler.java:535)
	at org.eclipse.jetty.server.handler.ScopedHandler.nextHandle(ScopedHandler.java:188)
	at org.eclipse.jetty.server.session.SessionHandler.doHandle(SessionHandler.java:1595)
	at org.eclipse.jetty.server.handler.ScopedHandler.nextHandle(ScopedHandler.java:188)
	at org.eclipse.jetty.server.handler.ContextHandler.doHandle(ContextHandler.java:1253)
	at org.eclipse.jetty.server.handler.ScopedHandler.nextScope(ScopedHandler.java:168)
	at org.eclipse.jetty.servlet.ServletHandler.doScope(ServletHandler.java:473)
	at org.eclipse.jetty.server.session.SessionHandler.doScope(SessionHandler.java:1564)
	at org.eclipse.jetty.server.handler.ScopedHandler.nextScope(ScopedHandler.java:166)
	at org.eclipse.jetty.server.handler.ContextHandler.doScope(ContextHandler.java:1155)
	at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:141)
	at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:132)
	at org.eclipse.jetty.server.Server.handle(Server.java:530)
	at org.eclipse.jetty.server.HttpChannel.handle(HttpChannel.java:347)
	at org.eclipse.jetty.server.HttpConnection.onFillable(HttpConnection.java:256)
	at org.eclipse.jetty.io.AbstractConnection$ReadCallback.succeeded(AbstractConnection.java:279)
	at org.eclipse.jetty.io.FillInterest.fillable(FillInterest.java:102)
	at org.eclipse.jetty.io.ChannelEndPoint$2.run(ChannelEndPoint.java:124)
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.doProduce(EatWhatYouKill.java:247)
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.produce(EatWhatYouKill.java:140)
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.run(EatWhatYouKill.java:131)
	at org.eclipse.jetty.util.thread.ReservedThreadExecutor$ReservedThread.run(ReservedThreadExecutor.java:382)
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:708)
	at org.eclipse.jetty.util.thread.QueuedThreadPool$2.run(QueuedThreadPool.java:626)
	at java.lang.Thread.run(Thread.java:748)
```

Any idea of how to get rid of it?

kasemir, 2019-03-15:
plugin tries to add a help file table-of-contents that doesn't exist.
Pull request removes it.

claudio-rosati, 2019-03-15:
Thank you very much.

PR #2566 (2019-03-15): Remove erroneous toc
Fixes #2565
closed 2019-03-15 by claudio-rosati