ControlSystemStudio/cs-studio #2567: AlarmConfigTool -send_jms_config fails

MichaelRitzert, 2019-03-20:
In 945aa221f69420fc823113c74bf4f1fb7dcd062c, the class JMSSender is introduced that takes the JMS server URL in its constructor.
In Application.java, the call is
```
                final JMSSender sender = new JMSSender(this.url, this.user,
                    this.password, this.root+"_CLIENT");
```
where
```
        final StringOption url = new StringOption(parser, "-rdb_url",
               "Alarm config database URL", Preferences.getRDB_Url());
```

So the RDB URL is passed as the JMS server URL. Accordingly, I get
```
Error sending JMS CONFIG message:                                                    
javax.jms.JMSException: Could not create Transport. Reason: java.io.IOException: Transport scheme NOT recognized: [jdbc]
```

kasemir, 2019-03-20:
Well, you're correct, it's using the RDB URL as the JMS URL.
But consider the use case:
This is for the `"-send_jms_config` command line option. So you can invoke it like this:
```
AlarmConfigTool -rdb_url THE_JMS_URL -send_jms_config
```

So in practice, this "works".
Still, you're correct, it might be more obvious to have a separate `-jms_url` option/preference/default value. Do you want to implement that and send a pull request?

MichaelRitzert, 2019-03-20:
I considered going for a fix myself, but didn't go the obvious way when I saw this in the log:
```
 org.apache.activemq.transport.failover.FailoverTransport (doReconnect) - Successfully connected to tcp://pxdioc1:61616
```
So the correct JMS URL I configured via
```
org.csstudio.alarm.beast/jms_url=failover:(tcp://pxdioc1:61616)
```
is available and already used in another place. Maybe it would be better to use the information from there.

kasemir, 2019-03-20:
So you use the existing `Preferences.getJMS_URL()` as a default, offering a new `-jms_url` option to fetch/adjust it.

MichaelRitzert, 2019-03-20:
OK, I see where the Preferences are coming from. I created a first pull request, #2568.
It would be more consistent to use the value passed in via the command line to also override the value retrieved from the preferences also in other places, but that is out of my reach at the moment.

closed 2019-04-26 by berryma4