ControlSystemStudio/cs-studio #2569: How to compile css with eclipse rcp?

lemrose, 2019-04-02:
I checked the Control System Studio Guide, but there are a lot of errors in importing studio into eclipse rcp as mentioned above. I don't know if there is a problem with my import method. Is there a detailed step?my eclipse version is  4.6，jdk version is 1.8

kasemir, 2019-04-02:
All I can say is that I feel your pain, it's complicated, I'm not aware of any error-free method.

What works for me:

 1) Follow steps on https://github.com/ControlSystemStudio/org.csstudio.sns to get sources and build from the command line. At this time that requires Oracle JDK 8.
 2) Setup Eclipse with a 'target platform' that uses the product's plugins directory which was created as a result of step 1, for example `org.csstudio.sns/repository/target/products/org.csstudio.sns.product.product/linux/gtk/x86_64/sns-css-4.5.0/plugins`
 3) Import the sources into the IDE, using 'Import', 'Maven', 'Existing Maven Projects'.
 4) Wait for it to compile, then delete all the plugins with errors, for example all the `*.rap.*` plugins that aren't used in the desktop RCP product.

willrogers, 2019-04-05:
I agree, it's very complicated. If you are free to use a version that is in development, consider trying Phoebus: https://github.com/shroffk/phoebus

closed 2019-04-26 by berryma4