ControlSystemStudio/cs-studio #2570: Unable to build branch issue_2530_EB

ifmalcuaz, 2019-04-08:
Building cs-studio with the issue_2530_EB branch (Java 11) generates these jfxtras errors:

```
...
[INFO] ------------------------------------------------------------------------
[INFO] Building org.csstudio.logbook.olog.property.fault 1.0.0-SNAPSHOT
[INFO] ------------------------------------------------------------------------
[INFO] 
[INFO] --- maven-clean-plugin:2.5:clean (default-clean) @ org.csstudio.logbook.olog.property.fault ---
[INFO] Deleting /scratch/tph19377/tmp/cs-studio/applications/logbook/logbook-plugins/org.csstudio.logbook.olog.property.fault/target
[INFO] 
[INFO] --- tycho-packaging-plugin:1.3.0:build-qualifier (default-build-qualifier) @ org.csstudio.logbook.olog.property.fault ---
[INFO] The project's OSGi version is 1.0.0.201904080905
[INFO] 
[INFO] --- tycho-packaging-plugin:1.3.0:validate-id (default-validate-id) @ org.csstudio.logbook.olog.property.fault ---
[INFO] 
[INFO] --- tycho-packaging-plugin:1.3.0:validate-version (default-validate-version) @ org.csstudio.logbook.olog.property.fault ---
[INFO] 
[INFO] --- maven-resources-plugin:2.4.3:resources (default-resources) @ org.csstudio.logbook.olog.property.fault ---
[INFO] Using 'UTF-8' encoding to copy filtered resources.
[INFO] skip non existing resourceDirectory /scratch/tph19377/tmp/cs-studio/applications/logbook/logbook-plugins/org.csstudio.logbook.olog.property.fault/src/main/resources
[INFO] 
[INFO] --- tycho-compiler-plugin:1.3.0:compile (default-compile) @ org.csstudio.logbook.olog.property.fault ---
[INFO] Compiling 23 source files to /scratch/tph19377/tmp/cs-studio/applications/logbook/logbook-plugins/org.csstudio.logbook.olog.property.fault/target/classes
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary:
[INFO] 
[INFO] applications ....................................... SUCCESS [  0.209 s]
[INFO] alarm .............................................. SUCCESS [  0.003 s]
[INFO] alarm-plugins ...................................... SUCCESS [  0.002 s]
[INFO] apputil ............................................ SUCCESS [  0.003 s]
[INFO] apputil-plugins .................................... SUCCESS [  0.003 s]
[INFO] org.csstudio.apputil ............................... SUCCESS [  4.756 s]
[INFO] org.csstudio.alarm.beast ........................... SUCCESS [  0.997 s]
[INFO] org.csstudio.utility.speech ........................ SUCCESS [  1.272 s]
[INFO] org.csstudio.alarm.beast.annunciator ............... SUCCESS [  0.731 s]
[INFO] org.csstudio.alarm.beast.annunciator.test .......... SUCCESS [  0.492 s]
[INFO] org.csstudio.alarm.beast.configtool ................ SUCCESS [  0.456 s]
[INFO] org.csstudio.alarm.beast.configtool.test ........... SUCCESS [  0.317 s]
[INFO] cs-studio logbook project .......................... SUCCESS [  0.001 s]
[INFO] Logbook plugins .................................... SUCCESS [  0.002 s]
[INFO] org.csstudio.logbook ............................... SUCCESS [  0.449 s]
[INFO] org.csstudio.apputil.ui ............................ SUCCESS [  1.041 s]
[INFO] appunorganized ..................................... SUCCESS [  0.002 s]
[INFO] appunorganized-plugins ............................. SUCCESS [  0.002 s]
[INFO] org.csstudio.email ................................. SUCCESS [  0.564 s]
[INFO] org.csstudio.alarm.beast.notifier .................. SUCCESS [  0.709 s]
[INFO] org.csstudio.email.ui .............................. SUCCESS [  0.411 s]
[INFO] org.csstudio.alarm.beast.ui ........................ SUCCESS [  1.019 s]
[INFO] org.csstudio.alarm.beast.ui.alarmtable ............. SUCCESS [  0.717 s]
[INFO] org.csstudio.alarm.beast.msghist ................... SUCCESS [  0.781 s]
[INFO] opibuilder ......................................... SUCCESS [  0.002 s]
[INFO] opibuilder-plugins ................................. SUCCESS [  0.002 s]
[INFO] org.csstudio.simplepv .............................. SUCCESS [  0.389 s]
[INFO] org.csstudio.swt.xygraph ........................... SUCCESS [  1.019 s]
[INFO] org.csstudio.swt.widgets ........................... SUCCESS [  1.490 s]
[INFO] org.csstudio.opibuilder ............................ SUCCESS [  4.240 s]
[INFO] org.csstudio.opibuilder.widgets .................... SUCCESS [  2.237 s]
[INFO] org.csstudio.alarm.beast.msghist.opiwidget ......... SUCCESS [  0.573 s]
[INFO] org.csstudio.alarm.beast.msghist.test .............. SUCCESS [  0.506 s]
[INFO] org.csstudio.alarm.beast.notifier.intro ............ SUCCESS [  0.070 s]
[INFO] org.csstudio.alarm.beast.notifier.logbook .......... SUCCESS [  0.433 s]
[INFO] org.csstudio.alarm.beast.notifier.test ............. SUCCESS [  0.453 s]
[INFO] org.csstudio.alarm.beast.test ...................... SUCCESS [  0.566 s]
[INFO] org.csstudio.alarm.beast.server .................... SUCCESS [  0.636 s]
[INFO] org.csstudio.alarm.beast.server.test ............... SUCCESS [  0.391 s]
[INFO] org.csstudio.alarm.beast.ui.alarmtree .............. SUCCESS [  0.518 s]
[INFO] org.csstudio.alarm.beast.ui.areapanel .............. SUCCESS [  0.465 s]
[INFO] org.csstudio.alarm.beast.ui.globaltable ............ SUCCESS [  0.491 s]
[INFO] org.csstudio.alarm.beast.ui.test ................... SUCCESS [  0.431 s]
[... 316 lines cut ...]
	at org.apache.maven.plugin.DefaultBuildPluginManager.executeMojo(DefaultBuildPluginManager.java:134)
	at org.apache.maven.lifecycle.internal.MojoExecutor.execute(MojoExecutor.java:208)
	... 20 more
[ERROR] 
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException
[ERROR] 
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <goals> -rf :org.csstudio.logbook.olog.property.fault
```

I have a local composite p2 repo which points to maven-osgi-bundles and cs-studio-thirdparty, both of which have also checked out the branch issue_2530_EB, as well the p2 repo at the main cs-studio repo.

The output of my `java --version` is:

```
(issue_2530_EB) [tph19377@ws012 applications]$ java --version
java 11.0.2 2019-01-15 LTS
Java(TM) SE Runtime Environment 18.9 (build 11.0.2+9-LTS)
Java HotSpot(TM) 64-Bit Server VM 18.9 (build 11.0.2+9-LTS, mixed mode)
```

And I have my JAVA_HOME variable pointing to my local 11.0.2 folder. Can anyone help with this?

willrogers, 2019-04-08:
See #2530. 

I am working with Ito and I can build this, but I don't know why he can't. Does anyone understand the error? Does it just mean that JavaFX is not there?

willrogers, 2019-04-08:
I just noticed that Ito doesn't seem to be using openjdk. We will try that.

ifmalcuaz, 2019-04-08:
Still the exact same error, even with open jdk11

```
(issue_2530_EB) [tph19377@ws012 applications]$ java --version
openjdk 11.0.2 2019-01-15
OpenJDK Runtime Environment 18.9 (build 11.0.2+9)
OpenJDK 64-Bit Server VM 18.9 (build 11.0.2+9, mixed mode)
```

willrogers, 2019-04-08:
@berryma4 @shroffk any ideas?

berryma4, 2019-04-08:
Let me try to reproduce the error, Kunal recently was able to build though.

willrogers, 2019-04-08:
I can build it without errors, that's the strange thing.

kasemir, 2019-04-08:
With Java 11, the JDK no longer includes JavaFX. It needs to be fetched from org.openjfx.
The error
```
[ERROR] /scratch/tph19377/tmp/cs-studio/applications/logbook/logbook-plugins/org.csstudio.logbook.olog.property.fault/src/jfxtras/controls/agenda/AgendaLastWeekDaysFromDisplayedSkin.java:[236] 
[ERROR] 	public static List<CssMetaData<? extends Styleable, ?>> getClassCssMetaData() {
[ERROR] 	              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[ERROR] The return type is incompatible with AgendaSkinTimeScale24HourAbstract<AgendaLastWeekDaysFromDisplayedSkin>.getClassCssMetaData()
[ERROR] /scratch/tph19377/tmp/cs-studio/applications/logbook/logbook-plugins/org.csstudio.logbook.olog.property.fault/src/jfxtras/controls/agenda/AgendaLastWeekDaysFromDisplayedSkin.java:[247] 
[ERROR] 	public List<CssMetaData<? extends Styleable, ?>> getCssMetaData() {
[ERROR] 	       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[ERROR] The return type is incompatible with AgendaSkinTimeScale24HourAbstract<AgendaLastWeekDaysFromDisplayedSkin>.getCssMetaData()
[ERROR] 2 problems (2 errors)
[ERROR] -> [Help 1]
```
could mean you somehow get two different JavaFX versions into the build.
One might be the OpenJavaFX 11 that you want to use, and the other some older version that the jfxtras dependency drags in.

shroffk, 2019-04-08:
I will double check the jfxtras dependency tree...
Last week I was able to build the entire cs-studio product from the Issue_2030_EB branch without issue.

ifmalcuaz, 2019-04-09:
> With Java 11, the JDK no longer includes JavaFX. It needs to be fetched from org.openjfx.
> The error
>
> ```
> [ERROR] [...]

Sorry, this was the problem; there was a line in my .mavenrc file which was setting JAVA_HOME to a jlink environment, closing the issue, thanks for the help.

closed 2019-04-09 by ifmalcuaz