ControlSystemStudio/cs-studio #2573: LCID BOY macro sometimes returns the same ID

berryma4, 2019-04-15:
Considering this an AtomicInteger, I'm not sure how yet.
(Really says a lot about encapsulation here) 
https://github.com/ControlSystemStudio/cs-studio/blob/87b849003769c9e93616f11d9873e2f9337d0314/applications/opibuilder/opibuilder-plugins/org.csstudio.opibuilder.widgets/src/org/csstudio/opibuilder/widgets/editparts/LinkingContainerEditpart.java#L165

berryma4, 2019-04-15:
although, I'm not sure why this is synchronized or a function to be used once
https://github.com/ControlSystemStudio/cs-studio/blob/87b849003769c9e93616f11d9873e2f9337d0314/applications/opibuilder/opibuilder-plugins/org.csstudio.opibuilder.widgets/src/org/csstudio/opibuilder/widgets/editparts/LinkingContainerEditpart.java#L164

berryma4, 2019-04-15:
```
loc://LCID_1633BIT
loc://LCID_1633BIT<VDouble>(0)
loc://LCID_1633BIT<VDouble>(1)
loc://LCID_1633BIT<VDouble>(10)
loc://LCID_1633BIT<VDouble>(11)
loc://LCID_1633BIT<VDouble>(12)
loc://LCID_1633BIT<VDouble>(13)
loc://LCID_1633BIT<VDouble>(14)
loc://LCID_1633BIT<VDouble>(15)
loc://LCID_1633BIT<VDouble>(16)
loc://LCID_1633BIT<VDouble>(17)
loc://LCID_1633BIT<VDouble>(18)
loc://LCID_1633BIT<VDouble>(19)
loc://LCID_1633BIT<VDouble>(2)
loc://LCID_1633BIT<VDouble>(20)
loc://LCID_1633BIT<VDouble>(21)
loc://LCID_1633BIT<VDouble>(22)
loc://LCID_1633BIT<VDouble>(23)
loc://LCID_1633BIT<VDouble>(24)
loc://LCID_1633BIT<VDouble>(25)
loc://LCID_1633BIT<VDouble>(26)
loc://LCID_1633BIT<VDouble>(27)
loc://LCID_1633BIT<VDouble>(28)
loc://LCID_1633BIT<VDouble>(29)
loc://LCID_1633BIT<VDouble>(3)
loc://LCID_1633BIT<VDouble>(30)
loc://LCID_1633BIT<VDouble>(31)
loc://LCID_1633BIT<VDouble>(4)
loc://LCID_1633BIT<VDouble>(5)
loc://LCID_1633BIT<VDouble>(6)
loc://LCID_1633BIT<VDouble>(7)
loc://LCID_1633BIT<VDouble>(8)
loc://LCID_1633BIT<VDouble>(9)
loc://LCID_1633Chain<VString>("FE_MEBT:BPM_D1056")
loc://LCID_1633Chain<VString>("FE_MEBT:BPM_D1072")
loc://LCID_1633Chain<VString>("FE_MEBT:BPM_D1094")
loc://LCID_1633Chain<VString>("FE_MEBT:BPM_D1111")
loc://LCID_1633Chain<VString>("FS1_BBS:BPM_D2421")
loc://LCID_1633Chain<VString>("FS1_BBS:BPM_D2466")
loc://LCID_1633Chain<VString>("FS1_BMS:BPM_D2502")
loc://LCID_1633Chain<VString>("FS1_BMS:BPM_D2537")
loc://LCID_1633Chain<VString>("FS1_BMS:BPM_D2587")
loc://LCID_1633Chain<VString>("FS1_BMS:BPM_D2600")
loc://LCID_1633Chain<VString>("FS1_BMS:BPM_D2665")
loc://LCID_1633Chain<VString>("FS1_BMS:BPM_D2690")
loc://LCID_1633Chain<VString>("FS1_BMS:BPM_D2702")
loc://LCID_1633Chain<VString>("FS1_BTS:BPM_D2436")
loc://LCID_1633Chain<VString>("FS1_BTS:BPM_D2467")
loc://LCID_1633Chain<VString>("FS1_CSS:BPM_D2212")
loc://LCID_1633Chain<VString>("FS1_CSS:BPM_D2223")
loc://LCID_1633Chain<VString>("FS1_CSS:BPM_D2248")
loc://LCID_1633Chain<VString>("FS1_CSS:BPM_D2278")
loc://LCID_1633Chain<VString>("FS1_CSS:BPM_D2313")
loc://LCID_1633Chain<VString>("FS1_CSS:BPM_D2369")
loc://LCID_1633Chain<VString>("FS1_CSS:BPM_D2383")
loc://LCID_1633Chain<VString>("LS1_BTS:BPM_D1967")
loc://LCID_1633Chain<VString>("LS1_BTS:BPM_D1980")
loc://LCID_1633Chain<VString>("LS1_BTS:BPM_D2027")
loc://LCID_1633Chain<VString>("LS1_BTS:BPM_D2054")
loc://LCID_1633Chain<VString>("LS1_BTS:BPM_D2116")
loc://LCID_1633Chain<VString>("LS1_BTS:BPM_D2130")
loc://LCID_1633Chain<VString>("LS1_CA01:BPM_D1129")
loc://LCID_1633Chain<VString>("LS1_CA01:BPM_D1144")
loc://LCID_1633Chain<VString>("LS1_CA02:BPM_D1163")
loc://LCID_1633Chain<VString>("LS1_CA02:BPM_D1177")
loc://LCID_1633Chain<VString>("LS1_CA03:BPM_D1196")
loc://LCID_1633Chain<VString>("LS1_CA03:BPM_D1211")
loc://LCID_1633Chain<VString>("LS1_CB01:BPM_D1231")
loc://LCID_1633Chain<VString>("LS1_CB01:BPM_D1251")
loc://LCID_1633Chain<VString>("LS1_CB01:BPM_D1271")
loc://LCID_1633Chain<VString>("LS1_CB02:BPM_D1295")
loc://LCID_1633Chain<VString>("LS1_CB02:BPM_D1315")
loc://LCID_1633Chain<VString>("LS1_CB02:BPM_D1335")
loc://LCID_1633Chain<VString>("LS1_CB03:BPM_D1359")
loc://LCID_1633Chain<VString>("LS1_CB03:BPM_D1379")
loc://LCID_1633Chain<VString>("LS1_CB03:BPM_D1399")
loc://LCID_1633Chain<VString>("LS1_CB04:BPM_D1423")
loc://LCID_1633Chain<VString>("LS1_CB04:BPM_D1442")
loc://LCID_1633Chain<VString>("LS1_CB04:BPM_D1462")
loc://LCID_1633Chain<VString>("LS1_CB05:BPM_D1486")
loc://LCID_1633Chain<VString>("LS1_CB05:BPM_D1506")
loc://LCID_1633Chain<VString>("LS1_CB05:BPM_D1526")
loc://LCID_1633Chain<VString>("LS1_CB06:BPM_D1550")
loc://LCID_1633Chain<VString>("LS1_CB06:BPM_D1570")
loc://LCID_1633Chain<VString>("LS1_CB06:BPM_D1590")
loc://LCID_1633Chain<VString>("LS1_CB07:BPM_D1614")
loc://LCID_1633Chain<VString>("LS1_CB07:BPM_D1634")
loc://LCID_1633Chain<VString>("LS1_CB07:BPM_D1654")
loc://LCID_1633Chain<VString>("LS1_CB08:BPM_D1677")
loc://LCID_1633Chain<VString>("LS1_CB08:BPM_D1697")
loc://LCID_1633Chain<VString>("LS1_CB08:BPM_D1717")
loc://LCID_1633Chain<VString>("LS1_CB09:BPM_D1741")
loc://LCID_1633Chain<VString>("LS1_CB09:BPM_D1761")
loc://LCID_1633Chain<VString>("LS1_CB09:BPM_D1781")
loc://LCID_1633Chain<VString>("LS1_CB10:BPM_D1805")
loc://LCID_1633Chain<VString>("LS1_CB10:BPM_D1825")
loc://LCID_1633Chain<VString>("LS1_CB10:BPM_D1845")
loc://LCID_1633Chain<VString>("LS1_CB11:BPM_D1869")
loc://LCID_1633Chain<VString>("LS1_CB11:BPM_D1889")
loc://LCID_1633Chain<VString>("LS1_CB11:BPM_D1909")
loc://LCID_1633Chain<VString>("LS1_WA01:BPM_D1155")
loc://LCID_1633Chain<VString>("LS1_WA02:BPM_D1188")
loc://LCID_1633Chain<VString>("LS1_WA03:BPM_D1222")
loc://LCID_1633Chain<VString>("LS1_WB01:BPM_D1286")
loc://LCID_1633Chain<VString>("LS1_WB02:BPM_D1350")
loc://LCID_1633Chain<VString>("LS1_WB03:BPM_D1413")
loc://LCID_1633Chain<VString>("LS1_WB04:BPM_D1477")
loc://LCID_1633Chain<VString>("LS1_WB05:BPM_D1541")
loc://LCID_1633Chain<VString>("LS1_WB06:BPM_D1604")
loc://LCID_1633Chain<VString>("LS1_WB07:BPM_D1668")
loc://LCID_1633Chain<VString>("LS1_WB08:BPM_D1732")
loc://LCID_1633Chain<VString>("LS1_WB09:BPM_D1796")
loc://LCID_1633Chain<VString>("LS1_WB10:BPM_D1859")
loc://LCID_1633Chain<VString>("LS1_WB11:BPM_D1923")
```
[image]

berryma4, 2019-04-15:
A horrible work around could be to replace the atomic int with a UUID? @shroffk

berryma4, 2019-04-15:
Let me first get rid of the extra sync, since I don't know what that would do.

willrogers, 2019-04-16:
I can't tell from that list of output exactly where the problem was, but the change you suggest is innocuous. I assume that it doesn't fix the problem.

berryma4, 2019-04-16:
I agree this won't fix the problem. But, locking on an atomic is silly (locking to do a CAS).
The clue in the problem might be that all the numbers are the same and it seems like a high number (the last linking container?), although macros get filled in correctly. So the problem is probably in the collection of linking container objects. This part of the BOY code is troublesome though.