ControlSystemStudio/cs-studio #2580: Appliance Archiver data retrieval not working with jdk 11 and latest eclipse

shroffk, 2019-04-23:
I believe the problem is similar to the Olog and Shift issue.

Currently I have verified that the reader is correctly discovered and activated.
The pv Name search and other archiver requests are correctly formulated and resolved.

The request for data fails and I am fairly certain is will be due to some jaxb dependency of pbraw of  one of its dependencies.

shroffk, 2019-04-23:
As expected...
I added the 
jaxb-api  & the com.sun.xml.bind.jaxb-impl to the list of dependencies in the plugin.xml of the archiver reader and now the data retrieval operation works.

@willrogers @berryma4 
Is it fine if we just add the dependency explicitly?

berryma4, 2019-04-23:
thank you for fixing this.

closed 2019-04-23 by berryma4