ControlSystemStudio/cs-studio #2582: resolving dependencies for jersey based applications in java 11 fails

shroffk, 2019-04-23:
```

2019-04-23 16:07:42.777 CONFIG [Thread 1] com.sun.jersey.core.spi.component.ProviderFactory (__getComponentProvider) - A dependent class, com/sun/syndication/io/FeedException, of the component class com.sun.jersey.atom.rome.impl.provider.entity.AtomFeedProvider is not found. The component is ignored.
2019-04-23 16:07:42.777 CONFIG [Thread 1] com.sun.jersey.core.spi.component.ProviderFactory (__getComponentProvider) - A dependent class, com/sun/syndication/io/FeedException, of the component class com.sun.jersey.atom.rome.impl.provider.entity.AtomEntryProvider is not found. The component is ignored.
2019-04-23 16:07:42.778 CONFIG [Thread 1] com.sun.jersey.core.spi.component.ProviderFactory (__getComponentProvider) - A dependent class, org/codehaus/jettison/json/JSONException, of the component class com.sun.jersey.json.impl.provider.entity.JSONArrayProvider$App is not found. The component is ignored.
2019-04-23 16:07:42.778 CONFIG [Thread 1] com.sun.jersey.core.spi.component.ProviderFactory (__getComponentProvider) - A dependent class, org/codehaus/jettison/json/JSONException, of the component class com.sun.jersey.json.impl.provider.entity.JSONArrayProvider$General is not found. The component is ignored.
2019-04-23 16:07:42.779 CONFIG [Thread 1] com.sun.jersey.core.spi.component.ProviderFactory (__getComponentProvider) - A dependent class, org/codehaus/jettison/json/JSONException, of the component class com.sun.jersey.json.impl.provider.entity.JSONObjectProvider$App is not found. The component is ignored.
2019-04-23 16:07:42.779 CONFIG [Thread 1] com.sun.jersey.core.spi.component.ProviderFactory (__getComponentProvider) - A dependent class, org/codehaus/jettison/json/JSONException, of the component class com.sun.jersey.json.impl.provider.entity.JSONObjectProvider$General is not found. The component is ignored.
2019-04-23 16:07:42.782 CONFIG [Thread 1] com.sun.jersey.core.spi.component.ProviderFactory (__getComponentProvider) - A dependent class, org/codehaus/jackson/map/JsonSerializableWithType, of the component class com.sun.jersey.json.impl.provider.entity.JSONWithPaddingProvider is not found. The component is ignored.
2019-04-23 16:07:42.784 CONFIG [Thread 1] com.sun.jersey.core.spi.component.ProviderFactory (__getComponentProvider) - A dependent class, org/codehaus/jackson/jaxrs/JacksonJsonProvider, of the component class com.sun.jersey.json.impl.provider.entity.JacksonProviderProxy is not found. The component is ignored.
2019-04-23 16:07:43.140 WARNING [Thread 46] javax.xml.bind.ContextFinder (getDeprecatedSystemProperty) - Using non-standard property: javax.xml.bind.context.factory. Property javax.xml.bind.JAXBContextFactory should be used instead.
```

While the current version of the third party libraries ensures that jaxb is a defined dependency of olgo, channelfinder, and shift. While using the client some additional dependencies of the jersey library are still not being correctly resolved.

berryma4, 2019-04-24:
I wouldn't say these fail. Do we plan on using an atom feed? These are just components we didn't add a provider, so they are ignored. If we need one of them, then yes let's add a provider for it. But, I don't see where we do.

shroffk, 2019-04-24:
Well...currently the olog requests fail with the above exception stack.
I am guessing it is needed somewhere, I can debug this a bit more after I merge your latest changes/fixes

berryma4, 2019-04-25:
The above is not an exception stack, they are config logs. But there is a runtime jaxb exception (see #2577) until this pull request is merged:
https://github.com/ControlSystemStudio/cs-studio-thirdparty/pull/77

To remove the last warning, you should replace 
```-Djavax.xml.bind.context.factory=com.sun.xml.bind.v2.ContextFactory```
with 
```-Djavax.xml.bind.JAXBContextFactory=com.sun.xml.bind.v2.ContextFactory```
in your xxx.product file.

closed 2019-08-23 by berryma4