ControlSystemStudio/cs-studio #2584: SVG batik warning on /BOY Examples/pictures/SymbolLibrary/SVG/PL General.svg

berryma4, 2019-04-25:
```java
2019-04-25 09:53:59.766 WARNING [Thread 1] org.csstudio.swt.widgets.symbol.SVGSymbolImage (loadDocument) - Error loading SVG image /BOY Examples/pictures/SymbolLibrary/SVG/PL General.svg
org.apache.batik.dom.util.SAXIOException: The prefix "xlink" for attribute "xlink:href" associated with an element type "use" is not bound.
	at org.apache.batik.dom.util.SAXDocumentFactory.createDocument(SAXDocumentFactory.java:459)
	at org.apache.batik.dom.util.SAXDocumentFactory.createDocument(SAXDocumentFactory.java:357)
	at org.apache.batik.anim.dom.SAXSVGDocumentFactory.createDocument(SAXSVGDocumentFactory.java:226)
	at org.csstudio.swt.widgets.symbol.SVGSymbolImage.loadDocument(SVGSymbolImage.java:312)
	at org.csstudio.swt.widgets.symbol.SVGSymbolImage$2.runWithInputStream(SVGSymbolImage.java:280)
	at org.csstudio.swt.widgets.util.AbstractInputStreamRunnable.run(AbstractInputStreamRunnable.java:32)
	at org.eclipse.swt.widgets.RunnableLock.run(RunnableLock.java:40)
	at org.eclipse.swt.widgets.Synchronizer.runAsyncMessages(Synchronizer.java:185)
	at org.eclipse.swt.widgets.Display.runAsyncMessages(Display.java:4906)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:4475)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1173)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:339)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1062)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:156)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:636)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:339)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:563)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:151)
	at org.csstudio.utility.product.Workbench.runWorkbench(Workbench.java:99)
	at org.csstudio.startup.application.Application.startApplication(Application.java:265)
	at org.csstudio.startup.application.Application.start(Application.java:119)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:203)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:137)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:107)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:400)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:659)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:595)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1501)
	at org.eclipse.equinox.launcher.Main.main(Main.java:1474)
Caused by: org.xml.sax.SAXParseException; systemId: file:///home/user/.cs-studio_testest/BOY%20Examples/pictures/SymbolLibrary/SVG/PL%20General.svg; lineNumber: 30; columnNumber: 89; The prefix "xlink" for attribute "xlink:href" associated with an element type "use" is not bound.
	at java.xml/com.sun.org.apache.xerces.internal.util.ErrorHandlerWrapper.createSAXParseException(ErrorHandlerWrapper.java:204)
	at java.xml/com.sun.org.apache.xerces.internal.util.ErrorHandlerWrapper.fatalError(ErrorHandlerWrapper.java:178)
	at java.xml/com.sun.org.apache.xerces.internal.impl.XMLErrorReporter.reportError(XMLErrorReporter.java:400)
	at java.xml/com.sun.org.apache.xerces.internal.impl.XMLErrorReporter.reportError(XMLErrorReporter.java:327)
	at java.xml/com.sun.org.apache.xerces.internal.impl.XMLErrorReporter.reportError(XMLErrorReporter.java:284)
	at java.xml/com.sun.org.apache.xerces.internal.impl.dtd.XMLNSDTDValidator.startNamespaceScope(XMLNSDTDValidator.java:178)
	at java.xml/com.sun.org.apache.xerces.internal.impl.dtd.XMLDTDValidator.handleStartElement(XMLDTDValidator.java:1887)
	at java.xml/com.sun.org.apache.xerces.internal.impl.dtd.XMLDTDValidator.startElement(XMLDTDValidator.java:728)
	at java.xml/com.sun.org.apache.xerces.internal.impl.XMLNSDocumentScannerImpl.scanStartElement(XMLNSDocumentScannerImpl.java:374)
	at java.xml/com.sun.org.apache.xerces.internal.impl.XMLDocumentFragmentScannerImpl$FragmentContentDriver.next(XMLDocumentFragmentScannerImpl.java:2708)
	at java.xml/com.sun.org.apache.xerces.internal.impl.XMLDocumentScannerImpl.next(XMLDocumentScannerImpl.java:605)
	at java.xml/com.sun.org.apache.xerces.internal.impl.XMLNSDocumentScannerImpl.next(XMLNSDocumentScannerImpl.java:112)
	at java.xml/com.sun.org.apache.xerces.internal.impl.XMLDocumentFragmentScannerImpl.scanDocument(XMLDocumentFragmentScannerImpl.java:534)
	at java.xml/com.sun.org.apache.xerces.internal.parsers.XML11Configuration.parse(XML11Configuration.java:888)
	at java.xml/com.sun.org.apache.xerces.internal.parsers.XML11Configuration.parse(XML11Configuration.java:824)
	at java.xml/com.sun.org.apache.xerces.internal.parsers.XMLParser.parse(XMLParser.java:141)
	at java.xml/com.sun.org.apache.xerces.internal.parsers.AbstractSAXParser.parse(AbstractSAXParser.java:1216)
	at java.xml/com.sun.org.apache.xerces.internal.jaxp.SAXParserImpl$JAXPSAXParser.parse(SAXParserImpl.java:635)
	at org.apache.batik.dom.util.SAXDocumentFactory.createDocument(SAXDocumentFactory.java:453)

```

berryma4, 2019-04-26:
Well, it seems this is not a well formed svg file.
The line:
```
<svg xmlns="http://www.w3.org/2000/svg" xmlns:v="http://schemas.microsoft.com/visio/2003/SVGExtensions/" width="0.374331in"
		height="0.374331in" viewBox="0 0 26.9518 26.9518" xml:space="preserve" color-interpolation-filters="sRGB" class="st1">
```
should include xmlns:xlink="http://www.w3.org/1999/xlink":
```
<svg xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink" xmlns:v="http://schemas.microsoft.com/visio/2003/SVGExtensions/" width="0.374331in"
		height="0.374331in" viewBox="0 0 26.9518 26.9518" xml:space="preserve" color-interpolation-filters="sRGB" class="st1">
```
When I correct the svg, there is no error.

berryma4, 2019-04-26:
This looks like the parser is more strict. I would hate to have a lot of broken screens, in which I tell people it's their fault and to add the line. But, on the other hand, this would be fixing the issue in the correct spot.

berryma4, 2019-04-26:
@shroffk @willrogers thoughts?

berryma4, 2019-04-26:
Even with this error gone, the SVGs still don't display because there are two batiks fighting each other.

berryma4, 2019-04-26:
```
An error has occurred. See error log for more details.
loader constraint violation: loader org.eclipse.osgi.internal.loader.EquinoxClassLoader @4bb745c7 (instance of org.eclipse.osgi.internal.loader.EquinoxClassLoader, child of 'platform' jdk.internal.loader.ClassLoaders$PlatformClassLoader) wants to load class org.apache.batik.util.ParsedURL. A different class with the same name was previously loaded by org.eclipse.osgi.internal.loader.EquinoxClassLoader @5a9d08df (instance of org.eclipse.osgi.internal.loader.EquinoxClassLoader, child of 'platform' jdk.internal.loader.ClassLoaders$PlatformClassLoader).
```

berryma4, 2019-04-26:
Continuation of #2448

berryma4, 2019-04-26:
Although now, until 1.11 is released we have to use the batik-all package:
https://issues.apache.org/jira/browse/BATIK-1249
(why the package was changed to 1.10 batik-all throughout cs-studio for java 11)

berryma4, 2019-06-20:
@willrogers  this is the last issue that needs to be fixed before a 4.6 release. Could you look at it?

willrogers, 2019-06-21:
I'll try to have a look next week.

closed 2019-06-28 by berryma4