ControlSystemStudio/cs-studio #2586: failure in org.csstudio.opibuilder.script.ScriptService

berryma4, 2019-04-25:
```java
2019-04-25 10:18:15.030 WARNING [Thread 1] org.csstudio.opibuilder.script.ScriptService (registerScript) - Failed to register moveDots.js. 
java.lang.IllegalArgumentException: Path for project must have only one segment.
java.lang.IllegalArgumentException: Path for project must have only one segment.
	at org.eclipse.core.runtime.Assert.isLegal(Assert.java:66)
	at org.eclipse.core.internal.resources.WorkspaceRoot.getProject(WorkspaceRoot.java:150)
	at org.eclipse.core.internal.resources.Resource.getProject(Resource.java:1127)
	at org.eclipse.core.internal.resources.Resource.getLocation(Resource.java:1068)
	at org.csstudio.opibuilder.util.ResourceUtilSSHelperImpl.workspacePathToSysPath(ResourceUtilSSHelperImpl.java:204)
	at org.csstudio.opibuilder.util.ResourceUtil.workspacePathToSysPath(ResourceUtil.java:216)
	at org.csstudio.opibuilder.script.AbstractScriptStore.init(AbstractScriptStore.java:151)
	at org.csstudio.opibuilder.script.AbstractScriptStore.<init>(AbstractScriptStore.java:82)
	at org.csstudio.opibuilder.script.RhinoScriptStore.<init>(RhinoScriptStore.java:42)
	at org.csstudio.opibuilder.script.ScriptStoreFactory.getJavaScriptStore(ScriptStoreFactory.java:201)
	at org.csstudio.opibuilder.script.ScriptStoreFactory.getScriptStore(ScriptStoreFactory.java:184)
	at org.csstudio.opibuilder.script.ScriptService.registerScript(ScriptService.java:97)
	at org.csstudio.opibuilder.editparts.AbstractBaseEditPart.activate(AbstractBaseEditPart.java:214)
	at org.csstudio.opibuilder.editparts.AbstractPVWidgetEditPart.activate(AbstractPVWidgetEditPart.java:37)
	at org.eclipse.gef.editparts.AbstractEditPart.activate(AbstractEditPart.java:160)
	at org.eclipse.gef.editparts.AbstractGraphicalEditPart.activate(AbstractGraphicalEditPart.java:195)
	at org.csstudio.opibuilder.editparts.AbstractBaseEditPart.activate(AbstractBaseEditPart.java:147)
	at org.csstudio.opibuilder.editparts.DisplayEditpart.activate(DisplayEditpart.java:68)
	at org.eclipse.gef.editparts.AbstractEditPart.addChild(AbstractEditPart.java:215)
	at org.eclipse.gef.editparts.SimpleRootEditPart.setContents(SimpleRootEditPart.java:105)
	at org.eclipse.gef.ui.parts.AbstractEditPartViewer.setContents(AbstractEditPartViewer.java:617)
	at org.eclipse.gef.ui.parts.AbstractEditPartViewer.setContents(AbstractEditPartViewer.java:626)
	at org.csstudio.opibuilder.runmode.OPIRuntimeDelegate$7$2.run(OPIRuntimeDelegate.java:451)
	at org.eclipse.swt.widgets.RunnableLock.run(RunnableLock.java:40)
	at org.eclipse.swt.widgets.Synchronizer.runAsyncMessages(Synchronizer.java:185)
	at org.eclipse.swt.widgets.Display.runAsyncMessages(Display.java:4906)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:4475)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1173)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:339)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1062)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:156)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:636)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:339)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:563)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:151)
	at org.csstudio.utility.product.Workbench.runWorkbench(Workbench.java:99)
	at org.csstudio.startup.application.Application.startApplication(Application.java:265)
	at org.csstudio.startup.application.Application.start(Application.java:119)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:203)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:137)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:107)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:400)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:659)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:595)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1501)
	at org.eclipse.equinox.launcher.Main.main(Main.java:1474)
```

dylan171, 2019-12-18:
This issue was reported this morning:
```
java.lang.IllegalArgumentException: Path for project must have only one segment.
2019-12-17 14:51:10 ERROR: Failed to register ../Scripts/formatTimeDifferenceWithoutSeconds.js. 
java.lang.IllegalArgumentException: Path for project must have only one segment.
2019-12-17 14:51:10 ERROR: Failed to register ../Scripts/formatTimeDifferenceWithoutSeconds.js. 
java.lang.IllegalArgumentException: Path for project must have only one segment.
```
Seems like there was a change in the behaviour Eclipse core, but I need to investigate further. 

@shroffk @kasemir any thoughts?

dylan171, 2019-12-18:
In my testing this error only occurs when the OPI file is loaded via HTTP, loading from the regular file system works as expected.

kasemir, 2019-12-18:
I don't think scripts were ever supported via http.
Once you're inside the script, an 'import' from jython certainly can't use a jython path that includes http://..
The BOY `*ScriptStore` may have supported compiling a 'standalone', one file script that's fetched as an input stream from http, but your stack trace suggests that's not supported. It tries to get the local file name for a potential workspace resource, calling `workspacePathToSysPath`, which requires local files.

dylan171, 2019-12-18:
This particular OPI works as expected with CS Studio v4.5.

The documentation for [IResouse.getLocation()](https://help.eclipse.org/2018-12/index.jsp?topic=%2Forg.eclipse.platform.doc.isv%2Freference%2Fapi%2Forg%2Feclipse%2Fcore%2Fresources%2FIResource.html&anchor=getLocation--) states: 

  ```Returns the absolute path in the local file system to this resource, or null if no path can be determined.```

But that doesn't seem to be strictly true with v2018-12, as it now throws an exception.

dylan171, 2020-01-06:
Regression fixed by PR #2624

closed 2020-01-06 by dylan171