ControlSystemStudio/cs-studio #2588: Unable to have annunciator working

claudio-rosati, 2019-04-26:
Hello all,

a colleague of mine has installed a local version of BEAST the test the Annunciator, but we're unable to have it working.

Beast was configured to have a single topic:

[image]

and it started successfully:

[image]

The alarms can be seen in the CS-Studio Alarm Table but not in the Annunciator one:

[image]

even if the Annunciator configuration seems good (confirmed by the voice annunciating it's ready):

[image]

What are we missing?

berryma4, 2019-04-26:
Is your ActiveMQ server really localhost?

berryma4, 2019-04-26:
I have an issue where the Annunciator stopped working in Debian Stretch (only tested on CS-Studio 4.5.x branch). But, it hangs on connecting to the audio.

claudio-rosati, 2019-04-26:
I've seen that if I set a bad address, when I open the Annunciator view nothing happens. But when I set the right one, it announce itself being connected to the network.
So, being I have that message, I assume it is correct.