ControlSystemStudio/cs-studio #2589: cs-studio project doesn't build with Java 11.0.2

claudio-rosati, 2019-04-29:
Hello,

my Java 11 pipeline fails at this point:

```
[2019-04-29T01:47:54.100Z] Running org.csstudio.logbook.olog.property.fault.FaultConfigurationParsingTest

[2019-04-29T01:47:54.100Z] Tests run: 1, Failures: 0, Errors: 1, Skipped: 0, Time elapsed: 0.041 s <<< FAILURE! - in org.csstudio.logbook.olog.property.fault.FaultConfigurationParsingTest

[2019-04-29T01:47:54.100Z] parserTest(org.csstudio.logbook.olog.property.fault.FaultConfigurationParsingTest)  Time elapsed: 0.018 s  <<< ERROR!

[2019-04-29T01:47:54.100Z] javax.xml.bind.JAXBException: Implementation of JAXB-API has not been found on module path or classpath.

[2019-04-29T01:47:54.100Z] 	at org.csstudio.logbook.olog.property.fault.FaultConfigurationParsingTest.parserTest(FaultConfigurationParsingTest.java:24)

[2019-04-29T01:47:54.100Z] Caused by: java.lang.ClassNotFoundException: com.sun.xml.internal.bind.v2.ContextFactory

[2019-04-29T01:47:54.100Z] 	at org.csstudio.logbook.olog.property.fault.FaultConfigurationParsingTest.parserTest(FaultConfigurationParsingTest.java:24)

[2019-04-29T01:47:54.100Z] 

[2019-04-29T01:47:54.100Z] Running org.csstudio.logbook.olog.property.fault.FaultAdapterTest

[2019-04-29T01:47:54.100Z] Tests run: 4, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.003 s - in org.csstudio.logbook.olog.property.fault.FaultAdapterTest

[2019-04-29T01:47:54.100Z] checkCauseCorrectionFaultTextParsin(org.csstudio.logbook.olog.property.fault.FaultAdapterTest)  Time elapsed: 0.001 s

[2019-04-29T01:47:54.100Z] checkDescRepairFaultTextParsin(org.csstudio.logbook.olog.property.fault.FaultAdapterTest)  Time elapsed: 0 s

[2019-04-29T01:47:54.100Z] checkSimpleFaultTextParsin(org.csstudio.logbook.olog.property.fault.FaultAdapterTest)  Time elapsed: 0 s

[2019-04-29T01:47:54.100Z] checkOnlyDescFaultTextParsin(org.csstudio.logbook.olog.property.fault.FaultAdapterTest)  Time elapsed: 0 s

[2019-04-29T01:47:54.100Z] 

[2019-04-29T01:47:54.100Z] Results:

[2019-04-29T01:47:54.100Z] 

[2019-04-29T01:47:54.100Z] Errors: 

[2019-04-29T01:47:54.100Z]   FaultConfigurationParsingTest.parserTest:24 » JAXB Implementation of JAXB-API ...

[2019-04-29T01:47:54.100Z] 

[2019-04-29T01:47:54.100Z] Tests run: 5, Failures: 0, Errors: 1, Skipped: 0

[2019-04-29T01:47:54.100Z] 

[2019-04-29T01:47:54.100Z] [ERROR] Failed to execute goal org.eclipse.tycho:tycho-surefire-plugin:1.3.0:test (default-test) on project org.csstudio.logbook.olog.property.fault.test: ```

claudio-rosati, 2019-05-06:
Is someone able to build the `cs-studio` project with Java 11?

berryma4, 2019-05-06:
You might need to either add the jaxb classes ie.
https://github.com/ControlSystemStudio/cs-studio/pull/2581/files

Or, add the runtime parameter:
``` -Djavax.xml.bind.context.factory=com.sun.xml.bind.v2.ContextFactory```

This error has been the main issue going to java 11.
And, I'm not sure where all of it is going after reading this blog post on Friday:
http://eclipse-foundation.blog/?p=1492

claudio-rosati, 2019-05-06:
So the situation is that the master branch of CS-Studio projects is broken because of this.
Can we go back to a compilable Java 8 version of the master branch and keep the Java 11 stuff in a different branch until everything is compilable with Java 11?

berryma4, 2019-05-06:
It compiles and runs fine under java 11 (except the current svg issue ticket). You might need to add the runtime parameter.

The master branch is not suppose to be stable. The release branches are stable. If you feel this should be different, we can change our processes in the management review meeting (happens once a year).

If you see problems on the master branch, I would ask you to please help and submit a pull request. We would like to mark it stable as soon as we can.

claudio-rosati, 2019-05-06:
OK I understand how master branch is supposed to be.
The problem about is is that I can't build `cs-studio` with Java 11, so the runtime parameter must be placed somewhere in the master branch. I can do it, but I've not understood where it should be placed.

kasemir, 2019-05-06:
Regarding branches, I agree that it's quite unfortunate but still possible for the master branch to be broken for a short time. The Java 9/10/11/12 move is huge, so master issues are expected.
What I don't quite understand is why the java8 branch, created just before we started to merge the J9-12 changes, is also broken. Fails for me with
`[ERROR]   Missing requirement: org.csstudio.utility.batik 1.1.0.qualifier requires 'osgi.bundle; org.apache.xmlgraphics.batik-all 1.10.0' but it could not be found`

berryma4, 2019-05-06:
You probably need to branch before the batik commit where all was used.

shroffk, 2019-05-31:
While this might not be related....we recently discovered that the master branch does not build with tycho 1.3 and maven 3.6
To build the master branch it is necessary to use maven version <= 3.5.3

claudio-rosati, 2019-06-03:
I've tried the newly released [Maven 3.6.1](https://maven.apache.org/docs/3.6.1/release-notes.html) and now the problem is no more in the `cs-studio` project, but in the `cs-studio-thirdparty` one:

```
[2019-06-03T12:28:16.599Z] + cd cs-studio-thirdparty
[2019-06-03T12:28:16.599Z] + mvn -q -B -P ess-css-settings,platform-default,csstudio-composite-repo-enable,eclipse-sites -Dmaven.repo.local=/home/srv_icsjenkins/workspace/CS-Studio-CE/css-maven-repo/.m2 -Dcsstudio.composite.repo=/home/srv_icsjenkins/workspace/CS-Studio-CE/ess-css-extra/ess_css_comp_repo --settings ../ess-css-extra/maven/settings.xml clean verify
[2019-06-03T12:28:23.178Z] [ERROR] Cannot resolve project dependencies:
[2019-06-03T12:28:23.178Z] [ERROR]   Software being installed: edu.msu.nscl.olog.api 2.2.8.qualifier
[2019-06-03T12:28:23.178Z] [ERROR]   Missing requirement: edu.msu.nscl.olog.api 2.2.8.qualifier requires 'osgi.bundle; org.eclipse.core.runtime 0.0.0' but it could not be found
[2019-06-03T12:28:23.178Z] [ERROR] 
[2019-06-03T12:28:23.178Z] [ERROR] See http://wiki.eclipse.org/Tycho/Dependency_Resolution_Troubleshooting for help.
[2019-06-03T12:28:23.178Z] [ERROR] Cannot resolve dependencies of MavenProject: org.csstudio:edu.msu.nscl.olog.api:2.2.8-SNAPSHOT @ /home/srv_icsjenkins/workspace/CS-Studio-CE/cs-studio-thirdparty/plugins/edu.msu.nscl.olog.api/pom.xml: See log for details -> [Help 1]
```

shroffk, 2019-06-03:
Sorry, 
I was pointing out that you need to use and older version of maven
use 3.5.3

claudio-rosati, 2019-06-03:
What you said is that there was a problem with tycho 1.3 and Maven 3.6, not 3.6.1 ;-)

Being that there was some fixes in Maven 3.6.1 I wanted to give it a try, and the funny thing is that the error happens earlier on another project. 

I’m not sure 3.5.3 is good enough for running some plugins required for java 11. Before moving to 3.6 I used 3.5.4... but I’ll leave that exercise to Georg and Fredrik

berryma4, 2019-06-04:
From the Tycho mailing list:

Maven 3.6.1 has a bug that prevents from running Tycho builds successfully in most cases: https://bugs.eclipse.org/bugs/show_bug.cgi?id=546463 . Any other version of Maven should work.

claudio-rosati, 2019-06-05:
Understood. I've tried on my macOS where I was able to build Java 8 CS-Studio with Maven 3.6.0 but not with 3.6.1.

claudio-rosati, 2019-06-05:
So yes, at lease Java 8 CS-Studio can be definitively build with Maven 3.6.0 but not 3.6.1.
So @berryma4 do you think, as @shroffk said, Maven 3.5.3 is really needed to build with Java 11? Is the problem I have building `cs-studio-thirdparty` related to the version of Maven used?

willrogers, 2019-06-10:
I have just built my CS-Studio product end-to-end using jdk-11.0.2 and Maven 3.5.0. I have opened it and everything I have tested works. The only strange thing I found was a long pause the first time I opened a JavaFX view. Perhaps this was some glitch.

I'm keen to make a proper release of 4.6 for Diamond soon. Thanks for everyone's work on this.

willrogers, 2019-06-12:
There are two tests failing on the master branch:

* org.csstudio.swt.rtplot.util.TemporalRoundingTest.testDST
* org.csstudio.logbook.olog.property.fault.FaultConfigurationParsingTest.parserTest

The first is to do with timezones, but I don't know what has changed:

```
java.lang.AssertionError: 
Expected: "2014-03-08 23:00:00 (EST, UTC -0500)"
     but: was "2014-03-08 23:00:00 (GMT-05:00, UTC -0500)"
	at org.hamcrest.MatcherAssert.assertThat(MatcherAssert.java:20)
	at org.junit.Assert.assertThat(Assert.java:956)
	at org.junit.Assert.assertThat(Assert.java:923)
	at org.csstudio.swt.rtplot.util.TemporalRoundingTest.testDST(TemporalRoundingTest.java:190)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at org.junit.runners.model.FrameworkMethod$1.runReflectiveCall(FrameworkMethod.java:50)
	at org.junit.internal.runners.model.ReflectiveCallable.run(ReflectiveCallable.java:12)
	at org.junit.runners.model.FrameworkMethod.invokeExplosively(FrameworkMethod.java:47)
	at org.junit.internal.runners.statements.InvokeMethod.evaluate(InvokeMethod.java:17)
	at org.junit.runners.ParentRunner.runLeaf(ParentRunner.java:325)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:78)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:57)
	at org.junit.runners.ParentRunner$3.run(ParentRunner.java:290)
	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:71)
	at org.junit.runners.ParentRunner.runChildren(ParentRunner.java:288)
	at org.junit.runners.ParentRunner.access$000(ParentRunner.java:58)
	at org.junit.runners.ParentRunner$2.evaluate(ParentRunner.java:268)
	at org.junit.runners.ParentRunner.run(ParentRunner.java:363)
	at org.eclipse.jdt.internal.junit4.runner.JUnit4TestReference.run(JUnit4TestReference.java:89)
	at org.eclipse.jdt.internal.junit.runner.TestExecution.run(TestExecution.java:41)
	at org.eclipse.jdt.internal.junit.runner.RemoteTestRunner.runTests(RemoteTestRunner.java:541)
	at org.eclipse.jdt.internal.junit.runner.RemoteTestRunner.runTests(RemoteTestRunner.java:763)
	at org.eclipse.jdt.internal.junit.runner.RemoteTestRunner.run(RemoteTestRunner.java:463)
	at org.eclipse.jdt.internal.junit.runner.RemoteTestRunner.main(RemoteTestRunner.java:209)
```

The second one is because of the JAXB loading. I have apparently made the same changes as elsewhere, but it still fails with 

```
Running org.csstudio.logbook.olog.property.fault.FaultConfigurationParsingTest
Tests run: 1, Failures: 0, Errors: 1, Skipped: 0, Time elapsed: 0.057 s <<< FAILURE! - in org.csstudio.logbook.olog.property.fault.FaultConfigurationParsingTest
testCheckParserTest(org.csstudio.logbook.olog.property.fault.FaultConfigurationParsingTest)  Time elapsed: 0.027 s  <<< ERROR!
javax.xml.bind.JAXBException: Implementation of JAXB-API has not been found on module path or classpath.
	at org.csstudio.logbook.olog.property.fault.FaultConfigurationParsingTest.testCheckParserTest(FaultConfigurationParsingTest.java:29)
Caused by: java.lang.ClassNotFoundException: com.sun.xml.internal.bind.v2.ContextFactory cannot be found by org.csstudio.logbook.olog.property.fault_1.0.0.201906120740
	at org.csstudio.logbook.olog.property.fault.FaultConfigurationParsingTest.testCheckParserTest(FaultConfigurationParsingTest.java:29)
```

Do the tests have a different classloading mechanism?

willrogers, 2019-06-12:
My attempts to fix the second test are in #2598

kasemir, 2019-06-12:
Regarding the time stamp tests, in the JFX version of that library we added this to the start of TemporalRoundingTest to get a known locale:
```
public TemporalRoundingTest()
{
    Locale.setDefault(Locale.US);
}
```

Does that help when you run the swt.rtplot.util.TemporalRoundingTest?

willrogers, 2019-06-12:
Yes, PR submitted.

closed 2019-06-20 by berryma4