ControlSystemStudio/cs-studio #2591: Beast Annunciation broken in debian openjdk8

berryma4, 2019-05-03:
The Beast Annunciation hangs on startup in debian openjdk 8.

berryma4, 2019-05-03:
The underline issue in openjdk8 is with a change that debian made to the sound properties.

Issue:
https://github.com/Debian/openjdk-8/issues/1

Fix:

Edit $JAVA_HOME/jre/lib/sound.properties and comment out everything related to ```org.classpath.icedtea.pulseaudio``` and uncomment these lines:

```java
javax.sound.sampled.Clip=com.sun.media.sound.DirectAudioDeviceProvider
javax.sound.sampled.Port=com.sun.media.sound.PortMixerProvider
javax.sound.sampled.SourceDataLine=com.sun.media.sound.DirectAudioDeviceProvider
javax.sound.sampled.TargetDataLine=com.sun.media.sound.DirectAudioDeviceProvider
```

closed 2019-05-03 by berryma4

kasemir, 2019-05-06:
Thanks for figuring this out!

mcgivrer, 2020-02-14:
You save my life !