ControlSystemStudio/cs-studio #2595: Large detector not displaying images.

kgofron, 2019-06-07:
When we are using Dexela detector with 3888 x 3072 pel, the images do not display initially in CSS. When hovering over image in PVA display it  shows 'null'.
The workaround is to bin 2x2 image, and images will display in CSS with binned dimensions. After image is unbined , and images will finally display in CSS with full size of 3888 x 3072.
Other clients such as caget are getting the images. The issue is present for both PVA and CA versions of CSS opi.

kasemir, 2019-06-10:
Works fine for me with recent software, i.e. using the Phoebus-based CS-Studio and Area Detector R3-3-2.

Specifically, I'm trying the VM image used in last week's EPICS 7 intro from https://controlssoftware.sns.ornl.gov/training/2019_USPAS/.
See 99 Spring 2019 EPICS 7 Intro document for starting the simulated area detector and PVA display.

In `/home/training/epics-train/tools/areaDetector/ADSimDetector/iocs/simDetectorIOC/iocBoot/iocSimDetector`, edit `st_base.cmd` to update XSIZE and YSIZE.

When I set the simulated area detector image size to 3888 x 3072, the PVA image shows  up fine:

[image]

But that's with the most recent PVA library, not sure what has changed in between.

shroffk, 2019-06-21:
Kaz are you using the nsls2 nightly builds?

kgofron, 2019-06-21:
On my laptop, I downloaded and am using 4.61 nightly build from February. But the 28ID where the problem is visible is standard beamline issue. Please work with Ji Li.