ControlSystemStudio/cs-studio #2602: Runtime problems with CS-Studio 4.6

willrogers, 2019-07-01:
I'm not sure I'm going to make any friends here, but:

CS-Studio 4.6 is now building for me reliably, which seems a huge step forwards. However, I'm seeing lots of different runtime problems. I don't know if this is indicating one problem somewhere or more than one, but there are a variety of errors reported:

* "Workspace is closed" - see [your discussion with Tom Schindl](https://www.eclipse.org/forums/index.php/t/1096912/)
* "cs-studio" is not responding - popup from Gnome, giving me the options to Force Quit or Wait
* `2019-07-01 16:00:30.695 SEVERE [Thread 1] org.csstudio.logging.PluginLogListener (logging) - Unhandled event loop exception
org.eclipse.e4.core.di.InjectionException: java.lang.NullPointerException
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:68)`
* "Error connecting to archive server URL ... Detail: null" (In a popup)
* `OpenJDK 64-Bit Server VM warning: Potentially dangerous stack overflow in ReservedStackAccess annotated method java.util.concurrent.locks.ReentrantReadWriteLock$Sync.tryAcquireShared(I)I [1]` - could be nothing
* `2019-07-01 15:56:46.369 CONFIG [Thread 1] sun.util.locale.provider.LocaleServiceProviderPool (config) - A locale sensitive service provider returned null for a localized objects,  which should not happen.  provider: sun.util.cldr.CLDRCalendarDataProviderImpl@4652c74d locale: en` - this sounds innocuous

I will try building the common product and see if I get any of the same problems.

berryma4, 2019-07-01:
"Workspace is closed" was a bug in the efx plugin which broke many things. Make sure your product has the correct version. Careful, the wrong version is in the eclipse p2, and you need the specific efx p2.

willrogers, 2019-07-01:
I've got the 3.4.3 versions from the `runtime-released` p2. `org.csstudio.product` is still using the nightly versions and is now up to 3.6.0. That seems to be working better but I don't know if that's why.

`cs-studio` is referring to the 3.4.3 repo I think.

willrogers, 2019-07-01:
The common product seems to have `org.eclipse.osgi.compatibility.state`, which isn't in my product.

berryma4, 2019-07-01:
You need this fix in the efxclipse runtime:
https://github.com/eclipse-efx/efxclipse-rt/issues/292
which it looks like is released now:
https://download.eclipse.org/efxclipse/runtime-released/3.5.0/

berryma4, 2019-07-01:
where is "org.eclipse.osgi.compatibility.state" in the common product?

willrogers, 2019-07-02:
Thanks, when I got 3.5.0 in my build the errors seem to be resolved.

closed 2019-07-02 by willrogers