ControlSystemStudio/cs-studio #2607: Phoebus find open port and app in path

berryma4, 2019-07-24:
I would like to have the integration plugin pick an open port, and find phoebus in the user path.

shroffk, 2019-07-30:
Using the random port seems to not make much sense to me.
In that case it might be best to not use the ```-server``` options, so that each launch is a standalone instance of phoebus

berryma4, 2019-07-30:
I was thinking one server instance (with port) per eclipse instance.

shroffk, 2019-07-30:
I see...
Can I implement the following:
```
# launch the phoebus application with the -server argument, use a random available port
use_phoebus_server=true
```

This way, if my launcher script takes care of the port number then I can disable that by setting the above property to false

berryma4, 2019-07-30:
My pull request you merged works. But, I'm combining two things, find
Phoebus in the user's path and pick an open port.

shroffk, 2019-07-30:
I know, I have added on top of that

shroffk, 2019-07-30:
https://github.com/ControlSystemStudio/cs-studio/pull/2610

closed 2019-08-23 by berryma4