ControlSystemStudio/cs-studio #2616: Csstudio prerequisites not defined: java tycho p2 etc.

antoinetran, 2019-10-15:
Dear all,

I tried to upgrade our Eclipse RCP product, from Eclipse mars with Csstudio 4.3, to Eclipse Neon/Csstudio 4.5 and Eclipse 2018-12/Csstudio 4.6. We use Maven 3.5.3 and tycho 1.2.0.

I can see prerequisites chapter in doc: [here](http://cs-studio.sourceforge.net/docbook/ch04.html#idm140287029307424).
> ask other CSS developers for the currently supported versions

While the version of Eclipse RCP is clear and indicated in each Csstudio release, it is not the case for JDK.
It is the same for Tycho, I had to update from 1.2.0 to 1.4.0, after discovering in some Github/stackeverflow page that this is due to Tycho.

I described build dependencies. But we also need to know the runtime dependencies. For example, I discovered that even if I build with OpenJdk 8 with success, the Eclipse RCP fails to launch because it needs JavaSe 11. Right now, our team might revert from 4.6 to 4.5 to verify if Java 8 is ok or not, we have no choice to take lots of time to downgrade because we are currently locked to Java 8.

Now related to P2 plugin dependencies, I took so much time fixing missing dependencies.
At build, migrating from 4.3 to 4.6 could not work without:

```
<unit id="org.eclipse.gmf.feature.group" version="0.0.0"/>
<repository location="https://download.eclipse.org/efxclipse/runtime-released/3.4.3/site"/>
<unit id="org.csstudio.platform.libs.epics" version="0.0.0"/>
<unit id="org.epics.pvdata" version="0.0.0"/>
<unit id="org.epics.pvaccess" version="0.0.0"/>
```

For runtime dependencies, I also had to add:
```
<unit id="org.epics.util" version="0.0.0"/>
<unit id="org.epics.jca" version="0.0.0"/>
<unit id="org.apache.xmlgraphics.batik-XXX" version="0.0.0"/>
```

And to be really sure, I added (might be too much):
```
<unit id="org.epics.caj" version="0.0.0"/>
<unit id="org.diirt.support.diirt-ca" version="0.0.0"/>
<unit id="org.epics.channelfinder.cf-client-api" version="0.0.0"/>
<unit id="org.epics.vtype" version="0.0.0"/>
<unit id="org.apache.xmlgraphics.batik-all" version="0.0.0"/>
```

When I tried downgrading to from 4.6 to 4.5, I missed:
`[ERROR]   Missing requirement: org.csstudio.utility.batik 1.1.0.201902070940 requires 'osgi.bundle; org.apache.batik.bridge [1.7.0,1.8.0)' but it could not be found`

And I have no idea where to find it.

To summarize, shouldn't csstudio defines its build and runtime dependencies? Thank you!!

antoinetran, 2019-10-15:
Linked to Java11 #2589

kasemir, 2019-10-15:
I think it's fair to say that the Eclipse-based build process has become quite difficult. At this time, the org.csstudio.product just started to compile and run with Java 11, but not all sites have been successful in adapting their site-specific product.
The new 'phoebus' development that offers most of the CS-Studio functionality without Eclipse, https://github.com/shroffk/phoebus, is much easier to build and a more practical way forward.
I'd look at that unless you really need to stay with the Eclipse-based version for some reason.

shroffk, 2019-10-17:
I had similar problems with the batik dependencies...I reverted to using tycho 1.3 

My nightly builds still seem to work... the build from last night succeeded.
Here is the recipe:
https://github.com/cs-studio-nsls2/build/blob/no-docker/jenkins/cs-studio-build

I have to admit that the process by which dependency resolution of tycho and eclipse bundles is become a bit of a mystery.
I think we as a community need to take a concerted effort to eliminate all "import packages" uses in cs-studio plugins.
We should also lock version numbers and not use version=0.0.0