ControlSystemStudio/cs-studio #2618: CS-Studio build improvements

shroffk, 2019-10-17:
During the developers meeting there were some concerns shared about the complexity of the build and the amount if time each build takes.

One potential idea to address this issue would be to merge the following repos into one.

- maven-osgi-bundles  
- cs-studio-thirdparty  
- cs-studio  
- org.csstudio.product.  

In my experience, these projects don't have an individual life cycle. They are very tightly coupled with each other, bringing them all together in a single build might reduce the time taken to fetch the artifact info for all the common p2 repos (eclipse, efxclipse, etc....) by eliminating the need to do so 4 times.

shroffk, 2019-10-17:
https://github.com/ControlSystemStudio/cs-studio/tree/2618_restructured_repo
Here is the branch on which I have tried to restructure the repo as described in the issue ticket.
The new structure is as follows:

[image]

Pros:  
Building each module independently took me a total of around 59 mins...with the new repo the time came down to around 30 mins.

The poms are a lot cleaner now. The number of lines of pom.xml has greatly reduced since the common profiles, plugins, configurations, properties, repositories, etc...  are all in one master pom. This makes them easier to manage and maintain and easier to understand.

Cons:  
Sometimes the build fails, with similar errors as described in the jdk 11 build issue ticket. This might have nothing to do with the restructured repo itself...

I moved the other repo with the history...but we need to rewrite the history with the new paths.

@willrogers @css-iter ISIS other...

willrogers, 2019-10-18:
@shroffk thanks for this, I expect it is an improvement.

I hope to get around to revisiting the CS-Studio build in a couple of weeks.

ifmalcuaz, 2019-11-01:
cc @willrogers 

I am able to build it without errors on my machine at Diamond; how do you build a separate product?

willrogers, 2019-11-01:
@shroffk how do you build the NSLS-II product with this structure?

shroffk, 2019-11-01:
After completing the consolidated build of the complete cs-studio repo I build the nsls2 repo pointing to this build p2 repo

https://github.com/cs-studio-nsls2/org.csstudio.nsls2.product.git

```
mvn -DskipTests=true clean verify -Dcsstudio.composite.repo=${WORKSPACE}/p2repo
```

where `${WORKSPACE}/p2repo` is the p2repo from the above structure.

shroffk, 2019-11-01:
@ifmalcuaz @willrogers how long did your build take?

ifmalcuaz, 2019-11-08:
@shroffk applications took 35 minutes, the others a bit less

shroffk, 2020-07-23:
So I would say that build improvement has generally been achieved.
Build times are now much lesser
Travis now builds & runs all unit tests

closed 2020-07-23 by shroffk